davidjo/snd_hda_macbookpro #89: Is it impossible to make changes before merging?

WOLFRAZOR, 2023-05-12:
I want to patch the microphone. The file was found in that issue, but the worst is the file version is too old (the version before the signature is removed). Even if I use UltraEdit to compare line by line and copy the necessary parts, it doesn't solve the problem that cannot be patched. 

(I hope the author merges these modifications and saves unnecessary trouble.)

While I rarely use my computer's built-in mic, sometimes the internal mic is important when wireless headphones can't be used.

davidjo, 2023-05-13:
I have just updated the repo with internal mike fixups which restores base kernel internal mike operation.
I have no idea what UltraEdit is - the linux way would be to use diff3.

WOLFRAZOR, 2023-05-14:
Note: UltraEdit is a multi-functional notepad, this application is an artifact on Windows and macOS.

Thanks to the developer (otherwise I would have to reinstall another system because of the microphone)😂.

I don't want to sell this machine (there is no difference in security between a normal machine and a machine with T2... This really doesn't bother those hackers. Replace the chip and fingerprint module together, and those computers obtained through illegal means will be reset then into the market)

closed 2023-05-14 by WOLFRAZOR