davidjo/snd_hda_macbookpro #91: T2 Ubuntu Lunar kernel 6.3.1 - linux-headers-6.3.1-t2-lunar - No source in /usr/src

lui75, 2023-05-17:
Hi David, I managed to got sound working on a vanilla installation of Ubuntu Lunar, but given the fact that I couldn't resolve a problem with the wifi on my iMac 19,1 I decided to try the T2 Ubuntu lunar iso which has 6.3.1 kernel. Wifi now works perfectly, but when I launch your script it can not find the linux headers it needs to build the hda modules necessary for my machine, even though they are installed inside the dir /usr/src/linux-headers-t2-lunar.
Is there a way to let your script do its job?
Thanks for your help!

davidjo, 2023-05-17:
I suspect this is a naming issue - the script assumes that uname -r  gives you the kernel version that is used under /lib/modules.
Once you have run the script you can go into the eg build/hda-6.3.1  (maybe a t2 somewhere) directory and just run eg make clean to clean, make to build and sudo make install to install (Note if you originally ran with sudo you may need to chown the files in the directory to your user - or add sudo to all make commands).
The Makefile can take a parameter KVER=xxxxx where xxxxx is the version name under /lib/modules - by default it uses uname -r to set KDIR - which is the primary path variable for building/installing. (The script is just for ease of use - my original building was done manually).
Or it maybe like hwe kernels which also mess things up.
(Curiously Ive just upgraded to lunar (6.2 kernel) and surprisingly the script just worked).
Have to say Ive had issues with Wifi on my MacbookPro - it needed some messing with the firmware - I have a special file under /lib/firmware/brcm brcmfmac43602-pcie.txt which sets some options which was needed - havent checked if this is still needed - certainly was still needed for the 22.04 LTS. Still sometimes it seems to get stuck.

lui75, 2023-05-17:
Wow that's fascinating! The problem is: I do not have the linux-source-6.3.1-t2-lunar.tar.bz2

Can I use the files that are inside my /usr/src/linux-headers-6.3.1-t2-lunar ?

davidjo, 2023-05-17:
No - those are just the header files - you need the actual source - which looking at T2 build.sh seems to be loading from a git repo for ubuntu kernels.
You need to figure out the source being used - if you built the T2 kernel  its likely stored under whereever you did the build.

lui75, 2023-05-17:
I am building the kernel right now. Found the sound/pci/hda dir, the build dir is /root/work/linux-source

davidjo, 2023-05-17:
Just copy the files from /root/work/linux-source/sound/pci/hda  - eg cp -p /root/work/linux-source/sound/pci/hda/* into the build/hda  dir of my repo clone (or eg rsync -av  /root/work/linux-source/sound/pci/hda build).

lui75, 2023-05-17:
the hda dir is ready inside the build dir, but then I got lost - don't know which patches I need to apply.
I moved the hda dir and renamed both hda dir and its original makefile.

davidjo, 2023-05-18:
So you need to do the following:
in build/hda-x.y.z do:
patch -b -p2 <../../patch_patch_cs8409.h.diff
then copy in my files:
cp ../../patch_cirrus/Makefile ../../patch_cirrus/patch_cirrus_* .
then do:
patch -b -p2 <../../patch_patch_cirrus_apple.h.diff

Now you can do the make commands.

(Complexity only comes from trying do deal with multiple kernel versions and other distributions).

lui75, 2023-05-18:
Unfortunately, it didn't work.

davidjo, 2023-05-18:
So I need to know exactly what happened not sure what you mean by it didnt work.
Did you get a module in build/hda-x.y.z ie the file snd-hda-codec-cs8409.ko?  If this file exists then the module building worked.
One issue I forgot is the install - you likely need to create an updates directory under /lib/modules/version  - sudo mkdir updates - before running sudo make install.
Also try going into /lib/modules/version and running sudo depmod -a.

lui75, 2023-05-18:
the dir updates was already there. Do I have to run sudo make & sudo make install from inside hda-6.3.3)

lui75, 2023-05-18:
This is what I get before the make install:
`luigi@luigi-iMac:~/snd_hda_macbookpro$ sudo mv build/hda-6.3.3/Makefile build/hda-6.3.3/Makefile.origin
luigi@luigi-iMac:~/snd_hda_macbookpro$ cd build/hda-6.3.3
luigi@luigi-iMac:~/snd_hda_macbookpro/build/hda-6.3.3$ sudo patch -b -p2 <../../patch_patch_cs8409.h.diff
patching file patch_cs8409.h
Hunk #2 succeeded at 303 (offset 1 line).
Hunk #3 succeeded at 330 (offset 1 line).
Hunk #4 succeeded at 366 (offset 1 line).
luigi@luigi-iMac:~/snd_hda_macbookpro/build/hda-6.3.3$ sudo cp ../../patch_cirrus/Makefile ../../patch_cirrus/patch_cirrus_* .
luigi@luigi-iMac:~/snd_hda_macbookpro/build/hda-6.3.3$ sudo patch -b -p2 <../../patch_patch_cirrus_apple.h.diff
patching file patch_cirrus_apple.h
luigi@luigi-iMac:~/snd_hda_macbookpro/build/hda-6.3.3$ sudo make
make -C /lib/modules/6.3.3-t2-lunar/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/luigi/snd_hda_macbookpro/build/hda-6.3.3 modules
make[1]: ingresso nella directory «/usr/src/linux-headers-6.3.3-t2-lunar»
  CC [M]  /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/patch_cs8409.o
  CC [M]  /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/patch_cs8409-tables.o
  LD [M]  /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/snd-hda-codec-cs8409.o
  MODPOST /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/Module.symvers
  CC [M]  /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/snd-hda-codec-cs8409.ko
  BTF [M] /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/snd-hda-codec-cs8409.ko
Skipping BTF generation for /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/snd-hda-codec-cs8409.ko due to unavailability of vmlinux
make[1]: uscita dalla directory «/usr/src/linux-headers-6.3.3-t2-lunar»
luigi@luigi-iMac:~/snd_hda_macbookpro/build/hda-6.3.3$ 
`

lui75, 2023-05-18:
Here's what I got after make:

luigi@luigi-iMac:~/snd_hda_macbookpro$ sudo mv build/hda-6.3.3/Makefile build/hda-6.3.3/Makefile.origin
luigi@luigi-iMac:~/snd_hda_macbookpro$ cd build/hda-6.3.3
luigi@luigi-iMac:~/snd_hda_macbookpro/build/hda-6.3.3$ sudo patch -b -p2 <../../patch_patch_cs8409.h.diff
patching file patch_cs8409.h
Hunk #2 succeeded at 303 (offset 1 line).
Hunk #3 succeeded at 330 (offset 1 line).
Hunk #4 succeeded at 366 (offset 1 line).
luigi@luigi-iMac:~/snd_hda_macbookpro/build/hda-6.3.3$ sudo cp ../../patch_cirrus/Makefile ../../patch_cirrus/patch_cirrus_* .
luigi@luigi-iMac:~/snd_hda_macbookpro/build/hda-6.3.3$ sudo patch -b -p2 <../../patch_patch_cirrus_apple.h.diff
patching file patch_cirrus_apple.h
luigi@luigi-iMac:~/snd_hda_macbookpro/build/hda-6.3.3$ sudo make
make -C /lib/modules/6.3.3-t2-lunar/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/luigi/snd_hda_macbookpro/build/hda-6.3.3 modules
make[1]: ingresso nella directory «/usr/src/linux-headers-6.3.3-t2-lunar»
  CC [M]  /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/patch_cs8409.o
  CC [M]  /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/patch_cs8409-tables.o
  LD [M]  /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/snd-hda-codec-cs8409.o
  MODPOST /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/Module.symvers
  CC [M]  /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/snd-hda-codec-cs8409.ko
  BTF [M] /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/snd-hda-codec-cs8409.ko
Skipping BTF generation for /home/luigi/snd_hda_macbookpro/build/hda-6.3.3/snd-hda-codec-cs8409.ko due to unavailability of vmlinux
make[1]: uscita dalla directory «/usr/src/linux-headers-6.3.3-t2-lunar»
luigi@luigi-iMac:~/snd_hda_macbookpro/build/hda-6.3.3$

lui75, 2023-05-18:
And here's the result of the install:

luigi@luigi-iMac:/snd_hda_macbookpro/build/hda-6.3.3$ sudo make install
#cp snd-hda-codec-cs8409.ko /lib/modules/6.3.3-t2-lunar/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/6.3.3-t2-lunar/build M=/home/luigi/snd_hda_macbookpro/build/hda-6.3.3 CONFIG_MODULE_SIG_ALL=n modules_install
make[1]: ingresso nella directory «/usr/src/linux-headers-6.3.3-t2-lunar»
  INSTALL /lib/modules/6.3.3-t2-lunar/updates/snd-hda-codec-cs8409.ko
  DEPMOD  /lib/modules/6.3.3-t2-lunar
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: uscita dalla directory «/usr/src/linux-headers-6.3.3-t2-lunar»
depmod -a
luigi@luigi-iMac:/snd_hda_macbookpro/build/hda-6.3.3$

lui75, 2023-05-18:
The module was created

davidjo, 2023-05-18:
So that all looks good - a primary issue is to ensure the patch commands all say Hunk X succeeded - which they do.
The messages from the build are the expected messages.
The install messages all look good  - the kernel internal build DEPMOD is skipped but then make install does an explicit depmod -a right at the end.
Just check the module is under /lib/modules/6.3.3-t2-lunar/updates.

So what doesnt work - does sound not play?

do a lsmod and grep for the module to ensure its loaded.

check the syslog and ensure you have the message which includes Primary patch_cs8409 NOT FOUND trying APPLE

lui75, 2023-05-18:
I don't know how to do that. There is no file inside /var/log

davidjo, 2023-05-18:
Well that suggests the module isnt loading - do the lsmod check.
No files at all under /var/log ?? that doesnt make sense.

lui75, 2023-05-18:
There are files but not syslog
here's lsmod:

luigi@luigi-iMac:~$ lsmod
Module                  Size  Used by
rfcomm                 94208  4
snd_seq_dummy          16384  0
snd_hrtimer            16384  1
ccm                    20480  0
algif_aead             16384  0
des_generic            16384  0
libdes                 36864  1 des_generic
md4                    16384  0
cmac                   16384  3
algif_hash             20480  1
algif_skcipher         16384  1
af_alg                 32768  7 algif_hash,algif_skcipher,algif_aead
bnep                   32768  2
brcmfmac_wcc           16384  0
amdgpu              14262272  29
binfmt_misc            24576  1
snd_sof_pci_intel_cnl    16384  0
snd_sof_intel_hda_common   196608  1 snd_sof_pci_intel_cnl
soundwire_intel        57344  1 snd_sof_intel_hda_common
soundwire_generic_allocation    16384  1 soundwire_intel
soundwire_cadence      40960  1 soundwire_intel
snd_sof_intel_hda      24576  1 snd_sof_intel_hda_common
intel_rapl_msr         20480  0
snd_sof_pci            24576  2 snd_sof_intel_hda_common,snd_sof_pci_intel_cnl
intel_rapl_common      40960  1 intel_rapl_msr
snd_sof_xtensa_dsp     16384  1 snd_sof_intel_hda_common
intel_tcc_cooling      16384  0
x86_pkg_temp_thermal    20480  0
snd_sof               319488  3 snd_sof_pci,snd_sof_intel_hda_common,snd_sof_intel_hda
intel_powerclamp       24576  0
coretemp               24576  0
snd_sof_utils          20480  1 snd_sof
snd_soc_hdac_hda       24576  1 snd_sof_intel_hda_common
nls_iso8859_1          16384  1
snd_hda_ext_core       36864  3 snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda
snd_soc_acpi_intel_match    77824  2 snd_sof_intel_hda_common,snd_sof_pci_intel_cnl
kvm_intel             512000  0
snd_soc_acpi           16384  2 snd_soc_acpi_intel_match,snd_sof_intel_hda_common
soundwire_bus         110592  3 soundwire_intel,soundwire_generic_allocation,soundwire_cadence
iommu_v2               24576  1 amdgpu
snd_soc_core          413696  4 soundwire_intel,snd_sof,snd_sof_intel_hda_common,snd_soc_hdac_hda
drm_buddy              20480  1 amdgpu
kvm                  1331200  1 kvm_intel
snd_hda_codec_cs8409    36864  1
gpu_sched              61440  1 amdgpu
snd_compress           28672  1 snd_soc_core
drm_ttm_helper         16384  1 amdgpu
irqbypass              16384  1 kvm
snd_hda_codec_generic   118784  1 snd_hda_codec_cs8409
ttm                   102400  2 amdgpu,drm_ttm_helper
ac97_bus               16384  1 snd_soc_core
crct10dif_pclmul       16384  1
mei_hdcp               28672  0
snd_hda_codec_hdmi     90112  1
mei_pxp                20480  0
ledtrig_audio          16384  1 snd_hda_codec_generic
polyval_clmulni        16384  0
snd_pcm_dmaengine      20480  1 snd_soc_core
drm_display_helper    208896  1 amdgpu
polyval_generic        16384  1 polyval_clmulni
ghash_clmulni_intel    16384  0
uvcvideo              135168  0
cec                    94208  1 drm_display_helper
sha512_ssse3           53248  0
aesni_intel           397312  4
[... 85 lines cut ...]
hid                   180224  3 usbhid,hid_apple,hid_generic
uas                    28672  2
usb_storage            86016  1 uas
nvme                   57344  1
crc32_pclmul           16384  0
thunderbolt           454656  0
spi_intel_pci          16384  0
i2c_i801               36864  0
spi_intel              32768  1 spi_intel_pci
nvme_core             196608  2 nvme
intel_lpss_pci         28672  2
i2c_smbus              20480  1 i2c_i801
tg3                   217088  0
nvme_common            28672  1 nvme_core
sdhci_pci              77824  0
intel_lpss             16384  1 intel_lpss_pci
cqhci                  40960  1 sdhci_pci
ahci                   49152  0
idma64                 20480  0
sdhci                  90112  1 sdhci_pci
xhci_pci               24576  0
libahci                57344  1 ahci
xhci_pci_renesas       20480  1 xhci_pci
video                  73728  1 amdgpu
wmi                    40960  1 video

davidjo, 2023-05-18:
maybe kern.log?
or dmesg?
This is Ubuntu right?

The lsmod shows the module is loaded  - but the question is which one - there is a system version.
(command would be lsmod | grep 8409 )

There should be one under kernel/sound/pci/hda under  /lib/modules/version.

Now one way is to move that version out of /lib/modules/version to somewhere else, run a sudo depmod -a and reboot and check again - maybe theyve messed with updates - it should be that if the same module name is in updates it should override anything else.
Clearly somethings different about your system with no syslog.

lui75, 2023-05-18:
Yes it is ubuntu. This is the result of the command you asked me to check:
luigi@luigi-iMac:~$  lsmod | grep 8409
snd_hda_codec_cs8409    36864  1
snd_hda_codec_generic   118784  1 snd_hda_codec_cs8409
snd_hda_codec         204800  6 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_codec_cs8409,snd_hda_intel,snd_soc_hdac_hda,snd_sof_intel_hda
snd_hda_core          139264  9 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_codec_cs8409,snd_hda_intel,snd_hda_ext_core,snd_hda_codec,snd_sof_intel_hda_common,snd_soc_hdac_hda,snd_sof_intel_hda
snd                   135168  19 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_codec_cs8409,snd_hda_intel,snd_hda_codec,snd_sof,snd_timer,snd_compress,snd_soc_core,apple_bce,snd_pcm,snd_rawmidi

Inside /lib/modules/version kernel/sound/pci/hda there is not the 8409 module, that is only inside update.

lui75, 2023-05-18:
Dmesg under /var/log

[    5.727596] snd_hda_codec_cs8409: loading out-of-tree module taints kernel.
[    5.727628] snd_hda_codec_cs8409: module verification failed: signature and/or required key missing - tainting kernel
[    5.732286] snd_hda_codec_cs8409 hdaudioC0D0: autoconfig for CS8409: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
[    5.732292] snd_hda_codec_cs8409 hdaudioC0D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
[    5.732295] snd_hda_codec_cs8409 hdaudioC0D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
[    5.732297] snd_hda_codec_cs8409 hdaudioC0D0:    mono: mono_out=0x0
[    5.732299] snd_hda_codec_cs8409 hdaudioC0D0:    inputs:
[    5.732301] snd_hda_codec_cs8409 hdaudioC0D0:      Internal Mic=0x45
[    5.732303] snd_hda_codec_cs8409 hdaudioC0D0:      Mic=0x3c
[    5.749254] AMD-Vi: AMD IOMMUv2 functionality not available on this system - This is not a bug.
[    5.940633] intel_tcc_cooling: Programmable TCC Offset detected

davidjo, 2023-05-18:
OK - so thats not my module - no Primary patch - on the other hand it must be loading the updates version as it says out-of-tree.

By the way note that the module file name is snd-hda-codec-cs8409.ko but the module name under lsmod is snd_hda_codec_cs8409.

Check again  that there isnt another module somewhere - do
find . -name snd-hda-codec-cs8409.ko -ls
under /lib/modules/version

The other option is that somehow the original files are being used in build/hda-x.y.z  instead of the patched files.

try doing
strings -a snd-hda-codec-cs8409.ko | grep Primary in the build directory build/hda-x.y.z.

lui75, 2023-05-18:
This is the reult of the first command
find . -name snd-hda-codec-cs8409.ko -ls
  5521237     92 -rw-r--r--   1 root     root        92969 mag 17 22:06 ./kernel/sound/pci/hda/snd-hda-codec-cs8409.ko
  4377578    760 -rw-r--r--   1 root     root       774880 mag 18 18:06 ./updates/snd-hda-codec-cs8409.ko

No result for the string command

davidjo, 2023-05-18:
So there is a base kernel module.

However it looks as though theres something wrong with the build - you should get 2 lines with Primary from the strings cmd.

Sorry - my fault - I missed a patch
you need to do
patch -b -p2 <../../patch_patch_cs8409.c.diff

then make clean; make; and sudo make install

lui75, 2023-05-18:
Perfect! Now I got sound! Thank you so much!

davidjo, 2023-05-18:
Great - we finally got there!!

lui75, 2023-05-18:
I got an answer from Aditya Greg 
https://github.com/t2linux/T2-Ubuntu/issues/88#issuecomment-1553479103

lui75, 2023-05-20:
The solution proposed by Aditya Greg https://github.com/t2linux/T2-Ubuntu/issues/88#issuecomment-1553479103
solved the issue.

closed 2023-05-20 by lui75

davidjo, 2023-05-20:
Curious - that means the T2 kernel is a plain mainline kernel (isubuntu=0 just downloads kernel source from the main linux kernel    distribution site).

lui75, 2023-05-20:
It seems like it is, but it's built with all the patches for T2 Mac. And they work for my iMac model that was the last one before the T2 chip was introduced.