davidjo/snd_hda_macbookpro #92: SteamOS kernel 6.1.21-valve1-3-neptune-61

gronesy, 2023-06-04:
First of all, I'd like to thank davodjo and the rest of the community for their hard work and patience! I am getting back into linux after a ten year break because of the recent abandonment by Apple for their bootcamp gamers.  I am not now, or have I ever been, a developer. Enough about me.

Thank you again for this fix which works great for my Ubuntu and derivatives! I have a 2017 iMac 5k 18,3 machine with a Radeon 580X card.  I recently installed SteamOS from the HoloISO installer. Other than some video oddities the main problem I've had with this and nearly EVERY linux distro I've installed is.....no sound.

Recently, I used your instructions for the Arch linux install for your fix. It finished without errors but here's how the sound works. 

Sometimes, when I'm just doing a speaker check the audio chirps and then works fine. Until I start another source, that is. Then, it may just chirp and play or just squeal. Other times, the audio will just be fuzzy. It's got me scratching my head.

I did try the Arch directions on a Garuda linux install and it didn't work at all and may have errored out as it did for the other Arch linux poster.

The funny thing is if I blacklist the cs8409 card just for the hell of it the system thinks everything's fine but no sound. I mean internet radio continues to show it's playing instead of stopping like they usually do when something's not working.

These are all bare metal external SSD installs if that matters.

I am dying to know, once and for all what driver my system needs so I can install it on ANY linux I work with but I'm afraid that's a pipe dream. Not pipewire, mind you-LOL!

Let me know what output you need and I'll try to figure out how to get it and copy and paste it!

Thank you!

Jeff

davidjo, 2023-06-05:
Unfortunately  I think these are sound issues I havent got a good solution to.
Things have to be done slowly - ensure any previous stream has stopped for a good few seconds before starting a new one (and if you have eg system sounds on dont know how to manage that).
The Apple way seems to involve a lot of HDA verbs (100s-1000s) to start/stop streams - these take a good few milliseconds to 1/2 seconds (if you have headphones working try playing a sound, plugging in the headphones and see how long before the speakers switch off - for my MacbookPro it is a good 1/2 second - on previous macs was near instantaneous - you can do this on OSX).
As of yet havent figured a good way of blocking cause dont really have enough knowledge of the fine details of the linux kernel sound system and the user audio side interface - on OSX Im guessing a lot of this sound switching is handled by the higher level sound driver CoreAudio.
Curiously the Windows driver (as seen from files on the Bootcamp image in particular the cs4208_{number}.inf file - where number is eg 24, 27, 39 - it seems to turn sound off with much smaller numbers of verbs.

gronesy, 2023-06-05:
Thank you, sir, for your incredibly quick response! 

The sound system parameters you speak of are WAY above my level of understanding. I'll try the headphone/speaker cut-off experiment to see what happens. It seems like it doesn't matter whether I switch quickly or slowly. I'm still getting either the chirp before working, the squeal and NOT working or possibly both before getting fuzzy audio. 

I was working with a person from Parallels who told me that the program gives the guest linux system a virtual chip to work with which I thought was genius but not much help for use bare metal people. EVERY VM system I've tried says the driver is AC'97 (I know there are as many of these as ice cream flavors-too!) but loading ANY AC'97 module does not work for any linux I've tried.

The thing that also has me scratching my head is after I've identified that sysdefault is the card being used and run aplay -D sysdefault someaudiofile.wav it plays flawlessly. No chirps, squeals or fuzziness. This gives me hope that there IS a way to make this work!

Finally, I wanted to mention that I've used egorenar's Manjaro fix on a Manjaro system and it works perfectly! But it works ONLY on Manjaro and not other Arch-based systems.

If anybody has a different experience with that fix, please let me know. I'm mainly interested in running Regata (opensuse) and Garuda (arch) for the gaming.

Thank you again, davidjo, for your excellent Ubuntu "fix"!

Jeff

davidjo, 2023-06-05:
ah - so I  use aplay for testing because you can explicitly tell it to use the direct machine chip - I suspect the aplay command you are using is directly driving the actual machine chip (apparently linux VMs map the audio system to a virtual AC97 chip - how that maps to the bare metal chip I dont know - but that may be causing the issues).
One thing you might try is turning off the system sounds (like terminal beeps, window click beeps etc) - you can do that under Ubuntu using the Settings gui app - both on the bare machine OS and the VM - because its not clear to me how such sounds would be multiplexed if occurring at the same time.
Just seen that AC97 (an older format) apparently specifies either 16 or 20 bit audio either at 96 khz or 48khz - however the Apple 8409 chip system uses a specific format of 24 bit audio at 48 khz per channel pair - any other format needs translating to this format before sending to the 8409 chip - from  my experience mismatches in these bit sizes/rates does give rise to bad audio - squeals, chirps etc.
Such format translation needs to handled by eg the pulse user audio system or ALSA user modules (unfortunately pipewire seems to be the new linux user audio handler so you may have to get used to it - even under Ubuntu 22.04 browser audio is mapped through pipewire to pulse - later Ubuntus switch to pipewire fully).

PS the Egorenar patch is now getting a bit old - one of the big issues Ive had is that apparently the linux kernel code for audio has had a lot of breaking updates during kernel version 5.
One big one is that the linux kernel source added a specific 8409 chip system for some Dell laptops (which I think is post the egorenar patch) - this completely messed up my original 8409 chip handling but my current version handles this.

davidjo, 2023-06-06:
Sorry - loosing track - the egorenar DOES have the separate linux 8409 kernel module update - but all the egorenar fixups (additional iMac models using the TAS5764L amp chip) have now been implemented in the main repo.

Both my repo and egorenar do have to be redone on a kernel update simply because  the updated kernel module needs rebuilding for each kernel ( could use dkms to auto rebuild but as there have been all the other issues with incompatible mainline kernel changes not a main priority).

Actually so far from ubuntu kernel verrsions 5.15.47 to 6.2.20 doesnt seem to have been breaking update.

We really should try and figure out why the current repo seems to failing for you.

gronesy, 2023-06-06:
I have noticed that the Ubuntu does NOT need repatched each time the kernel updates. Great work!

I tried to run the Arch patch instructions again yesterday on Garuda. The weird part was although I copied and pasted your kernel header code it started out with a list of over 50 404s looking for it? Oddly, I couldn't see where it EVER fetched them but the process did finish without errors but "mentioned" it couldn't find something from the headers that never downloaded?

When I checked the sound (with system sounds muted and unmuted) I could see the front left, front right audio was playing but no sound. I know pipewire was originally a front end for alsa but now you say it's the sound system?

I've checked the audio devices gui, alsamixer and I think pavucontrol at some point. I can't see where sound is muted or turned all the way down..

This is great working with somebody who's not hostile towards mac people AND really knows their stuff! Thanks again!

Let me know what output you need and I'll get it for you.

Jeff

davidjo, 2023-06-07:
I dont use Arch so have to rely on other users for info.
At the moment Arch attempts to download a mainline kernel using wget - however still not sure of kernel version mapping so try to use major.minor.revision - which looks like it fails hence the 404s - but then it tries just major.minor and likely worked.
I need to see the exact error message to suggest whats happening.
Cant really figure if it loaded the module or not without seeing a kernel log - look for messages with snd_hda (snd_hda_intel and snd_hda_codec) in your kernel log.

My current understanding - which is just piecing things together as havent really found a good overview - is that ALSA is an api to talk to the kernel sound drivers - like my module - from the user side.
It also has the ability to do some user side features by setting up a user file eg .asoundrc which is placed in your home directory - some attempts have been discussed in older issues for the Apple audio system eg issue #23 .
pulse audio - now replaced by pipewire - is as you say a user side front end for dealing with moving audio from user requests to the kernel sound driver - some features of which could also be done by definitions in an .asoundrc file.
One of the big issues is converting from an arbitrary sound file the user wants to play ( with a bit rate *eg 44.1 kHz, 48 kHz etc, resolution eg 16 bit, 24 bits per channel , number of channels, 2, 4, 5.1, 7.1 etc) to the drivers specific format (in my modules case 24 bits and 44.1 kHz and 4 channels) as the sound hardware may not be able to handle all sound formats directly).
pulse/pipewire also handle up/down mixing of different sounds and channels eg if you have a 7.1 sound source it needs downmixing to play on a stereo only (2 channels) hardware.
(I think the issue was ALSA didnt really do this in a user friendly way hence pulse was created to do these conversions with much less user intervention - but pulse/pipewire still end up sending sound to raw ALSA kernel sound devices - which each hardware sound device is mapped to).

gronesy, 2023-06-07:
I will re-run your patch on the Garuda system and copy the error/output file for you to peruse. 

I'll have to say I just heard a loud booming overhead which was a great deal of your explanation going over my head-LOL!

What's the easiest way to access the kernel log? I'm reasonably handy with vi/vim but not sure where I'm looking.

I have an audio background so the down mixing is not completely lost on me.

I'll take a look at the issue 23 and see if it rings any bells.

Other than that, thank you again for your help and I'll go through the last post and ask you any questions I have as I get to them.

For the Regata/OpenSuse patch. Do you know the kernel header fetching code? Or will your patch not work on those systems and THAT'S why it's not included?

davidjo, 2023-06-08:
Kernel logs are usually under /var/log - there are some different file naming conventions - the original name was syslog but ubuntu also has kern.log (ie /var/log/syslog).

As I said anything Ive stated is what Ive pieced together from googling and some intuition - I wouldnt say I understand the linux audio system.

As my fixup has to build a kernel module you need to install all the components in order to build kernel modules (compilers etc)
plus kernel headers.
Unfortunately each linux distribution seems to have its own way of packaging these requirements (and where thay are placed in the file system) - so you need to figure out where Regata/OpenSuse has packaged the kernel headers and how to install - cant help you with that as never used OpenSuse - but assume googling for eg kernel headers opensuse should get you the info needed.
So yes my repo wont work out of the box on a random linux distribution  - all of the info about kernel headers that is printed by the install script has been added as other users have got the repo working on these other distributions.

gronesy, 2023-06-08:
I was able to finally find some info on getting kernel headers in Regata/OpenSUSE by running:

sudo zypper install kernel-headers

And it returned this:

`jeff@RegataOSND:~/snd_hda_macbookpro> sudo zypper install kernel-headers
Loading repository data...
Reading installed packages...
kernel-headers' not found in package names. Trying capabilities.
'linux-glibc-devel' providing 'kernel-headers' is already installed.
Resolving package dependencies...
Nothing to do.
jeff@RegataOSND:~/snd_hda_macbookpro> 

Apparently, it sees them as already installed.

Then, I ran your install patch and got this:

jeff@RegataOSND:~/snd_hda_macbookpro> sudo ./install.cirrus.driver.sh
[sudo] password for root: 
linux kernel headers not found:
Debian (eg Ubuntu): /usr/src/linux-headers-6.3.5-lp154.6-default
Fedora: /usr/src/kernels/6.3.5-lp154.6-default
Arch: /usr/lib/modules/6.3.5-lp154.6-default
Void: /usr/src/kernel-headers-6.3.5-lp154.6-default
assuming the linux kernel headers package is not installed
please install the appropriate linux kernel headers package:
Debian/Ubuntu: sudo apt install linux-headers-6.3.5-lp154.6-default
Fedora: sudo dnf install kernel-headers
Arch (also Manjaro): Linux: sudo pacman -S linux-headers
Void Linux: xbps-install -S linux-headers
jeff@RegataOSND:~/snd_hda_macbookpro>

Which tells me the install patch is not looking for the kernel-headers where they reside?

Am I even warm?

davidjo, 2023-06-08:
Yes - we will need to tell the install script where the linux-headers are installed - it appears they are installed from the linux-glibc-devel package but we need the path of the include files - possibly /usr/src/linux.
Another option would be to comment the exit 1 after echo " Void Linux: xpbs...." - the header check is just a check - so if the headers exist then the script might work - this will be the isubuntu=0 path/

gronesy, 2023-06-08:
The path of the include files is /usr/src/linux/include

I understand you want me to comment (#) the exit 1 after the "Void Linux:xpbs..." entry but are you saying something needs to be changed in the isubuntu=0 code?

To give you output (I promise I checked the Github docs) do I need to create a file and upload it so it doesn't strike through all of the text?

osalbahr, 2023-06-08:
You can surround the text with three back ticks
```
Like this
```

davidjo, 2023-06-08:
Sorry - I just meant that as this is not an ubuntu system isubuntu will be 0 so the actions performed will be for isubuntu=0 - which is to download a mainline kernel source as I need files from the kernel source to patch and build the module (in addition to the headers) - as of yet I dont know if this is the right thing to do for Suse.

I think just do what works for you - I dont always get the git comment special format syntax right.

gronesy, 2023-06-08:
Thank you, @osalbahr for the help! If the output is lengthy, how do I create one of those scrollable entries to the post? Is that uploading a text file of sorts?

@davidjo I commented out the exit 1 you mentioned and it still gave the linux-headers missing warning, mentioned there would probably be build problems but the script DID finish this time and didn't stop at "no kernel headers" with instructions.

I rebooted and now I have chirping induction audio and squealing/ringing but no real audio.

Here's the output:

```jeff@RegataOSND:~/snd_hda_macbookpro> sudo ./install.cirrus.driver.sh
[sudo] password for root: 
linux kernel headers not found:
Debian (eg Ubuntu): /usr/src/linux-headers-6.3.5-lp154.6-default
Fedora: /usr/src/kernels/6.3.5-lp154.6-default
Arch: /usr/lib/modules/6.3.5-lp154.6-default
Void: /usr/src/kernel-headers-6.3.5-lp154.6-default
assuming the linux kernel headers package is not installed
please install the appropriate linux kernel headers package:
Debian/Ubuntu: sudo apt install linux-headers-6.3.5-lp154.6-default
Fedora: sudo dnf install kernel-headers
Arch (also Manjaro): Linux: sudo pacman -S linux-headers
Void Linux: xbps-install -S linux-headers
--2023-06-08 14:02:14--  https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.3.5.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 2a04:4e42:30::432, 146.75.45.176
Connecting to cdn.kernel.org (cdn.kernel.org)|2a04:4e42:30::432|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 136926988 (131M) [application/x-xz]
Saving to: ‘build/linux-6.3.5.tar.xz’

linux-6.3.5.tar.xz        100%[=====================================>] 130.58M   101MB/s    in 1.3s    

2023-06-08 14:02:17 (101 MB/s) - ‘build/linux-6.3.5.tar.xz’ saved [136926988/136926988]

linux-6.3.5/sound/pci/hda/
linux-6.3.5/sound/pci/hda/Kconfig
linux-6.3.5/sound/pci/hda/Makefile
linux-6.3.5/sound/pci/hda/ca0132_regs.h
linux-6.3.5/sound/pci/hda/cs35l41_hda.c
linux-6.3.5/sound/pci/hda/cs35l41_hda.h
linux-6.3.5/sound/pci/hda/cs35l41_hda_i2c.c
linux-6.3.5/sound/pci/hda/cs35l41_hda_spi.c
linux-6.3.5/sound/pci/hda/hda_auto_parser.c
linux-6.3.5/sound/pci/hda/hda_auto_parser.h
linux-6.3.5/sound/pci/hda/hda_beep.c
linux-6.3.5/sound/pci/hda/hda_beep.h
linux-6.3.5/sound/pci/hda/hda_bind.c
linux-6.3.5/sound/pci/hda/hda_codec.c
linux-6.3.5/sound/pci/hda/hda_component.h
linux-6.3.5/sound/pci/hda/hda_controller.c
linux-6.3.5/sound/pci/hda/hda_controller.h
linux-6.3.5/sound/pci/hda/hda_controller_trace.h
linux-6.3.5/sound/pci/hda/hda_cs_dsp_ctl.c
linux-6.3.5/sound/pci/hda/hda_cs_dsp_ctl.h
linux-6.3.5/sound/pci/hda/hda_eld.c
linux-6.3.5/sound/pci/hda/hda_generic.c
linux-6.3.5/sound/pci/hda/hda_generic.h
linux-6.3.5/sound/pci/hda/hda_hwdep.c
linux-6.3.5/sound/pci/hda/hda_intel.c
linux-6.3.5/sound/pci/hda/hda_intel.h
linux-6.3.5/sound/pci/hda/hda_intel_trace.h
linux-6.3.5/sound/pci/hda/hda_jack.c
linux-6.3.5/sound/pci/hda/hda_jack.h
linux-6.3.5/sound/pci/hda/hda_local.h
linux-6.3.5/sound/pci/hda/hda_proc.c
linux-6.3.5/sound/pci/hda/hda_sysfs.c
linux-6.3.5/sound/pci/hda/hda_tegra.c
linux-6.3.5/sound/pci/hda/hp_x360_helper.c
linux-6.3.5/sound/pci/hda/ideapad_s740_helper.c
linux-6.3.5/sound/pci/hda/patch_analog.c
linux-6.3.5/sound/pci/hda/patch_ca0110.c
linux-6.3.5/sound/pci/hda/patch_ca0132.c
[... 25 lines cut ...]
In file included from /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/patch_cs8409.c:1488:
/home/jeff/snd_hda_macbookpro/build/hda-6.3.5/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/jeff/snd_hda_macbookpro/build/hda-6.3.5/patch_cirrus_apple.h:1193:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1193 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  CC [M]  /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/patch_cs8409-tables.o
  LD [M]  /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/snd-hda-codec-cs8409.o
  MODPOST /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/Module.symvers
  CC [M]  /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/snd-hda-codec-cs8409.ko
make[1]: Leaving directory '/usr/src/linux-6.3.5-lp154.6-obj/x86_64/default'
#cp snd-hda-codec-cs8409.ko /lib/modules/6.3.5-lp154.6-default/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/6.3.5-lp154.6-default/build M=/home/jeff/snd_hda_macbookpro/build/hda-6.3.5 CONFIG_MODULE_SIG_ALL=n modules_install
make[1]: Entering directory '/usr/src/linux-6.3.5-lp154.6-obj/x86_64/default'
  INSTALL /lib/modules/6.3.5-lp154.6-default/updates/snd-hda-codec-cs8409.ko
  DEPMOD  /lib/modules/6.3.5-lp154.6-default
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-6.3.5-lp154.6-obj/x86_64/default'
depmod -a

contents of /lib/modules/6.3.5-lp154.6-default/updates
total 1872
-rw-r--r-- 1 root root  449488 Jun  8 14:02 snd-hda-codec-cs8409.ko
-rw-r--r-- 1 root root 1464036 Jun  3 06:07 wl.ko.xz
jeff@RegataOSND:~/snd_hda_macbookpro> ```

gronesy, 2023-06-09:
Here is the output from the Garuda kernel header fetch:

```sudo pacman -S wget make gcc linux-headers
[sudo] password for jeff:
warning: wget-1.21.4-1 is up to date -- reinstalling
warning: make-4.4.1-2 is up to date -- reinstalling
warning: gcc-13.1.1-1 is up to date -- reinstalling
warning: linux-headers-6.3.5.arch1-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (4) gcc-13.1.1-1  linux-headers-6.3.5.arch1-1  make-4.4.1-2  wget-1.21.4-1

Total Installed Size:  345.47 MiB
Net Upgrade Size:        0.00 MiB

:: Proceed with installation? [Y/n] Y
(4/4) checking keys in keyring                                     [------------------------------------] 100%
(4/4) checking package integrity                                   [------------------------------------] 100%
(4/4) loading package files                                        [------------------------------------] 100%
(4/4) checking for file conflicts                                  [------------------------------------] 100%
(4/4) checking available disk space                                [------------------------------------] 100%
:: Running pre-transaction hooks...
(1/2) Performing snapper pre snapshots for the following configurations...
==> root: 53
(2/2) Remove upgraded DKMS modules
:: Processing package changes...
(1/4) reinstalling wget                                            [------------------------------------] 100%
(2/4) reinstalling make                                            [------------------------------------] 100%
(3/4) reinstalling gcc                                             [------------------------------------] 100%
(4/4) reinstalling linux-headers                                   [------------------------------------] 100%
:: Running post-transaction hooks...
( 1/10) Arming ConditionNeedsUpdate...
( 2/10) Updating module dependencies...
( 3/10) Install DKMS modules
==> ERROR: Missing 6.3.5-arch1-1 kernel modules tree for module vboxhost/7.0.8_OSE.
( 4/10) Refreshing PackageKit...
( 5/10) Foreign/AUR package notification
ananicy-rules 1.r129.8ac5d97-1
libsidplay 1.36.59-10
ltris 1.2.6-1
( 6/10) Orphaned package notification...
cython 0.29.35-1
libsidplay 1.36.59-10
lua53 5.3.6-1
python-future 0.18.3-1
sysfsutils 2.1.1-1
( 7/10) Checking for .pacnew and .pacsave files...
.pac* files found:
/etc/pamac.conf.pacnew
/etc/pacman.conf.pacnew
/etc/pacman.d/mirrorlist.pacnew
/etc/plymouth/plymouthd.conf.pacnew
/etc/retroarch.cfg.pacnew
Please check and merge
( 8/10) Updating the info directory file...
( 9/10) Performing snapper post snapshots for the following configurations...
==> root: 54
(10/10) Syncing all file systems...```

Then I re-ran the install.cirrus script and this is the output:

```sudo ./install.cirrus.driver.sh
--2023-06-08 18:15:51--  https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.3.5.tar.xz
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving cdn.kernel.org (cdn.kernel.org)... 2a04:4e42:30::432, 151.101.205.176
Connecting to cdn.kernel.org (cdn.kernel.org)|2a04:4e42:30::432|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 136926988 (131M) [application/x-xz]
Saving to: ‘build/linux-6.3.5.tar.xz’
[... 68 lines cut ...]
/home/jeff/snd_hda_macbookpro/build/hda-6.3.5/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/jeff/snd_hda_macbookpro/build/hda-6.3.5/patch_cirrus_apple.h:1193:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
1193 |         hinfo = spec->gen.stream_analog_playback;
|               ^
CC [M]  /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/patch_cs8409-tables.o
LD [M]  /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/snd-hda-codec-cs8409.o
MODPOST /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/Module.symvers
CC [M]  /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/snd-hda-codec-cs8409.mod.o
LD [M]  /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/snd-hda-codec-cs8409.ko
BTF [M] /home/jeff/snd_hda_macbookpro/build/hda-6.3.5/snd-hda-codec-cs8409.ko
#cp snd-hda-codec-cs8409.ko /lib/modules/6.3.5-zen2-1-zen/updates
make INSTALL_MOD_DIR=updates -C /lib/modules/6.3.5-zen2-1-zen/build M=/home/jeff/snd_hda_macbookpro/build/hda-6.3.5 CONFIG_MODULE_SIG_ALL=n modules_install
INSTALL /lib/modules/6.3.5-zen2-1-zen/updates/snd-hda-codec-cs8409.ko
ZSTD    /lib/modules/6.3.5-zen2-1-zen/updates/snd-hda-codec-cs8409.ko.zst
DEPMOD  /lib/modules/6.3.5-zen2-1-zen
depmod -a

contents of /lib/modules/6.3.5-zen2-1-zen/updates
total 548
drwxr-xr-x 1 root root     96 Jun  5 17:01 dkms
-rw-r--r-- 1 root root 560283 Jun  8 18:16 snd-hda-codec-cs8409.ko.zst

╭─jeff@GarudaMJ in repo: snd_hda_macbookpro```

I can't find a syslog in /var/log or a kernel directory there.

davidjo, 2023-06-09:
You will still get the warning message - removing exit 1 just means the script doesnt stop at that point without building anything.
(Until the checking in the script is updated to handle Suse type distributions removing exit 1 would always be needed).

Both your log files are succesful builds, the mainline kernel source has been downloaded, the patches applied successfully (eg Hunk ```#```2 succeeded - and all hunks succeeded, the module build worked.
The last few lines from contents of /lib/modules/.... /updates show you have the module (snd-hda-codec-cs8409.ko or a compressed version snd-hda-codec-cs8409.ko.zst)  in the kernel modules updates directory - which is what should happen.

Apparently Suses version of syslog is the messages file in /var/log ie /var/log/messages.

davidjo, 2023-06-09:
Sorry - I should have mentioned this - both bluetooth and usb headphones/speakers etc work fine - and I do suggest it to users if they have issues or just want something thats going work without issues.
My driver is purely for the internal speakers, mike and standard audio jack headphone/headset path - for a laptop this is more significant as then you dont have to carry around additional speakers.
In your image above you are selecting a different audio device - but cant immediately see if devices from my module exist.
I was going to suggest you try aplay -l and aplay -L with my module and show the results - these just list the devices - aplay -l just does the pure hardware devices, aplay -L also includes software devices (for mixing etc).

gronesy, 2023-06-09:
No worries. The last time I was doing linux hardcore it was with LinuxPPC and YellowDog, if that tells you anything. I really just wanted my internal mike and audio jack speakers to work! Like I mentioned, I'd read about people doing USB but I must have had a bad sample because theirs either didn't work or the volume was pegged and couldn't be adjusted.

I did a search in the Regata OS /var/log/messages file and could NOT find any reference to snd_anything. Here is a copy of the /var/log directory. Could it be in another file?

```RegataOSND:/var/log # ls
acpid                 boot.omsg      messages              wpa_supplicant.log
alternatives.log      btmp           messages-20230530.xz  wtmp
apparmor              cups           messages-20230606.xz  Xorg.0.log
audit                 firewall       NetworkManager        Xorg.0.log.old
boot.log              firewalld      openvswitch           YaST2
boot.log-20230530.xz  hp             pbl.log               zypp
boot.log-20230601.xz  krb5           private               zypper.log
boot.log-20230605.xz  lastlog        regataos-logs         zypper.log-20230529.xz
boot.log-20230606.xz  localmessages  samba                 zypper.log-20230605.xz
boot.log-20230607.xz  mail           snapper.log           zypper.log-20230609.xz
boot.log-20230608.xz  mail.err       speech-dispatcher
boot.log-20230609.xz  mail.info      tallylog
boot.msg              mail.warn      warn```

Here's the aplay -l:

```RegataOSND:/var/log # aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: PCH [HDA Intel PCH], device 0: CS8409/CS42L83 Analog [CS8409/CS42L83 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 7: HDMI 1 [HDMI 1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 8: HDMI 2 [HDMI 2]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 9: HDMI 3 [HDMI 3]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 10: HDMI 4 [HDMI 4]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 11: HDMI 5 [HDMI 5]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 12: HDMI 6 [HDMI 6]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Everyman [FREETALK Everyman], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0```

And aplay -L:

```RegataOSND:/var/log # aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
speexrate
    Rate Converter Plugin Using Speex Resampler
pipewire
    PipeWire Sound Server
pulse
    PulseAudio Sound Server
upmix
    Plugin for channel upmix (4,6,8)
default
    Default ALSA Output (currently PulseAudio Sound Server)
sysdefault:CARD=PCH
    HDA Intel PCH, CS8409/CS42L83 Analog
    Default Audio Device
front:CARD=PCH,DEV=0
    HDA Intel PCH, CS8409/CS42L83 Analog
    Front output / input
surround21:CARD=PCH,DEV=0
    HDA Intel PCH, CS8409/CS42L83 Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=PCH,DEV=0
    HDA Intel PCH, CS8409/CS42L83 Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=PCH,DEV=0
    HDA Intel PCH, CS8409/CS42L83 Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=PCH,DEV=0
    HDA Intel PCH, CS8409/CS42L83 Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=PCH,DEV=0
    HDA Intel PCH, CS8409/CS42L83 Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=PCH,DEV=0
    HDA Intel PCH, CS8409/CS42L83 Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
hdmi:CARD=HDMI,DEV=0
    HDA ATI HDMI, HDMI 0
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=1
    HDA ATI HDMI, HDMI 1
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=2
    HDA ATI HDMI, HDMI 2
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=3
    HDA ATI HDMI, HDMI 3
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=4
    HDA ATI HDMI, HDMI 4
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=5
    HDA ATI HDMI, HDMI 5
    HDMI Audio Output
hdmi:CARD=HDMI,DEV=6
    HDA ATI HDMI, HDMI 6
    HDMI Audio Output
sysdefault:CARD=Everyman
    FREETALK Everyman, USB Audio
    Default Audio Device
front:CARD=Everyman,DEV=0
    FREETALK Everyman, USB Audio
    Front output / input
surround21:CARD=Everyman,DEV=0
    FREETALK Everyman, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Everyman,DEV=0
    FREETALK Everyman, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Everyman,DEV=0
    FREETALK Everyman, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Everyman,DEV=0
    FREETALK Everyman, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Everyman,DEV=0
    FREETALK Everyman, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Everyman,DEV=0
    FREETALK Everyman, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Everyman,DEV=0
    FREETALK Everyman, USB Audio
    IEC958 (S/PDIF) Digital Audio Output```

But alas, I spoke a little too soon. It looks like one of my games (Ghost Recon: Breakpoint) CAN use the USB output but it cannot find the mic attached to the same headset even though the Regata sound setting clearly DOES see it.

Let me know if this output helps or just muddies the waters more-LOL!

davidjo, 2023-06-10:
Hey I remember YellowDog - used it myself for a while  - have always booted linux on my Apple laptops (but Apple keeps trying to make it difficult).

Unfortunately logging is configurable on linux (either /etc/syslog.conf or /etc/rsyslog.conf plus /etc/rsyslog.d) and have no idea what default Suse setup is - maybe try eg grep Primary * in /var/log and see if the message I mentioned above appears - it should because the aplay logs show my module is loaded.

From the aplay logs my module is definitely loaded - card 0 gives CS8409/CS42L83 Analog [CS8409/CS42L83 Analog] - which is a specific name my module sets.

Looking back at your audio settings it appears the headphones and mike are set to Built-in Audio ... - which is from my module - but it not sure what Analog Stereo Duplex is - for me its a simple Analog Stereo Output - which I see you also have as an option.
You should set the output for internal speakers to Built-in Audio plus Analog Stereo Output.

gronesy, 2023-06-10:
Here is the output from the grep Primary * that I think confirms your module IS loaded in this kernel:

```RegataOSND:/var/log # grep Primary *
grep: apparmor: Is a directory
grep: audit: Is a directory
grep: cups: Is a directory
grep: hp: Is a directory
grep: krb5: Is a directory
localmessages:May 26 16:43:49 localhost kscreen_backend_launcher[6648]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 26 16:43:49 localhost kscreen_backend_launcher[6648]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 26 15:45:44 RegataOSND kscreen_backend_launcher[2421]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 26 15:45:44 RegataOSND kscreen_backend_launcher[2421]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 26 16:57:29 RegataOSND kscreen_backend_launcher[3558]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 26 16:57:29 RegataOSND kscreen_backend_launcher[3558]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 29 15:05:48 RegataOSND kscreen_backend_launcher[2256]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 29 15:05:48 RegataOSND kscreen_backend_launcher[2256]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 30 13:43:02 RegataOSND kscreen_backend_launcher[1812]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 30 13:43:02 RegataOSND kscreen_backend_launcher[1812]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 30 14:18:39 RegataOSND kscreen_backend_launcher[2205]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 30 14:18:39 RegataOSND kscreen_backend_launcher[2205]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 30 14:52:04 RegataOSND kscreen_backend_launcher[1791]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 30 14:52:04 RegataOSND kscreen_backend_launcher[1791]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 30 15:18:25 RegataOSND kscreen_backend_launcher[1817]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 30 15:18:25 RegataOSND kscreen_backend_launcher[1817]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 30 15:24:14 RegataOSND kscreen_backend_launcher[1810]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:May 30 15:24:14 RegataOSND kscreen_backend_launcher[1810]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  1 15:18:29 RegataOSND kscreen_backend_launcher[2305]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  1 15:18:29 RegataOSND kscreen_backend_launcher[2305]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  1 16:06:09 RegataOSND kscreen_backend_launcher[1791]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  1 16:06:09 RegataOSND kscreen_backend_launcher[1791]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  1 16:19:53 RegataOSND kscreen_backend_launcher[2137]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  1 16:19:53 RegataOSND kscreen_backend_launcher[2137]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  1 16:46:24 RegataOSND kscreen_backend_launcher[1984]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  1 16:46:24 RegataOSND kscreen_backend_launcher[1984]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  1 17:29:33 RegataOSND kscreen_backend_launcher[2015]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  1 17:29:33 RegataOSND kscreen_backend_launcher[2015]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  5 16:10:27 RegataOSND kscreen_backend_launcher[2153]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  5 16:10:27 RegataOSND kscreen_backend_launcher[2153]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  6 10:53:17 RegataOSND kscreen_backend_launcher[2251]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  6 10:53:17 RegataOSND kscreen_backend_launcher[2251]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  6 11:31:35 RegataOSND kscreen_backend_launcher[2058]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  6 11:31:35 RegataOSND kscreen_backend_launcher[2058]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  7 17:39:39 RegataOSND kscreen_backend_launcher[2208]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  7 17:39:39 RegataOSND kscreen_backend_launcher[2208]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  7 18:06:34 RegataOSND kscreen_backend_launcher[2102]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  7 18:06:34 RegataOSND kscreen_backend_launcher[2102]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  8 09:38:24 RegataOSND kscreen_backend_launcher[2176]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  8 09:38:24 RegataOSND kscreen_backend_launcher[2176]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  8 13:25:18 RegataOSND kscreen_backend_launcher[1783]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  8 13:25:18 RegataOSND kscreen_backend_launcher[1783]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  8 14:10:03 RegataOSND kscreen_backend_launcher[1786]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  8 14:10:03 RegataOSND kscreen_backend_launcher[1786]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  8 16:08:05 RegataOSND kscreen_backend_launcher[1948]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  8 16:08:05 RegataOSND kscreen_backend_launcher[1948]: kscreen.xrandr: RRSetOutputPrimary 
localmessages:Jun  8 22:00:31 RegataOSND kscreen_backend_launcher[2185]: kscreen.xrandr: RRSetOutputPrimary 
[... cut ...]
As far as the Garuda (Arch) system is concerned, the same sound settings work well. Egorenar's patch does NOT work on that flavor of Arch for some reason. But as I said it DOES work perfectly on the Manjaro it was written for.

I'm a little bummed because the Breakpoint game looks and plays even better than Windows 11 which is a first. BUT it also does not see the USB headset/mic at all. As most of my play is co-op this is a BIG BUT! Still, it's a HUGE victory for me to finally surpass my W11 install for ANY gaming endeavour!

Do you have any suggestions for trying to get my audio jack speakers and internal mic to work before I just go USB on them?

davidjo, 2023-06-10:
Yes this confirms the module is loaded.

You mentioned that aplay -D sysdefault works - is this for this system?

At the moment I think its something to do with the user side (this includes the pulse audio daemon which I think is being run - well your aplay logs show the default device is the pulse audio daemon)  audio setup on Regata  - we need to try using aplay to pin down what works and what doesnt.

gronesy, 2023-06-11:
I've run the aplay -D sysdefault on Regata and as long as the USB headset is connected, even with them NOT selected in the audio gui, it plays through the headset. When I disconnect the headset, again, with it switched to internal mic and analog stereo output for the headphone choice (speakers unavailable) it gives and error that there is NO sysdefault?

I think I originally ran this on SteamOS as this is where this post started-LOL!

I'll run it now and report back later. It's time for my buddy and I to play. On Windows, unfortunately. But I DO believe we'll get there with your help!

Thanks again!

gronesy, 2023-06-11:
Ok, one last quick one before I play-LOL!

I disconnected the USB headset before booting into SteamOS (Arch), selected the headphones (speakers unavailable) and internal mic, both using Analog Stereo Duplex.

When I run aplay -D sysdefault  StereoTest32.wav here it plays the audio file through the speakers. I just don't know how to tell the system to use that.

By the way, both of the systems, Regata and SteamOS show your module loaded on aplay -l for me again. I know you already said that but I may have had the headset connected when I ran it last and just wanted to make sure it wasn't muddying the device waters.

davidjo, 2023-06-11:
OK so I think this does mean the issue is in the "user" side audio setup - by this I mean configuration files - apparently the primary alsa setup config files is under /usr/share/alsa eg alsa.conf - plus also /etc/alsa - the ubuntu package is libasound2-data - OpenSuse also uses a libasound2 package name apparently.
You can also setup a personal configuration in .asoundrc in your home directory.
There must be some difference between the alsa.conf on ubuntu and OpenSuse as alsa.conf on ubuntu defines an hw  pcm and plughw pcm explicity.
The one issue maybe is that somehow the user side on OpenSuse is not recognising my modules user side naming  - its curious it knows about the headphones - that should now be an auto switch when headphones are plugged in.
One of my major tests was playing audio on speakers (aplay using plughw), plugging in a headphone and unplugging - all while playing - which for me seemed to work smoothly (Although I did have issues with booting with the headphone plugged in - I think Im still not handling that - so better to boot without jack headphones then plug them in).

I may understand the Analog Stereo Duplex - I think this means you are using both a mike and speaker/headphone at the same time (not something I ever really tested).

You also can tell pulse audio to do things using pactl or pacmd - I havent used them for a while - you can set the "sinks" (outputs ie speakers) and "sources" (mikes) explicitly.

Personally I primarily game on linux (standard PC - never with windows - but not on mac)  using Steam which seems to work.
On the other hand I dont really use a headset.

Arch does have quite a useful ALSA wiki page.

Ill check what Im seeing under Ubuntu for the audio speaker naming.

gronesy, 2023-06-12:
I've just had another one of those foggy brain experiences-LOL!

I've understood about half of what you were just teaching me. Sorry.

In my research, I've come across the config files stuff but I decided that I wasn't going down that rabbit hole. The USB discovery has made me buy some new speakers for my system which I think will help a LOT!

Please understand how much I appreciate your help. I would NEVER have come to the realizations I have without your help. I'm not jumping ship on you either. 

Have you always used Apple hardware for your gaming on linux? Is that how you got into the excellent solution you created?

Do you know people who have had success with Hackintosh systems? I know they have their own set of issues but with an AMD GPU and standard Intel audio, sound and video should NOT be one of them.

I have no knowledge of pacmd or pacl but will check out the Arch ALSA page.

I finally got a linux system to outperform Windows late last week on Garuda. But, of course, the game I was playing GR: Breakpoint didn't recognize either the USB headset or my internal mike! Sheesh! But it looked and played beautifully.

I'll keep you posted on how this is going. I'm going to replace the audio jack speakers with my new Creative USBs today. Wish me luck!

davidjo, 2023-06-12:
No problem - as I said I recommend to use bluetooth/usb if you want a simple experience - my primary use was to get speakers on the laptop working so didnt have to carry around speakers.

Sorry - I dont really use mac hardware for gaming - so PC hardware just linux never windows - only those you can buy without windows.
Not really used non-laptop Macs as the Apple hardware never fitted what I wanted to do.

Not really looked at Hackintoshes - not sure where they get their OSX drivers from  (they could be derived from linux drivers - the hardware would be handled just the interaction with the OSX kernel is the issue) - but you used to be able to create Apple kernel modules.

Ive always done gaming thro the Steam program on linux - although have used Lutris - not sure what Garuda is doing - Proton is now very good at handling Windows games in particular  with the Vulkan driver interface there is much less overhead due to the function call mapping.

If you want to try more with my module see if you can get pactl - this lets you list the pulse audio sinks and sources - its packaged in pulseaudio-utils on Ubuntu.

davidjo, 2023-06-13:
Correct - well I was used to linux for work - all supercomputers were and are essentially linux now - the really big ones use a combo of CPUs and GPUs (mainly Nvidia) now.

Apple did attempt to make a "server" at one stage which I used in a cluster of 30 or so but they gave up quite quickly - by 2010 I think.

I use Mac laptops (yes I used PowerPC originally) as since OSX they had a unix/linux environment but could load software on them for more general tasks, presentations etc. so Ive used those for a long time - always with dual boot to linux - sound used to work fine until around 2017 when they created this 8409 combo system (My guess is to create a 24 bit audio path for professional audio - both inputs, outputs and even internal speaker amps are 24 bit paths). But Apple systems seem to be getting more and more proprietary - next issue is the M1/M2 chips although some progress via reverse engineering (Apple refuses to document these systems) is being made for booting linux - see Asahi linux.

davidjo, 2023-06-13:
For a few years now (since around 2010) Ive been using something called rEFInd (originally rEFIt) to multi boot via EFI - sounds like you might still be using non-EFI booting - it still boots linux through a grub config but that grub system is limited to the actual linux boot partition.
I also was having issues with pre-EFI grub messing things up.

gronesy, 2023-06-14:
All of the linux I've set up has used an EFI partition unless it just didn't do that but that's rare. 

I know about rEFInd too. Both Garuda and Regata use it and I love it! That way, I can install the messy grub2 linux first (on like half of the SSD drive) and then FIX the bootloader by installing Garuda or Regata. Cleaned it right up.

What I don't know is how to force install rEFInd during an install. The only ones I've experimented with I added them manually and it worked but it was a bit of a pain to do. Is there a tool that makes it easier?

I think the thing that really threw me was telling the installer EXACTLY where to put the bootloader and then having it put it wherever else it saw fit-LOL!

Finally, I'm not completely sure but it seems the systems may NOT be preserving the USB setup and then takes a guess at which options to choose. These are usually wrong. I've tried to remember to keep the USB speaker ON when I boot in too.

davidjo, 2023-06-14:
Ah - I think I remember what I did - for the same problem you had ie grub writing stuff in random places - for Ubuntu towards end of installation it asks something like do you want to install grub (bootloader?) system wide - answer no!! (I think this is where you were trying to tell it where to install the bootloader) - refind is clever enough to find the linux installs by itself.
However I installed refind from the OSX side - never done it from the linux side.
Once I installed refind I never really had to install it again - apart from refind updates - although every time OSX updated OS it messed up refind and I had to redo the install - but it finds the existing install and just does some name updates.
(The issue is after OSX updates it just boots direct into OSX every time - you probably could use the option key during boot but I like it to just boot into refind screen so "reinstalling" resets a boot parameter which means the refind screen is chosen).
The linux install will still create a grub boot setup on the linux root partition (ie under /boot) (well it did under ubuntu).
I can confirm it also works very well with multiple SSD installs ie multiple SSDs plugged in each with one or more Linux root partitions.
I also use the efibootmgr command on linux to change boot order.

Unfortunately for USB you may have to get into udev config files - Ive also had issues with inconsistencies - in particular with controllers (I use PS5 controller as theres a good controller kernel module from Sony under linux now).
Works OK with Bluetooth but if I plug it in using USB it decides to send the audio to controllers rumble device!!.

As to games - its a well known issue with both Epic and Ubisoft that you have to run their stupid game interface apps - even on Steam. 
I had them setup on Lutris but mainly using Steam at the moment (I have some very old games eg ones I still have a CD for) so not really using the Lutris path at the moment.
The one thing I do have setup is using eg protontricks or direct running of a script which uses the game windows files to be able to start eg the EA app for the game on its own in order to setup the EA side or deal with EA app updates (of course you have to setup the EA app for each game under wine/proton).

gronesy, 2023-06-14:
I guess I still had that fear of losing the install completely. I going to be installing Ubuntu today in your honor on the same SSD I have Regata installed. I'll tell it NOT to install grub. Then, I'll go into the YAST2 bootloader app and have it see Ubuntu. Wish me luck!

I actually installed on OSX and tested rEFInd earlier but was bummed when I lost it when OSX updated. Mainly because I had done custom template configuration. But I did think it was an excellent bootloader interface and I can see why you want that instead of the old option key.

I wish SteamOS could install alongside another OS. The latest HoloISO installer lets you choose all those options and users and passwords and then stops-LOL! If you say erase entire disk it works great. It's really because it's a cool system and you could probably play ALL of the Steam games you do on it. You should give it a shot!

We'll have to talk more about the udev stuff after I get a chance to see how many systems keep doing this.

It's shameful trying to play Ubisoft games (the bulk of my gaming right now. wouldn't you just know it-HA!) compared to the beautifully functioning native linux app Steam has. I swear, every time I log into a system again SOMETHING has changed and I either have to reinstall Lutris AND the Ubisoft connect app. Right now, on Garuda, after playing beautifully not two days ago the game gets past the first title minibox and just goes away. When I try to quit the app it tells me what a bad idea it is. HUH? Staring at a screen where NOTHING is happening is MY version of a bad idea-LOL!

I'll have to look at protontricks because I have seen it and wondered what it was. I also tried PlayOnLinux but Meh!

gronesy, 2023-06-15:
I installed Ubuntu next to Regata yesterday and there was no option to not install grub. The only choices I had were from the dropdown menu and none of them were DON'T INSTALL. Of course, it put it on the first part of my Mac fusion drive and not where I told it to AGAIN. This in spite of their being a fully valid EFI/boot partition on Ubuntu's drive. 

Things pretty much degraded from that. I've still not been able to recover my Manjaro install despite following the Manjaro forum procedure to the letter using their installer medium. 

Have you EVER had any luck with the boot-repair thing from Ubuntu? I've never had it fix anything.

Any suggestions for rescuing an entire drive (Manjaro/SteamOS) that won't boot?

I made myself go to Tai chi class today just so I could walk away and cool down-LOL!

davidjo, 2023-06-16:
Sorry - I just cant remember now the exact point I figured where to choose an option so didnt mess up.

Lets be clear - we are talking about an iMac system here - do you have an OSX install?
You can mount the EFI partition under OSX (its what refind does when it sets things up) or Linux - its usually mounted on Linux boot under /boot/efi.
As I said refind does a search on each boot to find installations (windows, OSX or linux) so you can mount random USB drives and still get boot options (unlike grub which makes a static list in grub.cfg only when update-grub is done).

Can you mount the drive and its partitions?
One thing would be to use chroot under any linux boot - in this process you mount the root partition you want to fix, setup and run chroot (there are many sets of instructions on the web) and then run update-grub which Im pretty certain only updates the grub configuration of the current chroot partition  as Ive used it a few times (not grub-install).

gronesy, 2023-06-16:
I looked at the standard install and the "Something Else" option and neither gave me the no install choice. The frustrating part for me is why ask me where I want it if you're going to put it somewhere else anyway? Keep in mind there is and has been a valid EFI partition with all installs I've attempted.

This is a 2017 iMac 5k i5-7600 or 18,3. I have OSX on the primary (Fusion) drive. All linux have been installed on external SSDs (Samsung T7).  What I've been doing is every time a linux puts a grub on the primary drive (or Time Machine) I used the command line diskutil commands to see and mount the drives and delete the EFI and grub stuff. For some reason, just this week my mounted drives no longer show up on the desktop. Do I just need to disconnect everything (including the new USB speakers) and boot up so it resets the mounted cache?

I was very impressed by rEFInd and how easy it was to set up. My problem this time is I actually on purpose deleted the EFI partition from the Manjaro drive so I don't think rEFInd would be able to find the boot info. I was hoping that using boot-repair or starting up from the Manjaro USB would allow me to fix this but no go.

I've run update-grub after mounting the partition to no effect. I spent too much time yesterday and thought it was a time for a new start. Don't laugh. I completely erased the drive and have installed Fedora Workstation 38 on the beginning partitions. I will install Garuda, with it's rEFInd bootloader later and have a fully functioning dual-boot system that doesn't need the Fusion drive grub droppings to boot.

What I mainly need to. figure out right now is how to get my Ubuntu/Regata dual boot system to boot from its own EFI partition so I can delete the stuff it put on my Fusion drive and Time Machine. 

I also looked into grub and grub rescue command line stuff and my eyes just glazed over-LOL! I simply couldn't reproduce what the linux.com tutorial was telling me to do.

I appreciate you sticking with me and let me know if you think this topic needs to move elsewhere.

davidjo, 2023-06-20:
Im pretty certain you only need an EFI partition on the internal drive - and yes refind will find bootable systems on any plugged in USB drives (at least thats what happens on my MacBook Pro - at least if they are plugged in before a reboot - I do have issues if you plug them in at the actual boot screen stage - it just doesnt see new drives plugged in at that point).
I think you may need a grub.cfg under /boot/grub for any linux root partition on usb drives - as I said you can fix this running update-grub under chroot for a root partition.

Ive avoided any BattleEye games for this reason - but I thought Proton version of wine now handles quite a lot of them - under my lutris system there is a lutris ge proton runner - not sure how much of Proton is now included in the actual wine source.

gronesy, 2023-06-21:
This Lutris/Wine version/esync-fsync nonsense has me completely baffled! On all four linux distros (three different types-arch, ubuntu, opensuse) Lutris now tells me before it installs Ubisoft that either the Wine version is incompatible or doesn't support esync or fsync or both. If I tell it to launch anyway I get the wine debugger window which goes crazy until I quit Ubisoft.

This is recent and I'm wondering if anybody else is having this problem. I can't believe it's just a mac thing.

I've used the wine version app on Lutris and installed and uninstalled several versions and it still won't work.

The sad part is only last week I had Ghost Recon: Breakpoint running on Garuda and looking better than Windows. And then it just stopped working. 

Do you know of ANY linux distros that are still written and updated for Intel macs like ours? The distrowatch listings are no help.

davidjo, 2023-06-21:
I did have a UPlay lutris setup working but it seems its not longer working as Ubisoft has updated to new name - but dont think esync/fsync is a mac problem - these are linux kernel features - did come across them because I started before fsync was generally available.
However this is on my PC machines and now I dont seem to be able to get Ubisoft to send me an email to reset password (even on OSX) to fix this!!

gronesy, 2023-06-21:
Luckily, I don't have any problem with Ubisoft or Steam sending me the confirmation emails.  I've had to empty out my Trusted Devices (20 limit) on Ubisoft twice now-LOL

Thanks again for the insight. From my previous ancient experience with linux usually PC hardware solved almost everything. It's troubling to think even PCs are having issues with this stuff. 

I just happened again. I decided to try LinuxMint for the hell of it and after I remember WHY I didn't keep it (grub throwing made me have to go into Mac recovery mode-UGH!) I tried to use Lutris/Ubisoft and Steam. 

I was beside myself when ubisoft and Breakpoint both installed without errors. Of course, it didn't launch either....

The Steam game Division 1 just looked like crap.

A full system update on both failed to help.

I think as soon as I can get some time with my Hackintosh expert and decide it too is not more trouble than it's worth I may just buy a decent PC for Mac, Windows and Linux.

And yes, I completely erased both Garuda and LinuxMint-LOL! I may just try another distro outside of the top 20 if you have any recommendations? I really DO like Ubuntu but that Ubisoft/Wine stuff, which may have nothing to do with the distro at all) just boils my blood!

I know linux has gotten WAY more complicated than when I started but I'd actually pay money for a distro that would just work! It's the Mac guy in me surfacing again-HA!

gronesy, 2023-06-28:
Just a quick update for you and anybody following this thread still...

After testing them on Parallels, I installed Debian 12 and Fedora Workstation 38 on external SSD drives. Both impressed me because I thought both used grub2 but did NOT throw those files on my primary mac drive or other drives. They simply did what I asked them: install them on the EFI partition the installer creates at the front of the drive they were installed on.

**Another hot tip for anybody having the same trouble with the Lutris/Wine/Ubisoft thing:**

On BOTH, I was able to find and install Lutris-0.5.12 first and successfully install Ubisoft Connect on both. In case I hadn't mentioned, ALL of the trouble I had been having with "Incompatible Wine Version", "Not Compatible with e_sync, f_sync" or both was with 0.5.13. 0.5.12 got around this. Then updating to 0.5.13 did NOT break Ubisoft.

I'll be doing more testing with installing and running games but I've already gotten the first Division to install which also used the SteamHelper app successfully. This gets around the errors everybody was getting when the defunct U-Play app tried to log into Ubisoft.

I'm ver encouraged by this even if it seems initially that the Vulkan drivers may be pushing my 8 G Radeon Pro 580 a little too hard. I could almost cook an egg on the back support of my iMac yesterday-LOL!

davidjo, 2023-06-28:
I just reinstalled Ubisoft Connect on Lutris and I noticed it appears that the default runner for windows games is now lutris ge proton.
Also noticed that Lutris on Ubuntu (Lunar Lobster) is version 0.5.12 and as you say has run perfectly fine for me.

gronesy, 2023-06-28:
Always glad to hear another mac linux person having success and the same experience I am. Thanks again for sharing!

With the Lutris-0.5.13, it didn't matter WHICH wine version I installed. Ubisoft either wouldn't install or wouldn't run if it did.

I don't know if I mentioned this or not but you should try opening Steam in Big Picture Mode. Steam has done something really cool. Big Picture Mode is now what SteamOS looks like when you install it. I know you won't want to keep it that way for resources but I think you'll get a kick out of it. It works EXACTLY like SteamOS which blows my mind!

Steam is still going to have to do some serious work on the OS to make it a contender with other consoles. Even if their niche is probably technically MOBILE gaming consoles, people do not expect to log into a branded game console to be told the game they love isn't approved or mostly plays with tweaks. Not why I sign into Steam or Ubisoft, for sure.

gronesy, 2023-07-02:
Another update in case it might help somebody. Until the 13th of July, Steam is offering some major Ubisoft titles at a discount. And all they do is link to your Ubisoft account but run on STEAM-YAY!

I was able to install and run the Division 2 on the Mageia Linux machine I recently built and it looks as good as it does on Windows! 

The bonus for me is that the interplay between Steam and Ubisoft is WAY more smooth than the other way around. And you don't have to mess with that Lutris/Wine nonsense and adding another layer to your game before you even start it.

I'm going to install Ghost Recon Breakpoint on the same machine today and hope it will work at least as well.

Good news all the way around!  Thanks again for your help!

davidjo, 2023-07-02:
Well my achievement is Ive figured out how to get an Ubisoft acquired game (actually a freebie) to run under Steam (ie not a Steam acquired Ubisoft game) - because even linking accounts didnt seem to bring the Ubisoft game into Steam - apart from running Ubisoft Connect then starting the game in Ubisoft Connect.
You have to do it as a non-Steam game initially running the Ubisoft Connect Installer exe but then changing the parameters to directly run  the real game - how to set the parameters was taken from the way Lutris installs Ubisoft games (Ubisoft Connect is still involved).

gronesy, 2023-07-03:
That's quite an accomplishment! I've seen the option to install as a non-Steam game but had very little confidence it would actually work given the Ubisoft Connect issues I've had repeatedly since I've started this journey. Good on you!

I was blown away yesterday when I installed Ghost Recon Breakpoint! I DID purchase it from Steam at a fraction of what I paid for it originally on Ubisoft (see the sale mentioned above until 7/13). I totally could not believe how much smoother the Ubisoft helper app ran than the Steam Helper. All I had to do was watch the little app check for updates, install them and then ran the game without ANY input from me!

The BIG, BIG win was that the graphics had been turned up to Very High from the Low to Medium I've experienced EVERY time I've run a game on Ubisoft Connect under Lutris.  Looks and plays BETTER than my Windows 11 install for the FIRST TIME EVER! I actually had to turn down the graphics to high so the game would play even smoother. And I was now getting 25-30 FPS consistently! Not astounding for NEW machines but EPIC for my 2017 iMac 5k 18,3!

The updated Steam linux app has opened up all sorts of new gaming possibilities! I couldn't be happier! I may even reinstall Garuda because it's such a beautiful and functional system that I just couldn't get the Lutris/Wine thing to work on.

The only challenge I've had is for voice chat in game. Apparently, most games running either under windows or linux see my Creative T60 USB speakers as speakers AND microphone. I've figured out under Windows how to use the speakers as an AUX plugin into my regular speaker jack on the iMac. But when I do that on linux, no surprise, it kicks in the same "dummy output" stuff where we started way back when.

I'm afraid to try installing any additional drivers for the squealing and other stuff that's happened if it not matched perfectly.

From what I understand, I shouldn't be using these games in multiplayer anyway because of performance or banning problems? Does this ring a bell with you?

davidjo, 2023-07-04:
Have you got protontricks - if not get it then you can run it and you can update the windows environment - if you run the wine config this eg lets you see what audio drivers "windows" thinks it is using - both speakers and mikes - and you can explicitly set the options.
Remember you have to do it separately for each windows game as each game is a separate windows environment.
Can you select the speakers explicilty under linux using the audio settings - you will still have dummy output from the internal audio device which still exists so ensure linux audio is not set to that.

Cant say anything about multiplayer and/or banning - although as proton now works for the DRM eg EasyAntiCheat, DeNuvo I would think its OK.
Multiplayer I can see as the round trip time time could well be affacted by the proton/wine translation overhead.

davidjo, 2023-07-06:
You might want to try protontricks - I just noticed that under wine config it lists 4 windows audio devices - Input device, Output device, Voice input device and Voice output device -  this is set to default by default(!!!) but each can be set to the explicit linux audio devices - as under Ubuntu the sound settings doesnt seem to make such a distinction not clear what the voice devices are being mapped to.
This probably does need some form of udev fixups.

davidjo, 2023-07-08:
I wont claim to be an export in the udev  system - but its the system used to perform actions when eg usb devices are plugged/unplugged (although its used for all devices - even inbuilt devices - but those can only trigger actions at boot).
The main system defined scripts are under /usr/lib/udev/rules.d.
Every USB device has an idVendor and idProduct and most rules work by matching those and specifying some action to be performed.
Ive had to use this for a number of fixups for macs.

gronesy, 2023-07-09:
Well, the Mageia system went back to thinking the USB speaker was the mic too-LOL!

Could you tell me which of the protontricks settings you choose? I'm thinking you were talking about using it for wineconfig and that is an option I've pulled up. It's tempting to change the system to windows 10 but is there something else I need to config in protontricks before I try the udev/rules.d option.

Not because it's really beyond my skill level but mainly because I've had such poor luck with anything linux actually seeing the CS8409 card.

davidjo, 2023-07-10:
Yes - run protontricks, select the game then OK the 64 bit dialog then select default prefix next window then next window choose run winecfg.
Select the Audio tab of the windows config window.
Im just suggesting you try setting eg the Voice input to the explicit linux mike device rather than Default.
(is it possible your speakers have a mike eg for Alexa style behaviour?).
Close protontricks by clicking top right X of the protontricks window.

I dont choose anything in general for sound - but I dont use mikes and default output is the display HDMI port.
My main messing with it recently is to get ps5 controllers working - windows games in particular older windows games assume Xbox controllers I think and for some reason the Steam Input didnt want to work (something to do with 3 different styles of windows drivers for input the game developers could have used and one of those styles apparently doesnt work with Steam Input) but figured (googling) a registry fixup which weirdly means the game sees the controller but as an Xbox controller so you get the xbox style button names etc. but otherwise works.

You should also do some experiments - display the sound settings window for your system, plug in eg the speakers and see what changes.

I had an issue again with PS5 controllers where if you plug it in using USB it was sending linux audio output to the rumble device (apparently the rumble device is driven from an audio signal so linux sets up an audio device for it - it used to switch the default audio output to this on plugin but it has apparently been fixed now - note this is pure linux nothing to do with steam - it also adds in a mike device for the PS5 internal mike). These changes are driven by the udev system.
Using it via Bluetooth didnt have these problems.

gronesy, 2023-07-10:
I've gone through the steps you've recommended even before your reply. The weird thing is that the only choice  for the mic other than "Default" I see is Pulseaudio which works when the mic is plugged into the USB speaker as a mic but not for USB.

The main problem remains that, like your PS5 controller, as soon as the USB speaker is connected linux see it as the speaker AND the mic. If I plug another USB headset into the computer (not the speaker) it instantly shows up and I can select it but it still ignores the internal mic unless I plug the non-USB headset connector into the AUX port of the USB speaker.

I'll go back and see but I haven't seen where the Protontricks/wineconfig step even sees the internal (CS8409) mic.

I also tried bluetooth first but that completely confused the linux sound system-LOL! It was buggy and glitchy at best.

davidjo, 2023-07-10:
Unfortunately the CS8409 internal mike is not properly hooked up to the kernel user side yet - it should appear as an option for linux input audio (which means I think it should appear in the wine config audio device pull down) but it does need manually setting to become the default device on linux - and that may well be messed up by plugging in other devices which also have an input device so would need manual re-setting as default input device.
But there are some differences between iMacs and MacBookPros handling of the internal mike which of course I cant test as dont have iMac - also remember the internal mike has a very low volume without some pusle/pipewire amplification.

gronesy, 2023-07-11:
The only choices I have when I follow your directions is two choices that are the Creative USB speaker and Pulseaudio.

I promise I wasn't trying to complicate things by not just getting a cheap set of USB-only speakers. I really wanted to make sure I'd covered my bases.

Of course, it doesn't help that Windows has no problem using the USB speakers and the internal mic perfectly-LOL! The only issue I've run into gaming on Windows with this setup is that one of my games wants to keep switching the mic back to the USB speaker from the Cirrus Logic choice. I have to set it back EVERY time I start Ghost Recon: Breakpoint. 

I haven't even tried playing with my buddy from linux yet but that's the next challenge.

davidjo, 2023-07-12:
If I understand correctly you have a USB speaker which has a mike jack - I can see issues - if you plug in a mike then plug the USB I think linux may handle it - does it see both an output and input device? - however if you plug in the USB speaker first without the mike then plug in the mike I dont see at the moment how the udev/USB system handles this - is a USB event issued? as the USB device ids will be the same not clear how can be handled - the question is if no mike is plugged in is there an input device?? - Windows might specifically add an input device even if not plugged in.
You might want to search for the vendor id/product id under /usr/lib/udev/rules.d to see if there is a specific udev file or search the web for  udev sample files for your speakers. lsusb should list these ids for plugged in devices.

gronesy, 2023-07-12:
Creative thought they'd cover all the bases. The USB speaker has mic, headset and auxiliary jacks on the back of it. I just cycle through the power button to select USB, bluetooth or auxiliary. Since I've had like ONE linux bluetooth work out of the box, I tried to see what I could do with the others. 

I hadn't thought that the order the devices were presented to the system would matter but I can see your point with this "piggyback" USB, bluetooth, auxiliary setup and why it would confuse linux. I guess just using either mac or windows as a comparison and how they handled them without incident was not a great starting point-LOL! I'll try the mic into the USB and THEN connect the USB to see if it matters. I'll report back.

I'll have a look through the /usr/lib/udev/rules.d and see if I find anything that resembles my speakers. I can also check the web. Did you already tell me what I need to do if I find it?

Leave it to you to know the lsusb command-HA! I'll try that too.

Thanks again!

davidjo, 2023-07-12:
Creative sounds like a big enough name that udev files might exist for them.

davidjo, 2023-07-14:
What I did was just grep Creative *.udev  - this did bring up a Creative product - but  an old pci sound card I think.
So what we need to get are the idVendor and id Product - either lsusb or lsusb -v for more detail should show up the Creative item - then you can search for these ids (2 byte hexadecimal quantities) - so far doesnt look to be other Creative definitions.
Then can try googling for these hex ids and udev and see if anything comes up.

davidjo, 2023-07-15:
I did see a file http://www.linux-usb.org/usb.ids which lists ids - and does have a big Creative section so should find your ids.

gronesy, 2023-07-15:
Thank you for that! I looked up my Creative T60 which is 041e:3280 and it's not there. What would I have been able to do if I HAD found it? Is this something that may be worth trying other Creative ids to see if they work? Or is that just a stupid waste of time?

I'm assuming if I had found the correct ids I could have done whatever wizardry you would have me do and then I wouldn't need USB exclusively for the sound to work?

davidjo, 2023-07-16:
Yes - Im not seeing any udev files or examples for Creative - except for some sound cards - so maybe they should just work.
I was hoping to see some form of example udev if something was needed.
Just discovered 2 new commands tho
inxi -Az
to display system audio devices etc.
and
wpctl status
- this is for Pipewire systems - which is pretty much any recent linux distribution version.
wpctl status gives nice overview of the audio setup.
seems to be replacement for pactl for pulse audio systems (note pacmd apparently doesnt work for pipewire based pulse but pactl still works).
Again try this before speakers plugged in and afterwards.
In particular check under Audio for a source that appears - this would be a microphone - sinks are output devices eg speakers.

Well the idea would be to create a udev file which eg could detect if the mike was connected and setup the audio paths explicitly.
(I still have no idea how plugging in the jack mike with speakers already connected is working).

I dont think you want to try the internal mike - that will need more work.

gronesy, 2023-07-19:
inxi on both my Manjaro (Arch) and Mageia (Independent, leaning towards SUSE?) yielded the same results where the audio was the "Device-1: Intel 100 Series/C230 Series Family HD Audio driver: snd_hda_intel driver. Probably because it was being ignored because of this: "Device-3: Creative T60 type: USB driver: hid-generic,snd-usb-audio,usbhid"

wpctl on either system was not installed and I couldn't find it in their repos.

The Manjaro system showed Alsa as the sound module and Mageia had Pipewire. But I was able to start the Pulseaudio daemon with the pacmd command.

In both cases, unplugging the T60 just removed that Device and driver.

I'm WAY out of my league here-LOL! I wouldn't know how to make a udev file file or what it should include if I did.

Let me know if I've answered the questions you needed answering. I'd love to do the udev thing but if my internal mic is not part of the mix we can just let this audio thing ride.

[17 more replies]