davidjo/snd_hda_macbookpro #93: Note of Service iMAC

ScottBaetz, 2023-06-29:
Just for those who are wondering, yes, follow the instructions and you will NOW FINALLY have sound that functions!  Meaning, this is not STRICTLY for the MacBookPro!

Thanks to @davidjo and anyone who assisted in creating this GREAT resource.  There needs to be more recognition provided!

thijstriemstra, 2023-07-26:
> this is not STRICTLY for the MacBookPro!

Also worked for me on a 27inch imac (late 2013) running popOS 22.04 with `6.2.6-76060206-generic` kernel.

brycewray, 2023-08-19:
Worked for me on a mid-2017 27-inch iMac running Fedora 38 with `6.4.10-200.fc38.x86_64` kernel, although I agree with another issue that requested the instructions for each distro include the `patch` command, too.

Amazing to see --- and, with this patch, hear --- a Windows game (“Spyro Reignited Trilogy”) running pretty much perfectly under Proton 8 in Linux. Incidentally, the sound seems fine, so perhaps the CoreAudio concerns mentioned in the README don’t apply to the iMac?

fspkwon, 2024-03-18:
Ill be testing on the iMac 18,1 2017 1st Generation..... I have a Ubuntu Live-CD
and a Gentoo Install....

P2AK, 2024-06-23:
This patch worked flawlessly for me on my 2017 iMac 5K Retina (18,3) after spending days trying to use special modprobe options with the stock kernel modules to no avail.  Thank you very much for this davidjo.  Any chance you would submit this into the mainstream kernel so we don't get the tainted kernel warning and so that others can benefit from your hard and ingenious work please?

davidjo, 2024-06-24:
Thanks for the kind words.
Unfortunately Im not a kernel developer in any sense so I have no real idea how to do the submission plus there are some issues with this module which likely makes it unacceptable in its current form. At the moment Im trying to see if I can replace my hacked up blocking with proper kernel mutex locks but locks are a big pain to get right.

fspkwon, 2024-07-02:
@davidjo - Would you care to help me understand why building this on gentoo is the worlds next most impossible thing<- , it builds from what i can see in the source directory but any attempt to use what it compiles doesnt work, and the actual build folder and update directory are both empty... ..... ... i was wondering if maybe it was a issue with a newer kernel version as gentoo uses actually just the regular linux kernel this would be testable just on the whatever kernels just been posted to ftp, over in bsd world its just the 8409 driver thats missing 42l42 seems to be there

fspkwon, 2024-07-02:
@davidjo and i cant say theres not 1-single issue using ubuntu not --a-single-one

fspkwon, 2024-07-02:
@davidjo https://pastebin.com/AmHeFjBC imac 18,1 2017 base imac configuration owners glory time kernel-config

davidjo, 2024-07-02:
There is an old issue #28 which talks about Gentoo and it seems it uses a very different kernel/package build system to ubuntu and a lot of other distributions - its unlikely the current build system scripts here will work on Gentoo based on that.
I did give some suggestions in that old issue (which dont seem to have been followed up) but I have never used Gentoo so cant really offer any help.

davidjo, 2024-07-02:
As to these Cirrus chips, 8409, 42l42 plus Apples custom versions this is what Im guessing so far.
Apple uses the 8409 chips with their custom 42l83 (which is some form of 42l42). 
The 8409 chip seems to be undocumented but there is a pdf doc for the 42l42 - but not the 42l83 and I know there are some differences in registers between the 42l42 and 42l83.
Currently the linux kernel has the 8409/42l42 system under sound/pci/hda but Ive also noticed there is a separate direct chip driver for 42l42 (among other) under sound/soc/codecs - seems like BSD may be an implementation of these drivers - where the 42l42 is connected directly to a CPU i2c bus.
Using the 8409 the 42l42/42l83 chip is connected to a local private i2c bus between the 8409/42l42(42l83).
Also note that macs additionally drive amp chips from the 8409 for the speakers - the 42l83 has nothing to do with the speakers  (its unclear whats going on with the speakers for the Dell laptops).

P2AK, 2024-07-02:
@fspkwon are you using sys-kernel/gentoo-sources ~amd64 ?  I'm using them and this driver works great.

fspkwon, 2024-07-03:
I Figured as much , @davidjo yes they have a SoC driver device-tree firmware, @p2ak i build my kernel, everythings configured only for what i think is needed and im compiling alsa INTO the kernel no mods or anything and i DO specify a hostname 6.3.30-gentoo-x86_64 and i makes a DUPLICATE of the string when it goes to build it so like /lib/modules/6.6.30-gentoo6.6.30-gentoo-x86_64 and i dont know why and also the driver it does build because i looked as far as i can find what drivers are there what drivers arent it does say like soc-snd-cs8409 and i think thats what he said up there is that theyre 2 different hardware configurations so maybe its not patching the source maybe it doesnt know where it is see i dont know...

fspkwon, 2024-07-03:
so it just said so theyre gonna take a this is the easiest thing someone
didnt get wrong
is a green like catch a ball thing and stabs it to inactive activity -
or no its the room talks about something they then they re-enact like a
kids room

fspkwon, 2024-07-03:
Fedora and HTML 5 is its round peg square hole theory can be stapler latched

davidjo, 2024-07-03:
the module you need is snd-hda-codec-cs8409.ko under the sound/pci/hda directory - the sound/soc/codecs doesnt apply for macs