davidjo/snd_hda_macbookpro #94: Not working on Macbook 2015 retina with Ubuntu 22.4.2 LTS

amirian, 2023-07-01:
Title is expressive!

davidjo, 2023-07-01:
But Im afraid completely useless - I have no iead what a Macbook 2015 Retina is - I need to know the model identifier minimally eg mine is MacbookPro 14,3.

amirian, 2023-07-01:
That is also expressive, needs a simple search: [MacBook (Retina, 12-inch, Early 2015)](https://support.apple.com/kb/SP712)

davidjo, 2023-07-01:
That search result does not list your Model Identifier.
You need to boot OSX, click on Apple top left, click About this Mac then click on System Report - that lists your Model Identifier.

amirian, 2023-07-01:
It is Macbook 8,1

davidjo, 2023-07-01:
Then Im sorry - this repo doesnt work for your machine because thats older hardware - it is not a CS8409 system.
From what Ive quickly checked sound should work out of the box (see https://help.ubuntu.com/community/MacBookPro8-1/Natty#Sound) - well it did on older Ubuntu versions and I had no problems with sound on my previous Apple laptops when booted into linux.
Also try https://wiki.archlinux.org/title/MacBookPro8,x#Sound which also has information.

amirian, 2023-07-01:
But this is Macbook 8,1 not MacbookPro 8,1 which [seems to be unknown to Ubuntu community] (https://help.ubuntu.com/community/MacBook8-1). Then how can I uninstall this repo if not useful for my Macbook model?

amirian, 2023-07-01:
As mentioned in the previous bug report, microphone works, headphone jack works and only the built-in speaker doesn't work! Moreover, it is mentioned that due to the windows driver, the device type is Cirrus Logic CS4208 (AB 100).

davidjo, 2023-07-01:
Sorry - missed that my link was the MacboookPro 8,1 - but the CS4208 is also the wrong chip - this was the version prior to the CS8409.
And you have found one of the most useful kernel bug reports for this chip and that is the closest implementation it looks like you can get.

To remove this do rm -fr  "your repo clone directory name".
Then look under /lib/modules/kernel version-generic eg 5.19.0-21-generic for a directory updates.
In that directory look for snd-hda-codec-cs8409.ko and do rm snd-hda-codec-cs8409.ko - this needs to be done as superuser.

amirian, 2023-07-01:
Thank you. I wonder if this long-time kernel bug would be fixed by somebody. I found this patch on the bug report page: https://github.com/leifliddy/macbook12-audio-driver but it only works for Macbook9,1 and Macbook10,1 which are very similar to Macbook8,1

davidjo, 2023-07-01:
I suspect not - it is now a very old machine so the number of users really needing such a machine and having the knowledge to fix it is likely very small.

My suggestion would be to get some usb (or bluetooth) speakers - they have always worked - or  swap the laptop for one which does have working audio under linux.

DavidMorton, 2023-07-24:
I'll add my name into the hat of people trying to get this particular model working. I was just looking for this the other day. Tried both @leifliddy and your repo, but it did little except get rid of the volume controls on the title bar.

davidjo, 2023-07-25:
Your best starting point is the leifliddy version.
Unfortunately it seems that version is not fixed for the macbook 8,1.
I think its fixable but not something I can work on.

Your best hope is to get the 
cs4208_xx.inf file (eg cs4208_39.inf) from a Bootcamp installer for installing windows on macs - which is how I got my versions.
This does describe the setup required for a lot of cirrus systems for the windows drivers but the eg InitVerbs type descriptions give the HDA verbs to initialize the chips.

Try checking the cs420x_fixup_tbl  - this lists the CS420X_MBP81 as an entry (yes thats the pro version) 
- you need to find your codec SSID - add  the following print
codec_info(codec, "Picked ID=%d, VID=%08x, DEV=%08x\n", codec->fixup_id,
			 codec->bus->pci->subsystem_vendor,
			 codec->bus->pci->subsystem_device);
to patch_cs420x just before the snd_hda_pick_fixup.
(codec_info is a macro - hope its defined - if not take the format string and variables and convert into printk statement).

These give the ids needed to fix up the 8,1 - Im assuming for now the subsystem ids are different from those that exist - if they match an existing fixup that will need a different strategy.

You need these ids to trace through the cs4208_xx.inf file to see how windows does it.

Now add a new entry to the enum eg CS420X_MB81.
You then need to figure out how to fixup the following cs420x_fixup_tbl ,  cs4208_mac_fixup_tbl, cs420x_models with this new id (note that some go through the generic cx4208 fixups, some through the specific mac fixup table).

Youll to create an out of tree kernel module build (thats whats under the build directory of my repo once the install script has been run - leif liddys version will be similar).

It does look as though there is a default path set by CS4208_MAC_AUTO (theres also CS4208_AUTO) which may be the reason why the base kernel module gets somethings right but the leif liddy version doesnt as you seem to have observed.

DavidMorton, 2023-07-25:
> Any success report on Macbook8,1 would be appreciated

I’ll definitely update if I get something working. More than likely it’s gonna be a while though if I manage to crack it at all, so don’t hold your breath. There are tons of concepts for me to learn, and I have a day job :)Sent from my iPhone

DavidMorton, 2023-07-27:
So I've been working on this a bit...

My latest alsa info is here...

https://github.com/DavidMorton/macbook12-audio-driver/blob/master/alsa-output/2023-07-27-15-19.txt

The thing I'm kinda stumped on is that I'm manually setting the wcaps a couple of times, but it doesn't seem to be holding. I set it directly using snd_hda_override_wcaps, and then when I check the value after in the code, it's correct, but alsa-info is reporting the wrong value. 

Here's a link... I'm calling this everywhere but I'm baffled why it's changed by the time the computer boots up. 

https://github.com/DavidMorton/macbook12-audio-driver/blob/6c4e2943f556540c969ef5bccb9ec8e9afe72ac4/patch_cirrus/patch_cirrus.c#L864

Was wondering if @davidjo might have any quick ideas as to why. 

FYI, copious notes here: https://github.com/DavidMorton/macbook12-audio-driver/blob/master/info/notes.md.

I'm not a driver dev, but I play one on TV :)

davidjo, 2023-07-27:
Im not sure but the overrides may not override the chip values but cached values (snd_hda_override_wcaps is defined in hda_local.h - a base kernel source file) so alsa info may not report it - as I think that directly reads the chip values again.
Might want to download the intel HDA spec document - dont have to read it in detail (well I havent) but it gives some clues as to whats going on.

PS I am not a driver dev either - just read code and googled - had never worked on kernel code/audio driver code before this.

DavidMorton, 2023-08-03:
Alright, I gave it the old college try, but when the machine starts to overheat and auto-shutdown, to resume a moment later only to find the trackpad and keyboard don't work because... linux... I think I'm out for a while now. If anyone else wants to look at it, I've checked in all my code (with copious notes) here:

https://github.com/DavidMorton/macbook12-audio-driver