davidjo/snd_hda_macbookpro #97: Debian Kernel 5.10.0-25-amd64 MacBook Pro 14,1 Script does not work - patch too

Pablo12345678901, 2023-09-14:
I cannot get sound output/input. It seems that input is not available on MacBook Pro 14,1 which is my device - if it is available now please let me know. Main issue here is output sound.

I was not able to find the solution neither with the scripts of "snd_hda_macbookpro" and "https://github.com/leifliddy/macbook12-audio-driver" repositories nor by myself with your comment : I do not find "snd_hda_codec_cirrus" module and if I load it after reboot it is not anymore loaded - sound does not work.

I check if my sound was mute - it is not.

I have done #63 and this is the output when I run the script : 

Command : 
sudo ./install.cirrus.driver.sh 
reboot

Output : 
--2023-09-14 09:19:19--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.0.tar.xz
Résolution de cdn.kernel.org (cdn.kernel.org)… 2a04:4e42:8d::432, 146.75.117.176
Connexion à cdn.kernel.org (cdn.kernel.org)|2a04:4e42:8d::432|:443… connecté.
requête HTTP transmise, en attente de la réponse… 404 Not Found
2023-09-14 09:19:19 erreur 404 : Not Found.

Failed to download linux-5.10.0.tar.xz
Trying to download base kernel version linux-5.10.tar.xz
This may lead to build failures as too old
If this is an Ubuntu-based distribution this almost certainly will fail to build

--2023-09-14 09:19:19--  https://cdn.kernel.org/pub/linux/kernel/v5.x/linux-5.10.tar.xz
Résolution de cdn.kernel.org (cdn.kernel.org)… 2a04:4e42:8d::432, 146.75.117.176
Connexion à cdn.kernel.org (cdn.kernel.org)|2a04:4e42:8d::432|:443… connecté.
requête HTTP transmise, en attente de la réponse… 200 OK
Taille : 116606704 (111M) [application/x-xz]
Sauvegarde en : « build/linux-5.10.tar.xz »

linux-5.10.tar.xz   100%[===================>] 111.20M  24.6MB/s    ds 5.2s    

2023-09-14 09:19:24 (21.2 MB/s) — « build/linux-5.10.tar.xz » sauvegardé [116606704/116606704]

linux-5.10/sound/pci/hda/
linux-5.10/sound/pci/hda/Kconfig
linux-5.10/sound/pci/hda/Makefile
linux-5.10/sound/pci/hda/ca0132_regs.h
linux-5.10/sound/pci/hda/hda_auto_parser.c
linux-5.10/sound/pci/hda/hda_auto_parser.h
linux-5.10/sound/pci/hda/hda_beep.c
linux-5.10/sound/pci/hda/hda_beep.h
linux-5.10/sound/pci/hda/hda_bind.c
linux-5.10/sound/pci/hda/hda_codec.c
linux-5.10/sound/pci/hda/hda_controller.c
linux-5.10/sound/pci/hda/hda_controller.h
linux-5.10/sound/pci/hda/hda_controller_trace.h
linux-5.10/sound/pci/hda/hda_eld.c
linux-5.10/sound/pci/hda/hda_generic.c
linux-5.10/sound/pci/hda/hda_generic.h
linux-5.10/sound/pci/hda/hda_hwdep.c
linux-5.10/sound/pci/hda/hda_intel.c
linux-5.10/sound/pci/hda/hda_intel.h
linux-5.10/sound/pci/hda/hda_intel_trace.h
linux-5.10/sound/pci/hda/hda_jack.c
linux-5.10/sound/pci/hda/hda_jack.h
linux-5.10/sound/pci/hda/hda_local.h
linux-5.10/sound/pci/hda/hda_proc.c
linux-5.10/sound/pci/hda/hda_sysfs.c
linux-5.10/sound/pci/hda/hda_tegra.c
linux-5.10/sound/pci/hda/hp_x360_helper.c
linux-5.10/sound/pci/hda/patch_analog.c
linux-5.10/sound/pci/hda/patch_ca0110.c
linux-5.10/sound/pci/hda/patch_ca0132.c
linux-5.10/sound/pci/hda/patch_cirrus.c
linux-5.10/sound/pci/hda/patch_cmedia.c
linux-5.10/sound/pci/hda/patch_conexant.c
linux-5.10/sound/pci/hda/patch_hdmi.c
linux-5.10/sound/pci/hda/patch_realtek.c
[... 206 lines cut ...]

Command :
lspci | grep -i 'audio'

Output :
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)

Command :
sudo journalctl -b 0 -p 5 | grep -i 'audio|snd|sound|firmware'

Output :
sep 13 22:55:29 laptop-incognito kernel: ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored
sep 13 22:55:30 laptop-incognito kernel: brcmfmac 0000:02:00.0: firmware: failed to load brcm/brcmfmac4350c2-pcie.Apple Inc.-MacBookPro14,1.txt (-2)
sep 13 22:55:30 laptop-incognito kernel: firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
sep 13 22:55:30 laptop-incognito kernel: brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.Apple Inc.-MacBookPro14,1.txt failed with error -2
sep 13 22:55:30 laptop-incognito kernel: brcmfmac 0000:02:00.0: firmware: failed to load brcm/brcmfmac4350c2-pcie.txt (-2)
sep 13 22:55:30 laptop-incognito kernel: brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4350c2-pcie.txt failed with error -2
sep 13 22:55:30 laptop-incognito kernel: brcmfmac 0000:02:00.0: firmware: failed to load brcm/brcmfmac4350c2-pcie.clm_blob (-2)
sep 13 22:55:30 laptop-incognito kernel: bluetooth hci0: firmware: failed to load brcm/BCM.hcd (-2)
sep 13 22:55:30 laptop-incognito kernel: Bluetooth: hci0: BCM: firmware Patch file not found, tried:
sep 13 22:55:43 laptop-incognito libcanberra-login-sound.desktop[1584]: Failed to play sound: File or data not found

Please ask any other missing information and I will provide it.

Thank you.

davidjo, 2023-09-14:
Not sure where to start:

Why do you have a symbolic link in the updates directory (snd-hda-codec-cirrus.ko -> snd-hda-codec-cirrus.ko_speaker) - none of my scripts do this?

I dont have a 14,1  but from the issues it has been reported working for 14,1 machines - it does have the 8409 chip (I cant verify any of this tho).

However - from your logs the module is not being recognised as the actual module being loaded is snd_hda_codec_generic so you wont get any sound (hence the question about the symbolic link - Ive never tried to use them for kernel modules even tho it appears my updated module is being built and installed from your logs - for kernel 5.10 that is snd_hda_codec_cirrus).

Also - what version of debian are you using - kernel 5.10 is very old now - note my error message This may lead to build failures as too old - I would suggest upgrading debian to a much later version - preferably the current version - this module is currently mainly working on kernel version 6.

You did just clone my current repo and ran the install script - you didnt try and use an earler commit? (I would assume thats your first attempt).
Note that the current head of the repo does not need the 64.diff patch and it shouldnt be applied (it looks like the patch failed in any case - le patch ne s'applique pas - which would mean the 2nd run is just a repeat of the first run).

I attempted to have the install script still work for older kernel versions but I have not in any way tested this so it may not be working.

PS note that you are also having problems with wifi - the brcm log failures - and bluetooth - which should be fixed by a current kernel version.

Pablo12345678901, 2023-09-15:
Thank you for your quick reply.

I was with PureOs distro which is stuck on this kernel version. 

Command : 
sudo apt-cache search linux-image
[sudo] Mot de passe de connex-incognito : 

Output : 
linux-headers-5.10.0-22-amd64 - Header files for Linux 5.10.0-22-amd64
linux-headers-5.10.0-22-cloud-amd64 - Header files for Linux 5.10.0-22-cloud-amd64
linux-headers-5.10.0-22-rt-amd64 - Header files for Linux 5.10.0-22-rt-amd64
linux-image-5.10.0-22-amd64 - Linux 5.10 for 64-bit PCs (signed)
linux-image-5.10.0-22-amd64-dbg - Debug symbols for linux-image-5.10.0-22-amd64
linux-image-5.10.0-22-amd64-unsigned - Linux 5.10 for 64-bit PCs
linux-image-5.10.0-22-cloud-amd64 - Linux 5.10 for x86-64 cloud (signed)
linux-image-5.10.0-22-cloud-amd64-dbg - Debug symbols for linux-image-5.10.0-22-cloud-amd64
linux-image-5.10.0-22-cloud-amd64-unsigned - Linux 5.10 for x86-64 cloud
linux-image-5.10.0-22-rt-amd64 - Linux 5.10 for 64-bit PCs, PREEMPT_RT (signed)
linux-image-5.10.0-22-rt-amd64-dbg - Debug symbols for linux-image-5.10.0-22-rt-amd64
linux-image-5.10.0-22-rt-amd64-unsigned - Linux 5.10 for 64-bit PCs, PREEMPT_RT
linux-image-amd64 - Linux for 64-bit PCs (meta-package)
linux-image-amd64-dbg - Debugging symbols for Linux amd64 configuration (meta-package)
linux-image-amd64-signed-template - Template for signed linux-image packages for amd64
linux-image-cloud-amd64 - Linux for x86-64 cloud (meta-package)
linux-image-cloud-amd64-dbg - Debugging symbols for Linux cloud-amd64 configuration (meta-package)
linux-image-rt-amd64 - Linux for 64-bit PCs (meta-package)
linux-image-rt-amd64-dbg - Debugging symbols for Linux rt-amd64 configuration (meta-package)
linux-headers-5.10.0-25-amd64 - Header files for Linux 5.10.0-25-amd64
linux-headers-5.10.0-25-cloud-amd64 - Header files for Linux 5.10.0-25-cloud-amd64
linux-headers-5.10.0-25-rt-amd64 - Header files for Linux 5.10.0-25-rt-amd64
linux-image-5.10.0-25-amd64 - Linux 5.10 for 64-bit PCs (signed)
linux-image-5.10.0-25-amd64-dbg - Debug symbols for linux-image-5.10.0-25-amd64
linux-image-5.10.0-25-amd64-unsigned - Linux 5.10 for 64-bit PCs
linux-image-5.10.0-25-cloud-amd64 - Linux 5.10 for x86-64 cloud (signed)
linux-image-5.10.0-25-cloud-amd64-dbg - Debug symbols for linux-image-5.10.0-25-cloud-amd64
linux-image-5.10.0-25-cloud-amd64-unsigned - Linux 5.10 for x86-64 cloud
linux-image-5.10.0-25-rt-amd64 - Linux 5.10 for 64-bit PCs, PREEMPT_RT (signed)
linux-image-5.10.0-25-rt-amd64-dbg - Debug symbols for linux-image-5.10.0-25-rt-amd64
linux-image-5.10.0-25-rt-amd64-unsigned - Linux 5.10 for 64-bit PCs, PREEMPT_RT
linux-headers-5.10.0-23-amd64 - Header files for Linux 5.10.0-23-amd64
linux-image-5.10.0-23-amd64 - Linux 5.10 for 64-bit PCs (signed)

I will switch to another distro with kernel 6 and then check again if after the sound re-work with your script. Without any message from my part within the next week you can consider the issue as closed.

AGAIN : Thank you for your all your advices (much appreciated) - I will re-check them after changing distro.

davidjo, 2023-09-15:
Having looked at descriptions of PureOS I dont think this is a good distribution for working with Macs - it seems to be designed for pretty standard PC hardware and Macs have many peculiar hardware features.

davidjo, 2023-09-17:
Another thing to check for PureOS.
What I dont understand why its not loading snd_hda_codec_cirrus - the module is being built and I dont see build errors in your logs so it should be loaded - hence the issue with the symbolic link.
Only thing I can think is the patch_cirrus.c file is not being updated to add the cs8409 definition (I realised I never actually built for 5.10 - ubuntu LTS went from 5.8 to 5.13).
Check that your PureOS build had the following line in patch_cirrus.c in the repo build directory in the struct definition for snd_hda_id_cirrus (near bottom of file).
HDA_CODEC_ENTRY(0x10138409, "CS8409", patch_cs8409),
If this is not there then it will not load snd_hda_codec_cirrus.