davidjo/snd_hda_macbookpro #101: No sound on MacBookPro13,3 running Fedora 68 w/ kernel 6.2.9-300.fc38.x86_64

vascobrown, 2023-10-14:
I'm not sure if the patch is getting applied properly. When I initially ran the install script, I got the following error:

```
linux-6.2.9/sound/pci/hda/
linux-6.2.9/sound/pci/hda/Kconfig
linux-6.2.9/sound/pci/hda/Makefile
linux-6.2.9/sound/pci/hda/ca0132_regs.h
linux-6.2.9/sound/pci/hda/cs35l41_hda.c
linux-6.2.9/sound/pci/hda/cs35l41_hda.h
linux-6.2.9/sound/pci/hda/cs35l41_hda_i2c.c
linux-6.2.9/sound/pci/hda/cs35l41_hda_spi.c
linux-6.2.9/sound/pci/hda/hda_auto_parser.c
linux-6.2.9/sound/pci/hda/hda_auto_parser.h
linux-6.2.9/sound/pci/hda/hda_beep.c
linux-6.2.9/sound/pci/hda/hda_beep.h
linux-6.2.9/sound/pci/hda/hda_bind.c
linux-6.2.9/sound/pci/hda/hda_codec.c
linux-6.2.9/sound/pci/hda/hda_component.h
linux-6.2.9/sound/pci/hda/hda_controller.c
linux-6.2.9/sound/pci/hda/hda_controller.h
linux-6.2.9/sound/pci/hda/hda_controller_trace.h
linux-6.2.9/sound/pci/hda/hda_cs_dsp_ctl.c
linux-6.2.9/sound/pci/hda/hda_cs_dsp_ctl.h
linux-6.2.9/sound/pci/hda/hda_eld.c
linux-6.2.9/sound/pci/hda/hda_generic.c
linux-6.2.9/sound/pci/hda/hda_generic.h
linux-6.2.9/sound/pci/hda/hda_hwdep.c
linux-6.2.9/sound/pci/hda/hda_intel.c
linux-6.2.9/sound/pci/hda/hda_intel.h
linux-6.2.9/sound/pci/hda/hda_intel_trace.h
linux-6.2.9/sound/pci/hda/hda_jack.c
linux-6.2.9/sound/pci/hda/hda_jack.h
linux-6.2.9/sound/pci/hda/hda_local.h
linux-6.2.9/sound/pci/hda/hda_proc.c
linux-6.2.9/sound/pci/hda/hda_sysfs.c
linux-6.2.9/sound/pci/hda/hda_tegra.c
linux-6.2.9/sound/pci/hda/hp_x360_helper.c
linux-6.2.9/sound/pci/hda/ideapad_s740_helper.c
linux-6.2.9/sound/pci/hda/patch_analog.c
linux-6.2.9/sound/pci/hda/patch_ca0110.c
linux-6.2.9/sound/pci/hda/patch_ca0132.c
linux-6.2.9/sound/pci/hda/patch_cirrus.c
linux-6.2.9/sound/pci/hda/patch_cmedia.c
linux-6.2.9/sound/pci/hda/patch_conexant.c
linux-6.2.9/sound/pci/hda/patch_cs8409-tables.c
linux-6.2.9/sound/pci/hda/patch_cs8409.c
linux-6.2.9/sound/pci/hda/patch_cs8409.h
linux-6.2.9/sound/pci/hda/patch_hdmi.c
linux-6.2.9/sound/pci/hda/patch_realtek.c
linux-6.2.9/sound/pci/hda/patch_si3054.c
linux-6.2.9/sound/pci/hda/patch_sigmatel.c
linux-6.2.9/sound/pci/hda/patch_via.c
linux-6.2.9/sound/pci/hda/thinkpad_helper.c
Kernel version later than implemented version - there may be build problems
patching file patch_cs8409.c
Hunk #1 succeeded at 1444 (offset 174 lines).
Hunk #2 succeeded at 1453 (offset 174 lines).
Hunk #3 succeeded at 1482 (offset 174 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 303 (offset 1 line).
Hunk #3 succeeded at 330 (offset 1 line).
Hunk #4 succeeded at 366 (offset 1 line).
patching file patch_cirrus_apple.h
make -C /lib/modules/6.2.9-300.fc38.x86_64/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/vascobrown/Downloads/snd_hda_macbookpro/build/hda-6.2.9 modules
make[1]: Entering directory '/home/vascobrown/Downloads/snd_hda_macbookpro/build/hda-6.2.9'
make[1]: *** /lib/modules/6.2.9-300.fc38.x86_64/build: No such file or directory.  Stop.
make[1]: Leaving directory '/home/vascobrown/Downloads/snd_hda_macbookpro/build/hda-6.2.9'
make: *** [Makefile:23: all] Error 2
```

[... 113 lines cut ...]
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hda_gen_fix_pin_power
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol _snd_hda_set_pin_ctl
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hda_codec_set_name
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hda_get_path_from_idx
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hda_jack_unsol_event
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hda_gen_build_pcms
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_array_new
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hda_pick_fixup
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hda_apply_fixup
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hda_add_verbs
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hda_gen_parse_auto_config
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hdac_codec_read
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hda_gen_free
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hdac_regmap_sync
depmod: WARNING: /lib/modules/6.5.6-200.fc38.x86_64/updates/snd-hda-codec-cs8409.ko needs unknown symbol snd_hda_jack_add_kctl_mst
depmod: WARNING: could not open modules.builtin at /lib/modules/6.5.6-200.fc38.x86_64: No such file or directory
depmod: WARNING: could not open modules.builtin.modinfo at /lib/modules/6.5.6-200.fc38.x86_64: No such file or directory
make[1]: Leaving directory '/usr/src/kernels/6.5.6-200.fc38.x86_64'
depmod -a 6.2.9-300.fc38.x86_64

contents of /lib/modules/6.2.9-300.fc38.x86_64/updates
total 0
```

It looks like `depmod` is throwing a bunch of warnings including several folder that it is unable to find. I checked and they indeed don't exist, but I'm not sure where to go from here. Any suggestions?

vascobrown, 2023-10-14:
Also, when I run `inxi -Fxxrzc0`, the 'Audio' sections shows:

```
  Device-1: Intel 100 Series/C230 Series Family HD Audio driver: snd_hda_intel
    v: kernel bus-ID: 00:1f.3 chip-ID: 8086:a170
  Device-2: AMD Baffin HDMI/DP Audio [Radeon RX 550 640SP / 560/560X]
    driver: snd_hda_intel v: kernel pcie: speed: 8 GT/s lanes: 8 bus-ID: 01:00.1
    chip-ID: 1002:aae0
  API: ALSA v: k6.2.9-300.fc38.x86_64 status: kernel-api
  Server-1: PipeWire v: 0.3.67 status: active with: 1: pipewire-pulse
    status: active 2: wireplumber status: active 3: pipewire-alsa type: plugin
    4: pw-jack type: plugin
```

However, in Settings itI shows my Radeon card as the only output device option, with the Intel sound card nowhere to be found:

[image]
[image]

davidjo, 2023-10-14:
The patch is being applied properly (eg Hunk #1 succeeded messages).

I think the primary issue is you must install the source for kernel 6.2.9-300.fc38.x86_64 and not use 6.2.9-200.fc38.x86_64.
I dont know Fedora but you need to figure out how to install the source for 6.2.9-300.fc38.x86_64 just as the source for the previous version  6.2.9-200.fc38.x86_64 was installed.

By using the symbolic kink to the wrong kernel version instead of installing to /lib/modules/6.2.9-300.fc38.x86_64  it installed to /lib/modules/6.2.9-200.fc38.x86_64  which appears to have been cleaned out leading to missing symbol errors.
In any case it doesnt matter - it did not load the module into the current kernels module directory ie /lib/modules/6.2.9-300.fc38.x86_64 (see lines after "content of.../updates")  but the older kernel - so the current kernel cannot load the module hence no sound device.

vascobrown, 2023-10-14:
Thanks for the response. I'm a linux beginner so your help is greatly appreciated.

On your suggestion, I went to kernels.org downloaded the source `linux-6.2.9.tar.xz` and copied the extracted folder contents to to `/usr/src/kernels/6.2.9`.  I then removed the symlink I created previously and replaced it with `sudo ln -s /usr/src/kernels/6.2.9 /lib/modules/6.2.9-300.fc38.x86_64/build`. However, now when I run the install script I  get an error that a file `auto.conf` can't be found. Here's the full output...

```
--2023-10-14 16:45:57--  https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.2.9.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 146.75.93.176, 2a04:4e42:87::432
Connecting to cdn.kernel.org (cdn.kernel.org)|146.75.93.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-6.2.9/sound/pci/hda/
linux-6.2.9/sound/pci/hda/Kconfig
linux-6.2.9/sound/pci/hda/Makefile
linux-6.2.9/sound/pci/hda/ca0132_regs.h
linux-6.2.9/sound/pci/hda/cs35l41_hda.c
linux-6.2.9/sound/pci/hda/cs35l41_hda.h
linux-6.2.9/sound/pci/hda/cs35l41_hda_i2c.c
linux-6.2.9/sound/pci/hda/cs35l41_hda_spi.c
linux-6.2.9/sound/pci/hda/hda_auto_parser.c
linux-6.2.9/sound/pci/hda/hda_auto_parser.h
linux-6.2.9/sound/pci/hda/hda_beep.c
linux-6.2.9/sound/pci/hda/hda_beep.h
linux-6.2.9/sound/pci/hda/hda_bind.c
linux-6.2.9/sound/pci/hda/hda_codec.c
linux-6.2.9/sound/pci/hda/hda_component.h
linux-6.2.9/sound/pci/hda/hda_controller.c
linux-6.2.9/sound/pci/hda/hda_controller.h
linux-6.2.9/sound/pci/hda/hda_controller_trace.h
linux-6.2.9/sound/pci/hda/hda_cs_dsp_ctl.c
linux-6.2.9/sound/pci/hda/hda_cs_dsp_ctl.h
linux-6.2.9/sound/pci/hda/hda_eld.c
linux-6.2.9/sound/pci/hda/hda_generic.c
linux-6.2.9/sound/pci/hda/hda_generic.h
linux-6.2.9/sound/pci/hda/hda_hwdep.c
linux-6.2.9/sound/pci/hda/hda_intel.c
linux-6.2.9/sound/pci/hda/hda_intel.h
linux-6.2.9/sound/pci/hda/hda_intel_trace.h
linux-6.2.9/sound/pci/hda/hda_jack.c
linux-6.2.9/sound/pci/hda/hda_jack.h
linux-6.2.9/sound/pci/hda/hda_local.h
linux-6.2.9/sound/pci/hda/hda_proc.c
linux-6.2.9/sound/pci/hda/hda_sysfs.c
linux-6.2.9/sound/pci/hda/hda_tegra.c
linux-6.2.9/sound/pci/hda/hp_x360_helper.c
linux-6.2.9/sound/pci/hda/ideapad_s740_helper.c
linux-6.2.9/sound/pci/hda/patch_analog.c
linux-6.2.9/sound/pci/hda/patch_ca0110.c
linux-6.2.9/sound/pci/hda/patch_ca0132.c
linux-6.2.9/sound/pci/hda/patch_cirrus.c
linux-6.2.9/sound/pci/hda/patch_cmedia.c
linux-6.2.9/sound/pci/hda/patch_conexant.c
linux-6.2.9/sound/pci/hda/patch_cs8409-tables.c
linux-6.2.9/sound/pci/hda/patch_cs8409.c
linux-6.2.9/sound/pci/hda/patch_cs8409.h
linux-6.2.9/sound/pci/hda/patch_hdmi.c
linux-6.2.9/sound/pci/hda/patch_realtek.c
linux-6.2.9/sound/pci/hda/patch_si3054.c
linux-6.2.9/sound/pci/hda/patch_sigmatel.c
linux-6.2.9/sound/pci/hda/patch_via.c
linux-6.2.9/sound/pci/hda/thinkpad_helper.c
Kernel version later than implemented version - there may be build problems
patching file patch_cs8409.c
Hunk #1 succeeded at 1444 (offset 174 lines).
Hunk #2 succeeded at 1453 (offset 174 lines).
Hunk #3 succeeded at 1482 (offset 174 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 303 (offset 1 line).
Hunk #3 succeeded at 330 (offset 1 line).
Hunk #4 succeeded at 366 (offset 1 line).
patching file patch_cirrus_apple.h
make -C /lib/modules/6.2.9-300.fc38.x86_64/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/vascobrown/Downloads/snd_hda_macbookpro/build/hda-6.2.9 modules
Makefile:752: include/config/auto.conf: No such file or directory
make: *** [Makefile:23: all] Error 2
```

Is there a step I might of missed, or maybe grabbed the wrong kernel source files?

davidjo, 2023-10-15:
Unfortunately building kernels is complicated - just downloading the source is not enough - configuration values need setting - and you need to build the full kernel before you can build modules for it.
(auto.conf is a file created by the configuration setting process)

Because of this most distributions package kernels using their packaging system (rather than using kernel.org).
You did download the distribution kernel headers for 6.2.9-300.fc38.x86_64?
This should contain the data needed to build modules (you dont actually need the kernel source if you have the kernel headers package as it should be pre-packaged for building modules - except that my module DOES need a kernel source to copy some files from it).
What Ive been told is the packaging command in Fedora is dnf (see install.cirrus.driver.sh - echo "Fedora: sudo dnf install kernel-headers") - there must be some way to list the packages that can be installed (rather than those you actually have installed).

vascobrown, 2023-10-15:
Thank you so much, audio output is working now! Still no audio input, but I'll take what I can get for now. Your last response put me on the right track to figuring out what I was missing. Here's what I did for the benefit of anyone who is running the same hardware...

Based on your explanation, I started by deleting the folder `/usr/src/kernels/6.2.9` that I had manually downloaded and copied from kernel.org.

I then did some more research and noticed that although `uname -r` is showing `6.2.9-300.fc38.x86_64`, it appears that when the package `kernel-devel` was installed, a folder  `6.5.6-300.fc38.x86_64` was created in `/usr/src/kernels/`. Since that did no match my kernel version, I uninstalled `kernel-devel` and ran `dnf install kernel-devel-6.2.9` which created a `6.2.9-300.fc38.x86_64` folder in `/usr/src/kernels`. So far so good.

Next, in regards to the kernel headers, I ran `sudo dnf install kernel-headers.i686` and got the output:

```
sudo dnf install kernel-headers.i686
Last metadata expiration check: 3:24:08 ago on Sun 15 Oct 2023 01:26:46 PM EDT.
Dependencies resolved.
================================================================================================================================================================================
Package                                       Architecture                        Version                                          Repository                             Size
================================================================================================================================================================================
Installing:
kernel-headers                                i686                                6.2.6-300.fc38                                   fedora                                1.5 M

Transaction Summary
================================================================================================================================================================================
Install  1 Package

Total download size: 1.5 M
Installed size: 5.9 M
Is this ok [y/N]: y
Downloading Packages:
kernel-headers-6.2.6-300.fc38.i686.rpm                                                                                                          466 kB/s | 1.5 MB     00:03    
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Total                                                                                                                                           386 kB/s | 1.5 MB     00:04     
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                                                                        1/1 
  Installing       : kernel-headers-6.2.6-300.fc38.i686                                                                                                                     1/1 
  Verifying        : kernel-headers-6.2.6-300.fc38.i686                                                                                                                     1/1 

Installed:
  kernel-headers-6.2.6-300.fc38.i686                                                                                                                                            

Complete!

```
Then I ran `dnf list --installed | grep kernel` and confirmed that all the kernel related packages were matching my kernel version.

```
abrt-addon-kerneloops.x86_64          2.16.1-1.fc38     @anaconda
kernel.x86_64                         6.2.9-300.fc38    @anaconda
kernel-core.x86_64                    6.2.9-300.fc38    @anaconda
kernel-devel.x86_64                   6.2.9-300.fc38    @fedora  
kernel-headers.i686                   6.2.6-300.fc38    @fedora  
kernel-modules.x86_64                 6.2.9-300.fc38    @anaconda
kernel-modules-core.x86_64            6.2.9-300.fc38    @anaconda
kernel-modules-extra.x86_64           6.2.9-300.fc38    @anaconda
libreport-plugin-kerneloops.x86_64    2.17.9-1.fc38     @anaconda
```                                       

I then fixed the symlinks in `/lib/modules/6.2.9-300.fc38.x86_64` to point to the correct kernel source folder by running `unlink build` and `sudo unlink source`, then `sudo ln -s /usr/src/kernels/6.2.9-300.fc38.x86_64 /lib/modules/6.2.9-300.fc38.x86_64/build` and `sudo ln -s /usr/src/kernels/6.2.9-300.fc38.x86_64 /lib/modules/6.2.9-300.fc38.x86_64/source`. 

After that, I re-ran the install script, this was the output:

```
Ensure the patch package is installed
dnf install wget make gcc kernel-devel patch
--2023-10-15 18:25:00--  https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.2.9.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 146.75.93.176, 2a04:4e42:87::432
Connecting to cdn.kernel.org (cdn.kernel.org)|146.75.93.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

[... 72 lines cut ...]
  CC [M]  /home/vascobrown/Downloads/snd_hda_macbookpro/build/hda-6.2.9/patch_cs8409-tables.o
  LD [M]  /home/vascobrown/Downloads/snd_hda_macbookpro/build/hda-6.2.9/snd-hda-codec-cs8409.o
  MODPOST /home/vascobrown/Downloads/snd_hda_macbookpro/build/hda-6.2.9/Module.symvers
[... cut ...]  play an audio/video source like YouTube or a video file from my HD, I can hear output on both external speakers. I'll need to find a multi-channel audio file and see if is playing both the left channel to both speakers.

[image]

Regarding audio input, I am unable to record anything when testing with the apps Tenacity and Ocenaudio, despite both (and the Settings app) seeing that there is a mic installed.

Once again, I greatly appreciate your help on this and writing this module to begin with.

davidjo, 2023-10-16:
It curious the header file is .i686 rather than . x86_64  .i686 is 32 bit, x86_64 is 64 bit - but it could be the headers work both for 32 and 64 bit - as it worked for you.

Mike input is not fully implemented - you do need to manually set the input in Settings.
Are you talking about the internal mike or an external mike?
The internal mike is at a very low volume even when working so you do need to amplify it using an audio pulse/pipewire module (or even alsa module) - my testing was using Audacity when I could see if I made loud noises the volume meters did respond - and in Audacity you can apply an amplifier.

So USB speakers and bluetooth headsets/headphones have nothing to do with this module - this is purely for the internal mike, internal speakers and the audio 3.5mm jack - are your external speakers plugged into the audio jack?
Its not really something Ive tested as I dont have any external speakers.

jpdonnell, 2024-03-17:
This driver needs added to 6.5 kernel  upstream. 

Out of box drivers are absolute headaches. 

Please make it happen.