davidjo/snd_hda_macbookpro #107: `warning: assignment discards ‘const’ qualifier from pointer target type`

osalbahr, 2023-11-22:
Hello, I noticed the following when upgrading my system:

```
In file included from /var/lib/dkms/snd-hda-macbookpro/0.1/build/build/hda/patch_cs8409.c:1488:
/var/lib/dkms/snd-hda-macbookpro/0.1/build/build/hda/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/var/lib/dkms/snd-hda-macbookpro/0.1/build/build/hda/patch_cirrus_apple.h:1193:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1193 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
At main.c:167:
```
Just wanted to double-check if this warning is expected.

davidjo, 2023-11-22:
yes this is an expected error - I seem to remember at the time I couldnt figure a way to remove this warning.

davidjo, 2024-01-24:
Added fix for this.

closed 2024-01-24 by davidjo