davidjo/snd_hda_macbookpro #112: How do I completely uninstall this driver?

radbirb, 2023-12-02:


davidjo, 2023-12-04:
the name is snd-hda-codec-cs8409.ko - because some Dell laptops apparently now also use the cs8409 HDA chip - but  different from Apples usage of the cs8409 chip.
(which is why my module goes to updates because for ubuntu a module of the same name in updates overrides a system provided module of that name).
oh - should have said to run sudo depmod -A - doesnt remove files but does update some table which stores which modules to load - should ensure the original module is being used - but will need a reboot after removing modules

radbirb, 2024-01-18:
hey, that seems to be the case!
I've decided to go back to this driver and it's been working mostly fine, moving aside the audio jack issue.

I'm also not sure if I should open a new issue for this but the speakers/mic seemingly disconnect entirely when I connect my monitor, and it quickly goes to only show my monitor as the only available audio source, disconnecting my monitor seems to fix the issue but I'd like to see if there's a way I could at least keep the microphone working with a monitor connected, though I should note, it seems like Alsamixer does see .

Another question I have is if it's possible to entirely disable the audio jack? Sorry to bother ,:l