davidjo/snd_hda_macbookpro #114: I don't know how to work with mint cinnamon using this patch

ol0w0lo, 2023-12-30:
sudo ./install.cirrus.driver.sh 
linux-source-5.15.0/sound/pci/hda/
linux-source-5.15.0/sound/pci/hda/hda_intel.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.c
linux-source-5.15.0/sound/pci/hda/patch_ca0132.c
linux-source-5.15.0/sound/pci/hda/patch_cmedia.c
linux-source-5.15.0/sound/pci/hda/hda_controller.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-5.15.0/sound/pci/hda/hda_component.h
linux-source-5.15.0/sound/pci/hda/patch_analog.c
linux-source-5.15.0/sound/pci/hda/patch_conexant.c
linux-source-5.15.0/sound/pci/hda/ca0132_regs.h
linux-source-5.15.0/sound/pci/hda/hda_generic.c
linux-source-5.15.0/sound/pci/hda/patch_via.c
linux-source-5.15.0/sound/pci/hda/hda_local.h
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.h
linux-source-5.15.0/sound/pci/hda/hda_jack.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda.c
linux-source-5.15.0/sound/pci/hda/hda_codec.c
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.h
linux-source-5.15.0/sound/pci/hda/hda_sysfs.c
linux-source-5.15.0/sound/pci/hda/hda_proc.c
linux-source-5.15.0/sound/pci/hda/hda_intel_trace.h
linux-source-5.15.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-5.15.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-5.15.0/sound/pci/hda/hp_x360_helper.c
linux-source-5.15.0/sound/pci/hda/hda_tegra.c
linux-source-5.15.0/sound/pci/hda/hda_controller_trace.h
linux-source-5.15.0/sound/pci/hda/hda_auto_parser.c
linux-source-5.15.0/sound/pci/hda/patch_ca0110.c
linux-source-5.15.0/sound/pci/hda/Kconfig
linux-source-5.15.0/sound/pci/hda/patch_realtek.c
linux-source-5.15.0/sound/pci/hda/hda_intel.h
linux-source-5.15.0/sound/pci/hda/patch_si3054.c
linux-source-5.15.0/sound/pci/hda/hda_eld.c
linux-source-5.15.0/sound/pci/hda/Makefile
linux-source-5.15.0/sound/pci/hda/hda_bind.c
linux-source-5.15.0/sound/pci/hda/hda_controller.h
linux-source-5.15.0/sound/pci/hda/hda_jack.h
linux-source-5.15.0/sound/pci/hda/hda_hwdep.c
linux-source-5.15.0/sound/pci/hda/hda_beep.c
linux-source-5.15.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-5.15.0/sound/pci/hda/patch_sigmatel.c
linux-source-5.15.0/sound/pci/hda/thinkpad_helper.c
linux-source-5.15.0/sound/pci/hda/patch_hdmi.c
linux-source-5.15.0/sound/pci/hda/patch_cs8409.h
linux-source-5.15.0/sound/pci/hda/hda_beep.h
linux-source-5.15.0/sound/pci/hda/patch_cirrus.c
linux-source-5.15.0/sound/pci/hda/hda_generic.h
Kernel version later than implemented version - there may be build problems
patching file patch_cs8409.c
Hunk #1 succeeded at 1442 (offset 172 lines).
Hunk #2 succeeded at 1451 (offset 172 lines).
Hunk #3 succeeded at 1480 (offset 172 lines).
patching file patch_cs8409.h
patching file patch_cirrus_apple.h
make -C /lib/modules/5.15.0-91-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/olowolo/snd_hda_macbookpro/build/hda modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-91-generic'
  CC [M]  /home/olowolo/snd_hda_macbookpro/build/hda/patch_cs8409.o
In file included from /home/olowolo/snd_hda_macbookpro/build/hda/patch_cs8409.c:1486:
/home/olowolo/snd_hda_macbookpro/build/hda/patch_cirrus_apple.h: In function ‘cs_8409_apple_boot_init’:
/home/olowolo/snd_hda_macbookpro/build/hda/patch_cirrus_apple.h:1193:15: warning: assignment discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1193 |         hinfo = spec->gen.stream_analog_playback;
      |               ^
  CC [M]  /home/olowolo/snd_hda_macbookpro/build/hda/patch_cs8409-tables.o
  LD [M]  /home/olowolo/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.o
  MODPOST /home/olowolo/snd_hda_macbookpro/build/hda/Module.symvers
  CC [M]  /home/olowolo/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/olowolo/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.ko
  BTF [M] /home/olowolo/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.ko
Skipping BTF generation for /home/olowolo/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-91-generic'
make INSTALL_MOD_DIR=updates -C /lib/modules/5.15.0-91-generic/build M=/home/olowolo/snd_hda_macbookpro/build/hda CONFIG_MODULE_SIG_ALL=n modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-91-generic'
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
  INSTALL /lib/modules/5.15.0-91-generic/updates/snd-hda-codec-cs8409.ko
  DEPMOD  /lib/modules/5.15.0-91-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-91-generic'
depmod -a

contents of /lib/modules/5.15.0-91-generic/updates
total 1512
-rw-r--r-- 1 root root 1546128 Dec 30 02:29 snd-hda-codec-cs8409.ko

ol0w0lo, 2023-12-30:
Happy new year!

ol0w0lo, 2023-12-30:
nvm somehow it worked!

davidjo, 2023-12-31:
The last few lines starting at contents of ...
show that the updated module does exist in the kernel updates directory - so your build was successful.

In general its best to reboot after doing this install to ensure the new module is used.