davidjo/snd_hda_macbookpro #115: Sound doesn't work when heaphones are plugged in on startup

Adamillo, 2024-01-13:
This is a weird one. When I plug my headphones in while starting up Linux, the audio stops working entirely, even after the headphone was removed from the plug. The only way to fix this is to restart Linux and making sure that no headphones are plugged in during startup. The hardware I have is a 2017 iMac 21,5 inch with a Cirrus Logic CS8409 (AB 14) sound chip, and the distro I'm using is Linux Mint Cinnamon 21.2

davidjo, 2024-01-14:
Sorry - this is a known issue - I havent sorted out booting with headphones in - requires a different code path on boot - think I made some attempts but obviously not fixed yet

davidjo, 2024-02-02:
I have just added commits which may have fixed booting with headphones plugged in.

Adamillo, 2024-02-02:
I should just use the sh installer in order to install the newest version of the driver right?

davidjo, 2024-02-02:
yes - git pull (or reclone) then rerun the install script - can remove build/hda eg sudo rm -fr build/hda  before running script to make sure old build directory gone.

Adamillo, 2024-02-03:
Yeah this only worked once and then it goes back to it's usual not working state when rebooting

davidjo, 2024-02-03:
annoying - so need more details.
Are you using a headphone (no mike) or headset (with mike)?
if using a headset are you using an Apple one or non-Apple one?
The jack code on OSX detects these 3 different styles - I actually just checked things using a headphone so maybe  a headset needs more fixing.
The other problem could be its an imac vs macbook issue - these are different (believe it or not theres a polarity reversal) and fixes already exist for this but maybe something needs to be done at the boot point.
Just a thought - when you say plug in while starting Linux has the linux boot already started or not?
Im using refind so my checking was done plugging in at the refind boot screen ie before any linux code is being executed - there may well be issues if you plugin during boot code execution.
Ill probably need some logging to get further.

Adamillo, 2024-02-03:
> annoying - so need more details. Are you using a headphone (no mike) or headset (with mike)? if using a headset are you using an Apple one or non-Apple one? The jack code on OSX detects these 3 [...]

I'm using a non Apple headset, and the headset are always plugged even before starting Linux Mint. If it makes a difference, I'm using Limine for booting Linux Mint specifically instead of Grub. Everything else is stock, even the boot options are stock (holding the options aka alt key to choose an operating system). If it also matters, I'm using a triple boot set up (meaning I have macOS, Windows and Linux)

davidjo, 2024-02-03:
So we need some logging - you need to add -DMYSOUNDDEBUG to the beginning of the string assigned to KBUILD_EXTRA_CFLAGS in the Makefile.
Then DONT rerun the install script (which would overwrite the Makefile), just do make clean, make then sudo make install.
If you have permission problems with files for make clean or plain make then change the ownership of everything under build to your normal user.
Reboot to ensure new module version used.
Now you should be able to see lines in the log file with snd_hda... eg snd_hda_intel or snd_hda_codec_cs8409.
Under Ubuntu there is still the base syslog file under /var/log/syslog (however some linux distributions no longer have it).
You can then do eg tail -f /var/log/syslog - which will show the log lines as they appear - in particular if try an unplug or plugin.
It also will show commands as audio starts to play and stops playing - even just using the change volume buttons.
For the boot stage I just grep for snd_hda - but just from the last boot time (syslog at least under Ubuntu saves over multiple reboots).

Adamillo, 2024-02-04:
Did I add the string correctly like this? 
[image]
Sorry if this might take a while, I wanna be sure that I'm doing things right and not like brick my Linux set up like I did more than once and had to ask help from someone else. Bare in mind that I'm pretty new to Linux so I hope you're not too annoyed when I ask multiple questions before I do things on Linux

davidjo, 2024-02-04:
Oops - not quite - I was meaning KBUILD_EXTRA_CFLAGS = "-DMYSOUNDDEBUG -DAPPLE.....".
Things like -DNAME are what are called CFLAGS or flags to the C compiler - KBUILD_EXTRA_CFLAGS is a specific variable for kernel Makefiles (as created by linux kernel developers) for adding extra CFLAGS.

However - I have tried my Apple headset and think Ive found your issue - I needed some more interrupt waiting (it appears that at this early stage the unsolicited response is not working (unsol responses are something like interrupts) - so I need to manually wait for the interrupt flag - this was needed for the headset/headphone detect - now for a headset also need to wait for the button detect.

Dont be afraid to ask questions - a lot of this is new to me eg just exactly how unsol responses are working at this stage - I just have some experience in how to ask the right questions and where to look for answers.

By tomorrow should have a new version ready to push to the main repo.

But its good you can do this because its still possible there are iMac specific issues.
Plus I have to say the implementation is not robust yet - I observed some lock up issues because sometimes I didnt wait long enough before eg doing an unplug - I dont think it likes it if you do an unplug while the PCM stream is still open - so waiting for 3 seconds or so after last sound so the PCM stream is closed seems to be required. WIth the logging you can see when the stream is closed and the HDA goes into suspend mode.

davidjo, 2024-02-05:
New version uploaded which may work for you.

However fixing the robustness is going take a while - in particular the 2.5 sec delayed close stream is going to make overlapping actions more likely to occur and need a totally new fix.

davidjo, 2024-02-06:
OK so you are going to need to get the logging working.
It works for me plus someone else in another issue #109 has reported it works for them (another macbook version).
So either its the non-Apple headset or the iMac.
Can you try with an Apple headset - or even a headphone?

Adamillo, 2024-03-03:
I don't have an apple headset anymore. Your new version however fixed the problem with headphones not outputting sound on boot. Though I do notice a new bug with the newest version of your driver now, no idea what caused it to change. There's some sort of weird pattern that happens. First boot with headphone works with no issues, then second boot is where the sounds get distorted. Then when I reboot it goes back to normal, rinse and repeat.

davidjo, 2024-03-04:
Without getting logs cant say whats going on - as you say its likely it still thinks the headphones are plugged in - which suggests its not setting up the unplug actions right - but again it could be because its a non-Apple headset - would be really good if you could try with a pure headphone ie no mike as the mike adds a whole other set of complications.