davidjo/snd_hda_macbookpro #116: zorin os error building

Om0019, 2024-01-22:
These is what is presented to me after running. I only get this when running it in zorin OS. in Ubuntu it works perfectly fine.

linux-6.5/sound/pci/hda/
linux-6.5/sound/pci/hda/Kconfig
linux-6.5/sound/pci/hda/Makefile
linux-6.5/sound/pci/hda/ca0132_regs.h
linux-6.5/sound/pci/hda/cs35l41_hda.c
linux-6.5/sound/pci/hda/cs35l41_hda.h
linux-6.5/sound/pci/hda/cs35l41_hda_i2c.c
linux-6.5/sound/pci/hda/cs35l41_hda_spi.c
linux-6.5/sound/pci/hda/hda_auto_parser.c
linux-6.5/sound/pci/hda/hda_auto_parser.h
linux-6.5/sound/pci/hda/hda_beep.c
linux-6.5/sound/pci/hda/hda_beep.h
linux-6.5/sound/pci/hda/hda_bind.c
linux-6.5/sound/pci/hda/hda_codec.c
linux-6.5/sound/pci/hda/hda_component.h
linux-6.5/sound/pci/hda/hda_controller.c
linux-6.5/sound/pci/hda/hda_controller.h
linux-6.5/sound/pci/hda/hda_controller_trace.h
linux-6.5/sound/pci/hda/hda_cs_dsp_ctl.c
linux-6.5/sound/pci/hda/hda_cs_dsp_ctl.h
linux-6.5/sound/pci/hda/hda_eld.c
linux-6.5/sound/pci/hda/hda_generic.c
linux-6.5/sound/pci/hda/hda_generic.h
linux-6.5/sound/pci/hda/hda_hwdep.c
linux-6.5/sound/pci/hda/hda_intel.c
linux-6.5/sound/pci/hda/hda_intel.h
linux-6.5/sound/pci/hda/hda_intel_trace.h
linux-6.5/sound/pci/hda/hda_jack.c
linux-6.5/sound/pci/hda/hda_jack.h
linux-6.5/sound/pci/hda/hda_local.h
linux-6.5/sound/pci/hda/hda_proc.c
linux-6.5/sound/pci/hda/hda_sysfs.c
linux-6.5/sound/pci/hda/hda_tegra.c
linux-6.5/sound/pci/hda/hp_x360_helper.c
linux-6.5/sound/pci/hda/ideapad_s740_helper.c
linux-6.5/sound/pci/hda/patch_analog.c
linux-6.5/sound/pci/hda/patch_ca0110.c
linux-6.5/sound/pci/hda/patch_ca0132.c
linux-6.5/sound/pci/hda/patch_cirrus.c
linux-6.5/sound/pci/hda/patch_cmedia.c
linux-6.5/sound/pci/hda/patch_conexant.c
linux-6.5/sound/pci/hda/patch_cs8409-tables.c
linux-6.5/sound/pci/hda/patch_cs8409.c
linux-6.5/sound/pci/hda/patch_cs8409.h
linux-6.5/sound/pci/hda/patch_hdmi.c
linux-6.5/sound/pci/hda/patch_realtek.c
linux-6.5/sound/pci/hda/patch_si3054.c
linux-6.5/sound/pci/hda/patch_sigmatel.c
linux-6.5/sound/pci/hda/patch_via.c
linux-6.5/sound/pci/hda/thinkpad_helper.c
Kernel version later than implemented version - there may be build problems
patching file patch_cs8409.c
Hunk #1 succeeded at 1444 (offset 174 lines).
Hunk #2 succeeded at 1453 (offset 174 lines).
Hunk #3 succeeded at 1482 (offset 174 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 303 (offset 1 line).
Hunk #3 succeeded at 330 (offset 1 line).
Hunk #4 succeeded at 366 (offset 1 line).
patching file patch_cirrus_apple.h
make -C /lib/modules/6.5.0-14-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/orlando/Downloads/snd_hda_macbookpro-master/build/hda modules
make[1]: Entering directory '/usr/src/linux-headers-6.5.0-14-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           
  CC [M]  /home/orlando/Downloads/snd_hda_macbookpro-master/build/hda/patch_cs8409.o
/bin/sh: 1: gcc-12: not found
make[3]: *** [scripts/Makefile.build:251: /home/orlando/Downloads/snd_hda_macbookpro-master/build/hda/patch_cs8409.o] Error 127
make[2]: *** [/usr/src/linux-headers-6.5.0-14-generic/Makefile:2037: /home/orlando/Downloads/snd_hda_macbookpro-master/build/hda] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.5.0-14-generic'
make: *** [Makefile:16: all] Error 2

davidjo, 2024-01-22:
I think this is the issue
/bin/sh: 1: gcc-12: not found
looks like it cant find a C compiler on Zorin - need to check your build system on Zorin

Om0019, 2024-01-22:
Manually installing gcc12 rebooting and trying to build again worked. Thank you.

closed 2024-01-24 by Om0019