davidjo/snd_hda_macbookpro #118: DKMS issue when updating Kernel on Zorin 17 (Ubuntu 22.04) with Macbookpro 14,1

BrewCoffeeAddict, 2024-02-11:
Dear all,

First thanks a lot for providing these amazing drivers. Without these, I would not have sound when using my Macbookpro 14,1 with Linux.
I used to manually reinstall the drivers after each kernel update but wanted to try DKMS for automatic update since you added support for it.
I installed DKMS for these drivers by running the following:
````
1) Installed DKMS following instructions: https://github.com/dell/dkms
2) git clone https://github.com/davidjo/snd_hda_macbookpro
3) cd snd_hda_macbookpro
4) chmod +x ./dkms.sh
5) sudo ./dkms.sh
````

The setup was successful.
But when installing a new kernel, the DKMS update failed and as a result my kernel update failed too. Please find below kernel update output:
````
sudo apt install linux-image-6.5.0-17-generic linux-headers-6.5.0-17-generic linux-modules-6.5.0-17-generic linux-modules-extra-6.5.0-17-generic
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
linux-headers-6.5.0-17-generic is already the newest version (6.5.0-17.17~22.04.1).
linux-image-6.5.0-17-generic is already the newest version (6.5.0-17.17~22.04.1).
linux-modules-6.5.0-17-generic is already the newest version (6.5.0-17.17~22.04.1).
linux-modules-extra-6.5.0-17-generic is already the newest version (6.5.0-17.17~22.04.1).
The following package was automatically installed and is no longer required:
  libwmf0.2-7-gtk
Use 'sudo apt autoremove' to remove it.
0 to upgrade, 0 to newly install, 0 to remove and 0 not to upgrade.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
Setting up linux-image-6.5.0-17-generic (6.5.0-17.17~22.04.1) ...
Processing triggers for linux-image-6.5.0-17-generic (6.5.0-17.17~22.04.1) ...
/etc/kernel/postinst.d/dkms:
 * dkms: running auto installation service for kernel 6.5.0-17-generic
Deprecated feature: MODULES_CONF (/var/lib/dkms/facetimehd/0.1/source/dkms.conf)
Deprecated feature: MODULES_CONF (/var/lib/dkms/facetimehd/0.1/source/dkms.conf)

snd-hda-codec-cs8409.ko:
Running module version sanity check.
Module version  for snd-hda-codec-cs8409.ko
exactly matches what is already found in kernel 6.5.0-17-generic.
DKMS will not replace this module.
You may override by specifying --force.
Error! Installation aborted.
dkms autoinstall on 6.5.0-17-generic/x86_64 succeeded for facetimehd
dkms autoinstall on 6.5.0-17-generic/x86_64 failed for snd_hda_macbookpro(6)
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
 * dkms: autoinstall for kernel 6.5.0-17-generic
   ...fail!
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-6.5.0-17-generic (--configure):
 installed linux-image-6.5.0-17-generic package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 linux-image-6.5.0-17-generic
E: Sub-process /usr/bin/dpkg returned an error code (1)
````

Do you have any idea what happened and how I could resolve this?
Thank you.

davidjo, 2024-02-12:
I dont actually use dkms (Im always fixing something so dkms isnt that useful) - and I did not do this setup.
What it looks to me is that its not actually creating a new module - Im looking at the line
exactly matches what is already found in kernel 6.5.0-17-generic.
I think should not run dkms.sh  - you need to run the install.cirrus.driver.sh script but give it the -i (or --install) option (as sudo).

BrewCoffeeAddict, 2024-02-12:
> I dont actually use dkms (Im always fixing something so dkms isnt that useful) - and I did not do this setup. What it looks to me is that its not actually creating a new module - Im looking at the line exactly matches what is already found in kernel 6.5.0-17-generic. I think should not run dkms.sh directly - you need to run the install.cirrus.driver.sh script but give it the -i (or --install) option (as sudo). This runs the dkms.sh script (see actions for dkms_action shell variable).

Ah thanks, this is really useful to know. I completely missed the "-i" flag in the install.cirrus.driver.sh.
I ran the script with "-i" and reinstalled the kernel 6.5.0-17-generic, it seems to have successfully worked this time.
I guess, I will really find out at the next kernel update ;-)

BrewCoffeeAddict, 2024-02-17:
Hello @davidjo,

I just installed the latest Kernel available for my distro 6.5.0-18-generic but sadly the installation failed once again with DKMS:

```
Running the pre_build script:
--2024-02-17 15:08:24--  https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.5.0.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 146.75.73.176, 2a04:4e42:82::432
Connecting to cdn.kernel.org (cdn.kernel.org)|146.75.73.176|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-02-17 15:08:24 ERROR 404: Not Found.
Failed to download linux-6.5.0.tar.xz
Trying to download base kernel version linux-6.5.tar.xz
This may lead to build failures as too old
If this is an Ubuntu-based distribution this almost certainly will fail to build

--2024-02-17 15:08:24--  https://cdn.kernel.org/pub/linux/kernel/v6.x/linux-6.5.tar.xz
Resolving cdn.kernel.org (cdn.kernel.org)... 146.75.73.176, 2a04:4e42:82::432
Connecting to cdn.kernel.org (cdn.kernel.org)|146.75.73.176|:443... connected.
HTTP request sent, awaiting response... 416 Range Not Satisfiable

    The file is already fully retrieved; nothing to do.

linux-6.5/sound/pci/hda/
linux-6.5/sound/pci/hda/Kconfig
linux-6.5/sound/pci/hda/Makefile
linux-6.5/sound/pci/hda/ca0132_regs.h
linux-6.5/sound/pci/hda/cs35l41_hda.c
linux-6.5/sound/pci/hda/cs35l41_hda.h
linux-6.5/sound/pci/hda/cs35l41_hda_i2c.c
linux-6.5/sound/pci/hda/cs35l41_hda_spi.c
linux-6.5/sound/pci/hda/hda_auto_parser.c
linux-6.5/sound/pci/hda/hda_auto_parser.h
linux-6.5/sound/pci/hda/hda_beep.c
linux-6.5/sound/pci/hda/hda_beep.h
linux-6.5/sound/pci/hda/hda_bind.c
linux-6.5/sound/pci/hda/hda_codec.c
linux-6.5/sound/pci/hda/hda_component.h
linux-6.5/sound/pci/hda/hda_controller.c
linux-6.5/sound/pci/hda/hda_controller.h
linux-6.5/sound/pci/hda/hda_controller_trace.h
linux-6.5/sound/pci/hda/hda_cs_dsp_ctl.c
linux-6.5/sound/pci/hda/hda_cs_dsp_ctl.h
linux-6.5/sound/pci/hda/hda_eld.c
linux-6.5/sound/pci/hda/hda_generic.c
linux-6.5/sound/pci/hda/hda_generic.h
linux-6.5/sound/pci/hda/hda_hwdep.c
linux-6.5/sound/pci/hda/hda_intel.c
linux-6.5/sound/pci/hda/hda_intel.h
linux-6.5/sound/pci/hda/hda_intel_trace.h
linux-6.5/sound/pci/hda/hda_jack.c
linux-6.5/sound/pci/hda/hda_jack.h
linux-6.5/sound/pci/hda/hda_local.h
linux-6.5/sound/pci/hda/hda_proc.c
linux-6.5/sound/pci/hda/hda_sysfs.c
linux-6.5/sound/pci/hda/hda_tegra.c
linux-6.5/sound/pci/hda/hp_x360_helper.c
linux-6.5/sound/pci/hda/ideapad_s740_helper.c
linux-6.5/sound/pci/hda/patch_analog.c
linux-6.5/sound/pci/hda/patch_ca0110.c
linux-6.5/sound/pci/hda/patch_ca0132.c
linux-6.5/sound/pci/hda/patch_cirrus.c
linux-6.5/sound/pci/hda/patch_cmedia.c
linux-6.5/sound/pci/hda/patch_conexant.c
linux-6.5/sound/pci/hda/patch_cs8409-tables.c
linux-6.5/sound/pci/hda/patch_cs8409.c
linux-6.5/sound/pci/hda/patch_cs8409.h
linux-6.5/sound/pci/hda/patch_hdmi.c
linux-6.5/sound/pci/hda/patch_realtek.c
linux-6.5/sound/pci/hda/patch_si3054.c
[... 48 lines cut ...]
snd-hda-codec-cs8409.ko:
Running module version sanity check.
Module version  for snd-hda-codec-cs8409.ko
exactly matches what is already found in kernel 6.5.0-18-generic.
DKMS will not replace this module.
You may override by specifying --force.
Error! Installation aborted.
dkms autoinstall on 6.5.0-18-generic/x86_64 succeeded for facetimehd
dkms autoinstall on 6.5.0-18-generic/x86_64 failed for snd_hda_macbookpro(6)
 * dkms: autoinstall for kernel 6.5.0-18-generic
   ...fail!
Error! One or more modules failed to install during autoinstall.
Refer to previous errors for more information.
run-parts: /etc/kernel/postinst.d/dkms exited with return code 11
dpkg: error processing package linux-image-6.5.0-18-generic (--configure):
 installed linux-image-6.5.0-18-generic package post-installation script subprocess returned error exit status 1
Processing triggers for initramfs-tools (0.140ubuntu13.4) ...
update-initramfs: Generating /boot/initrd.img-6.5.0-17-generic
I: The initramfs will attempt to resume from /dev/nvme0n1p4
I: (UUID=5fb7b4e2-cc75-412a-a9b3-65626ac65672)
I: Set the RESUME variable to override this.
Errors were encountered while processing:
 linux-image-6.5.0-18-generic
Log ended: 2024-02-17  15:09:05
```

davidjo, 2024-02-18:
Not sure I can be much help - as I said I dont use dkms for this.
What Ive worries me is this:
contents of /lib/modules/6.5.0-18-generic/updates
total 2468
drwxr-xr-x 2 root root    4096 Feb 17 15:08 dkms
-rw-r--r-- 1 root root 2522232 Feb 17 15:08 snd-hda-codec-cs8409.ko
I believe dkms installs under updates/dkms - so check if there is something under updates/dkms - Im thinking somehow there
are 2 identical modules - the one listed above and one under the updates/dkms - which is causing the issue.
What I have no idea is how the updates/snd-hda-codec-cs8409.ko was installed - which it appears to have been because of the file date - you didnt re-run the install script manually at some point?
Try removing the updates/snd-hda-codec-cs8409.ko module (redo sudo depmod -a) and reboot - at least if theres a version under updates/dkms.

BrewCoffeeAddict, 2024-02-18:
Thanks for the suggestion @davidjo and I appreciate that you don't use dkms. I think I might know what is happening. 
When you run the script as below the following is happening:
` sudo ./install.cirrus.driver.sh -i`

1. the script goes there into install: https://github.com/davidjo/snd_hda_macbookpro/blob/master/install.cirrus.driver.sh#L39
2. Then it goes into the dkms script: https://github.com/davidjo/snd_hda_macbookpro/blob/master/dkms.sh#L26
3. Then this run the dkms program using the configuration file dkms.conf
4. Inside the dkms.conf, you can see the pre_build parameter here: https://github.com/davidjo/snd_hda_macbookpro/blob/master/dkms.conf#L3, this runs the install.cirrus.driver.sh as usual with the -k parameter.
`install.cirrus.driver.sh -k $kernelver`
6. This then install snd-hda-codec-cs8409.ko under here: /lib/modules/6.5.0-18-generic/updates as seen here: https://github.com/davidjo/snd_hda_macbookpro/blob/master/Makefile#L22
7. Finally dkms runs and install snd-hda-codec-cs8409.ko under /lib/modules/6.5.0-18-generic/updates/dkms/

I believe this is why there are two identical modules.
Now, when I run this manually, I don't get any failures but when the kernel gets updated and dkms runs with autoinstall, then it somehow fails.

davidjo, 2024-02-20:
My guess is that dkms is seeing the 2 modules which leads to the exactly matches what is already found in kernel 6.5.0-18-generic error which looks like it causes an error.

My suggestion would be to add an input argument eg --dkms and use this to set a shell variable in install.cirrus.driver.sh.
Then I think use this to skip the last code in the script which does the build and install (it appears dkms is doing its own build and install but it should be using the source code under the repo build/hda - check whats under /var/lib/dkms/snd_hda_macbookpro - has the updated source code been copied to there as that seems to be where dkms is building.
Im talking about the whole if block - from the if to the ending fi - starting if [ $PATCH_CIRRUS = true ]; then.
The logs do suggest the source code is being generated under /var/lib/dkms....
Also update the dkms.conf to add this new argument eg --dkms to the PRE_BUILD line which runs the install script.
This is because it looks as though dkms is doing its own build and install ( Cleaning build area... etc).

BrewCoffeeAddict, 2024-02-22:
Dear @davidjo,

Thanks a lot for the instructions. I made your suggestions on my fork and it seems to work fine now.
I reinstalled my current kernel and also re-ran the dkms_autoinstaller, it all worked without problem.
To answer your question, I have the following under /var/lib/dkms/snd_hda_macbookpro:
```
total 8
drwxr-xr-x 4 root root 4096 Feb 22 18:41 0.1
lrwxrwxrwx 1 root root   27 Feb 22 18:41 kernel-6.5.0-17-generic-x86_64 -> 0.1/6.5.0-17-generic/x86_64
lrwxrwxrwx 1 root root   27 Feb 22 18:35 kernel-6.5.0-18-generic-x86_64 -> 0.1/6.5.0-18-generic/x86_64
drwxr-xr-x 4 root root 4096 Feb 22 18:41 original_module
```
I want to wait the next kernel update for my distro to be 100% sure that the dkms auto install will works on new kernel.
If all work as expected, are you happy for me to raise a pull request? If so, would it be ok if I also add a section about dkms in the README?

BrewCoffeeAddict, 2024-02-24:
Great news @davidjo. I just had a kernel update to 6.5.0.21 and dkms auto install worked successfully with your suggested changes. 
Thanks a lot for your help. Also thanks again for making these drivers. You managed to improve them quite a lot recently. Now, I can't notice a difference between my sound quality under Mac OSX and Linux. Now that Apple has dropped future updates of mac OSX for my laptop, I feel that I can migrate to linux.

davidjo, 2024-02-25:
Thats good to know.
Sure - raise a pull request.