davidjo/snd_hda_macbookpro #122: Why isn't this upstream in 6.5?

jpdonnell, 2024-03-16:
Hi

I recently installed Mint 21.1 for a Mac-pro 2021 model and it came with 5.15 . I manually updated to 6.5.x.25 that was rumored to contain this fix for the audio being marked as DUMMY OUTPUT. 

It didn't resolve it .

Manually adding this driver cured my audio setup. 

Thank you. 

Why isn't this driver in 6.5 ?  

I happened  to be able to apply this work . There are less savvy users who will never be able to.

osalbahr, 2024-03-17:
See also:

- https://github.com/Dunedan/mbp-2016-linux?tab=readme-ov-file#audio-input--output
- https://bugzilla.kernel.org/show_bug.cgi?id=195671
- https://bugzilla.kernel.org/show_bug.cgi?id=110561
- http://mailman.alsa-project.org/pipermail/alsa-devel/2018-November/141942.html

And as far as packaging goes:

[image]

jpdonnell, 2024-03-17:
Hello

The progress seems stalled.  

https://bugzilla.kernel.org/show_bug.cgi?id=195671  - last update as 2020

https://bugzilla.kernel.org/show_bug.cgi?id=110561 - last update is 8/2023 

If you have functional patches in the repo  now ,  why haven't they been posted for review  for the current tip of Linux 6.8 ?

Thank you.

jpdonnell, 2024-03-17:
Notes:
See also:

    https://github.com/Dunedan/mbp-2016-linux?tab=readme-ov-file#audio-input--output
    https://bugzilla.kernel.org/show_bug.cgi?id=195671
    https://bugzilla.kernel.org/show_bug.cgi?id=110561
    http://mailman.alsa-project.org/pipermail/alsa-devel/2018-November/141942.html

packaging 

https://repology.org/project/snd-hda-macbookpro-dkms/versions

2023-11-09 01:55 	History start, no known latest or devel versions for this project, present in [AUR](https://repology.org/repository/aur)

jpdonnell, 2024-03-17:
Hi 

As you likely know, you will  continue to have these operational issues on upgrades  on  various distributions until you get this incorporated into  the base driver. 

It appears to me you are taking the existing driver code and patching  it in a out of tree driver for whatever reason..

An alternative ( proper ) method would be to work with the  Cirrus  maintainers in the appropriate kernel.org email group and come up with a solution that produces a " in kernel " snd_hba_macbookpro driver based on your changes here.

The proper way to get this in distribution is to apply the changes at the current tip (6.8 today)  , get it tested in a RC build, then back port the changes to branches you desire ( 5.15, 6.5 ) 

If you don't do this , these changes are going to end up be abandoned and orphaned eventually. 

This type of activity is a cancer that plagues open source software.

Thank you for this driver . 

Bomber.

davidjo, 2024-03-17:
First - I am not a kernel developer and not an audio developer either - I happened to be able to get sound for my 14,3 machine internal speakers working (which is all I wanted  - this was the first Apple laptop since Pismo which didnt have working speaker audio) - so thought I would share it - but if you are not a savvy user you probably shouldnt be using this.

"It appears to me you are taking the existing driver code and patching it in a out of tree driver for whatever reason."

Well I had no idea what I was doing when I started - not sure what you are referring to - there is a generic driver for HDA chips which has specific fixups for specific manufacturers and chips which seemed to be how to add new chips.
(Although because the 8409 is so different from standard HDA chips probably shouldnt have gone that route).
What I discovered is the HDA chip driver has a base module snd_hda_intel (because Intel wrote the HDA specification) which is generic then other kernel modules provide specific fixups - Im still using the base kernel snd_hda_intel module.

The Cirrus 8409 chip had no driver when I started (linux kernel 4.15 - the Dell laptop driver was added at 5.13 - so far my best guess - and this is what everything Ive done is  my best guess - is this is not similar to the Apple laptop system as it doesnt seem to have the separate audio chip drivers - just drives the jack.

You are aware the Apple chips are propietary versions so dont think Cirrus workers would be giving me info on them.
If you can find documentation for the 8409 chip please let me know.
The Dell laptop uses the CS42l42 chip for which there is documentation but Apple uses its own version CS42L83 - no documentation - although can be guessed from the CS42l42 docs - but I have found specific differences.
And again this is just for the jack - its not the internal speaker amp chips.

By the way I dont desire 5.15 or 6.5 - just want it to work with my current kernel - in fact dont understand why Mint is still on 5.15 - I only use Ubuntu.
Another point the HDA chip driver pre 5.13 was actually unable to handle the 8409 as an array size definition was too small so I had to hack a fix up so could still use the distributed snd_hda_intel module.

As you seem to be a kernel developer maybe you can tell me how the linux's mutex locks interact with the HDA unsolicited response.
Whats an RC build?? I have no idea as not a kernel developer.
Primary reason I did out of tree is so DONT have to do full kernel build each time so could test things out quickly.

But the reality is this is old software now - Ill shortly be upgrading to M3 laptops which is whole other complication - never mind Apples locking down of the OS so I couldnt do what I did now - but supposedly audio is working for M1/M2 linux - Asahi linux.

jpdonnell, 2024-03-18:
There are plenty of "proprietary"  hardware supported by the Linux kernel.   You could likely get a programmers guide through Cirrus if you were a employee of a company that works with them and have signed the appropriate legal NDA. 

A "RC"  build is the incremental Linux kernel build  at kernel.org , done usually once a week,  that includes the recent accepted changes on the tip and maintained branches ( 5.15, 4.x, 3.x) . Getting something in a RC build needs a maintainer  and peers to approve changes for the subsystem they represent. You submit changes to kernel.org email lists via a patch. 

Downstream distributions like Ubuntu , Mint, Red-hat, Oracle will pull in kernel,org  updates and churn their own special version  for their implementations.
 
Everything done in Linux is peer reviewed in open forums. 

So it appears this will become another orphaned driver . That is unfortunate. There are millions of Macs out there that could take advantage of your work and Linux. 

 I just happened to have a Mac and I wanted something that wasn't so tied down by Apple. 

I'm retired. 

I don't want to work on my computers anymore. I just want to use them.

davidjo, 2024-03-18:
Well you see I am the same - I dont work any more.

risen, 2024-03-18:
While it would be fantastic to have this upstream, I'm still grateful for your work. It made it possible to have sound on my MacBookPro14,1 (on 6.4 currently). I tend to use https://github.com/egorenar/snd-hda-codec-cs8409 which is easier to build, but AFAICT it was based on this driver.

So, thanks!

osalbahr, 2024-03-20:
> I tend to use https://github.com/egorenar/snd-hda-codec-cs8409 which is easier to build, but AFAICT it was based on this driver.

@risen easier than using [`install.cirrus.driver.sh`](https://github.com/davidjo/snd_hda_macbookpro/blob/master/install.cirrus.driver.sh)?