davidjo/snd_hda_macbookpro #123: microphone in code?

nnylip, 2024-03-19:


nnylip, 2024-03-19:
Is it possible to describe what code, headers etc is specific for the microphone. Am considering adapting  leifliddy/macbook12-audio-driver to handle microphone input. I'm a driver newbie though so the direction would be appreciated.

davidjo, 2024-03-20:
Im not sure the code for handling the internal mike in this repo would be useful for the macbook12 as 2 very different chips are being used, the  8409 in this repo vs the 4208 in leifliddy (the 4208 seems to be some version of Cirrus 4207 chip for which there is a documentation pdf).
If its an external mike then I think its even more unlikely this repo will help as thats is handled by a non-HDA chip in 8409 machines.

For the internal mike look for the intmike string - this is the HDA nid that seems to be handling the internal mike.
If you dont have already you should get the HDA specification document from Intel - just google for it.
You need to have some understanding of how the HDA system overall is designed, what a nid is etc.
(I didnt know anything about HDA systems when I started so this was all new to me as well).
Also you should look at some of the other code in the sound/pci/hda directory in the linux kernel source as this will likely also give more insight as to how to handle things.

nnylip, 2024-03-20:
ok thanks. I am currently using an external usb mic but would prefer to use the internal. thanks for 
the direction