davidjo/snd_hda_macbookpro #124: Error on Imac 19.2

green-51, 2024-05-02:
I got an error on an Imac 19.2 when attempting to build/install the driver. the following is the full output when attempting to run `sudo ./install.cirrus.driver.sh`

<pre>linux-source-6.5.0/sound/pci/hda/
linux-source-6.5.0/sound/pci/hda/hda_jack.h
linux-source-6.5.0/sound/pci/hda/hda_beep.c
linux-source-6.5.0/sound/pci/hda/hda_local.h
linux-source-6.5.0/sound/pci/hda/hda_generic.c
linux-source-6.5.0/sound/pci/hda/ca0132_regs.h
linux-source-6.5.0/sound/pci/hda/hp_x360_helper.c
linux-source-6.5.0/sound/pci/hda/hda_beep.h
linux-source-6.5.0/sound/pci/hda/patch_analog.c
linux-source-6.5.0/sound/pci/hda/cs35l41_hda.c
linux-source-6.5.0/sound/pci/hda/patch_si3054.c
linux-source-6.5.0/sound/pci/hda/hda_tegra.c
linux-source-6.5.0/sound/pci/hda/patch_ca0110.c
linux-source-6.5.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-6.5.0/sound/pci/hda/hda_controller.c
linux-source-6.5.0/sound/pci/hda/patch_conexant.c
linux-source-6.5.0/sound/pci/hda/patch_cs8409.h
linux-source-6.5.0/sound/pci/hda/hda_eld.c
linux-source-6.5.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-6.5.0/sound/pci/hda/patch_ca0132.c
linux-source-6.5.0/sound/pci/hda/hda_controller.h
linux-source-6.5.0/sound/pci/hda/hda_cs_dsp_ctl.c
linux-source-6.5.0/sound/pci/hda/hda_codec.c
linux-source-6.5.0/sound/pci/hda/thinkpad_helper.c
linux-source-6.5.0/sound/pci/hda/hda_generic.h
linux-source-6.5.0/sound/pci/hda/patch_sigmatel.c
linux-source-6.5.0/sound/pci/hda/cs35l41_hda_property.c
linux-source-6.5.0/sound/pci/hda/hda_intel.c
linux-source-6.5.0/sound/pci/hda/hda_auto_parser.c
linux-source-6.5.0/sound/pci/hda/hda_sysfs.c
linux-source-6.5.0/sound/pci/hda/Kconfig
linux-source-6.5.0/sound/pci/hda/hda_intel_trace.h
linux-source-6.5.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-6.5.0/sound/pci/hda/hda_intel.h
linux-source-6.5.0/sound/pci/hda/patch_cirrus.c
linux-source-6.5.0/sound/pci/hda/hda_hwdep.c
linux-source-6.5.0/sound/pci/hda/hda_bind.c
linux-source-6.5.0/sound/pci/hda/hda_jack.c
linux-source-6.5.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-6.5.0/sound/pci/hda/patch_hdmi.c
linux-source-6.5.0/sound/pci/hda/hda_component.h
linux-source-6.5.0/sound/pci/hda/hda_proc.c
linux-source-6.5.0/sound/pci/hda/patch_cmedia.c
linux-source-6.5.0/sound/pci/hda/cs35l41_hda.h
linux-source-6.5.0/sound/pci/hda/Makefile
linux-source-6.5.0/sound/pci/hda/patch_via.c
linux-source-6.5.0/sound/pci/hda/patch_realtek.c
linux-source-6.5.0/sound/pci/hda/hda_controller_trace.h
linux-source-6.5.0/sound/pci/hda/hda_auto_parser.h
linux-source-6.5.0/sound/pci/hda/cs35l41_hda_property.h
linux-source-6.5.0/sound/pci/hda/hda_cs_dsp_ctl.h
linux-source-6.5.0/sound/pci/hda/patch_cs8409.c
Kernel version later than implemented version - there may be build problems
patching file patch_cs8409.c
Hunk #1 succeeded at 1445 (offset 175 lines).
Hunk #2 succeeded at 1454 (offset 175 lines).
Hunk #3 succeeded at 1483 (offset 175 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 304 (offset 2 lines).
Hunk #3 succeeded at 331 (offset 2 lines).
Hunk #4 succeeded at 367 (offset 2 lines).
patching file patch_cirrus_apple.h
make -C /lib/modules/6.5.0-28-generic/build CFLAGS_MODULE=&quot;-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function&quot; M=/home/user/snd_hda_macbookpro/build/hda modules
make[1]: Entering directory &apos;/usr/src/linux-headers-6.5.0-28-generic&apos;
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           
  CC [M]  /home/user/snd_hda_macbookpro/build/hda/patch_cs8409.o
/bin/sh: 1: gcc-12: not found
make[3]: *** [scripts/Makefile.build:251: /home/user/snd_hda_macbookpro/build/hda/patch_cs8409.o] Error 127
make[2]: *** [/usr/src/linux-headers-6.5.0-28-generic/Makefile:2039: /home/user/snd_hda_macbookpro/build/hda] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory &apos;/usr/src/linux-headers-6.5.0-28-generic&apos;
make: *** [Makefile:16: all] Error 2
</pre>

davidjo, 2024-05-04:
What OS is this - there is a now closed issue on Zorin (#116) where for some reason the wrong c compiler seemed to be installed
- the fix was to install  specifically the required gcc-12 version.

green-51, 2024-05-04:
This was on Ubuntu 22.04. I don't remember the exact sub-version off the top of my head.

green-51, 2024-05-09:
Installing gcc-12 fixed the error and audio is working just fine now. Closing this issue.