davidjo/snd_hda_macbookpro #125: Warning on fresh Ubuntu 24.04

vrilutza, 2024-05-07:
Hello, 
         I tried last fresh ubuntu 24.04 on Macbook pro 13 2017

```
vik@vik-MacBookPro14-1:~/snd_hda_macbookpro$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=noble
DISTRIB_DESCRIPTION="Ubuntu 24.04 LTS"
```
`sudo apt install gcc linux-headers-generic make patch wget`
`sudo apt install build-essential`
`sudo apt install git`
`sudo apt install linux-source`

**first step log :** 

`vik@vik-MacBookPro14-1:~$ git clone https://github.com/davidjo/snd_hda_macbookpro.git
Cloning into 'snd_hda_macbookpro'...
remote: Enumerating objects: 345, done.
remote: Counting objects: 100% (126/126), done.
remote: Compressing objects: 100% (45/45), done.
remote: Total 345 (delta 101), reused 93 (delta 81), pack-reused 219
Receiving objects: 100% (345/345), 8.26 MiB | 3.35 MiB/s, done.
Resolving deltas: 100% (217/217), done.
vik@vik-MacBookPro14-1:~$ cd snd_hda_macbookpro/

vik@vik-MacBookPro14-1:~/snd_hda_macbookpro$ sudo ./install.cirrus.driver.sh
Ubuntu linux kernel source not found in /usr/src: /usr/src/linux-source-6.8.0.tar.bz2
assuming the linux kernel source package is not installed
please install the linux kernel source package:
sudo apt install linux-source-6.8.0
NOTE - This does not work for HWE kernels  
vik@vik-MacBookPro14-1:~/snd_hda_macbookpro$ sudo apt install linux-source
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  linux-source-6.8.0
Suggested packages:
  libncurses-dev | ncurses-dev kernel-package libqt3-dev
The following NEW packages will be installed:
  linux-source linux-source-6.8.0
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 354 MB of archives.
After this operation, 357 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ro.archive.ubuntu.com/ubuntu noble/main amd64 linux-source-6.8.0 all 6.8.0-31.31 [354 MB]
Get:2 http://ro.archive.ubuntu.com/ubuntu noble/main amd64 linux-source all 6.8.0-31.31 [9,836 B]                
Fetched 354 MB in 59s (5,966 kB/s)                                                                               
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend requires a screen at least 13 lines tall and 31 columns wide.)
debconf: falling back to frontend: Readline
Selecting previously unselected package linux-source-6.8.0.
(Reading database ... 151728 files and directories currently installed.)
Preparing to unpack .../linux-source-6.8.0_6.8.0-31.31_all.deb ...
Unpacking linux-source-6.8.0 (6.8.0-31.31) ...
Selecting previously unselected package linux-source.
Preparing to unpack .../linux-source_6.8.0-31.31_all.deb ...
Unpacking linux-source (6.8.0-31.31) ...
Setting up linux-source-6.8.0 (6.8.0-31.31) ...
Setting up linux-source (6.8.0-31.31) ...`

**and the second log when compile from terminal :** 

```
vik@vik-MacBookPro14-1:~/snd_hda_macbookpro$ sudo ./install.cirrus.driver.sh
linux-source-6.8.0/sound/pci/hda/
linux-source-6.8.0/sound/pci/hda/hda_intel.c
linux-source-6.8.0/sound/pci/hda/hda_cs_dsp_ctl.h
[... 135 lines cut ...]
/home/vik/snd_hda_macbookpro/build/hda/patch_cirrus_new84.h:1465:6: warning: no previous prototype for ‘cs_8409_headcapture_cleanup’ [-Wmissing-prototypes]
 1465 | void cs_8409_headcapture_cleanup(struct hda_codec *codec)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /home/vik/snd_hda_macbookpro/build/hda/patch_cs8409-tables.o
  LD [M]  /home/vik/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.o
  MODPOST /home/vik/snd_hda_macbookpro/build/hda/Module.symvers
  CC [M]  /home/vik/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/vik/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.ko
  BTF [M] /home/vik/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.ko
Skipping BTF generation for /home/vik/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-31-generic'
make INSTALL_MOD_DIR=updates -C /lib/modules/6.8.0-31-generic/build M=/home/vik/snd_hda_macbookpro/build/hda CONFIG_MODULE_SIG_ALL=n modules_install
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-31-generic'
  INSTALL /lib/modules/6.8.0-31-generic/updates/snd-hda-codec-cs8409.ko
  SIGN    /lib/modules/6.8.0-31-generic/updates/snd-hda-codec-cs8409.ko
  DEPMOD  /lib/modules/6.8.0-31-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-31-generic'
depmod -a

contents of /lib/modules/6.8.0-31-generic/updates
total 2612
-rw-r--r-- 1 root root 2672736 May  7 19:54 snd-hda-codec-cs8409.ko
vik@vik-MacBookPro14-1:~/snd_hda_macbookpro$ 
```

davidjo, 2024-05-08:
Thanks for the input - you beat me to it - my linux machines arent listing 24.04 as available yet.

My initial guess is it looks like the new compiler version (gcc-13) has some flag changes (-Wmissing-prototypes) which seems to require a slightly different coding style to the one that has worked - looks like I need to define some prototypes for functions before using them (most of the warnings are for new functions I created).
The actual patching of files seems to be OK though - and the messages are only warnings so it did actually create the module - as seen in the contents of listing.

vrilutza, 2024-06-02:
Hello again,
sorry for my late reply, yes with a new version (gcc-13) has some flag changes i just make a issue to know about that but the driver is working fine, my dream is to one day try to push upstream. I hope one day try to help you some people from opensource.cirrus.com to push upstream I good start i think is this ticket https://bugzilla.kernel.org/show_bug.cgi?id=110561

davidjo, 2024-06-02:
Thanks - updates from 23.10 to 24.04 have now been allowed so will be doing this shortly.

fspkwon, 2024-07-24:
kernel has to be 6.5

davidjo, 2024-07-29:
Ubuntus default kernel is 6.8 for 24.04 which Im running and sound works for me - the issue is the repo works with the default distribution kernel but using a non-default kernel can give problems.

paulshedley, 2025-03-08:
I'm having a similar kernel issue with 24.04 LTS - mainline kernel is 6.8, but the driver seems to be asking for 6.11.  In an alternative test system, the install works straightforwardly on 24.10 (presumably as this is on the later kernel).  Is there a workaround for 24.04 with a 6.8 kernel?  The error message I get indicates that the driver "does not work for HWE kernels".

fspkwon, 2025-03-09:
again youll find this is no issue you have to LOCK the kernel in synaptic and tell it not to update until you want it to, also you yes have to install a kernel that has everything in line sequentiation with what it asks -generic is 1. -unsigned would be the other as there would probably be confusion asking users to sign their own kernel modules.. so its 1 or so kernel will be available to you as you want the version it is, its headers, its source, debug thereof , its modules, its firmware, and i think thats about it... even the -modules-extra package and you will find the driver will work for you

davidjo, 2025-03-09:
I just did some updates and kernel 6.8 is still default for 24.04.2 LTS - and sound still works for me.
Dont know why it would be asking for 6.11 (although that appears to be the 24.10  kernel version).
Under Ubuntu only the current generic kernel has a default dpkg kernel source - which is what needs to be downloaded to build the sound module.
Its not that the module wont work with HWE kernels its that the kernel source and building is much more complicated to setup - there is no dpkg you have to download the source package from ubuntu, apply all the patches, modify the install script so it points to where you installed the kernel source code etc.

paulshedley, 2025-03-09:
> I just did some updates and kernel 6.8 is still default for 24.04.2 LTS - and sound still works for me. Dont know why it would be asking for 6.11 (although that appears to be the 24.10 kernel [...]

Thanks David.  Things to try.  Or just upgrade to 24.10 where it 'just works' ....

fspkwon, 2025-03-10:
looks like only 2 more chips to go for freebsd support the headphone ic and the parent card ic one of the 42lxxx 43lxxx and 8409 itself and of course an "Apple-Properties-Patch"