davidjo/snd_hda_macbookpro #126: Missing license

therainydev, 2024-06-11:
There is no license file in the repository, making it unclear how we can reuse this driver.

Can you add a license file? Thanks!

davidjo, 2024-06-14:
Not sure what you mean by reuse - you can use the repo now - are you talking about re-distributing the repo or the module?

therainydev, 2024-06-15:
Right, since without a license your code falls under exclusive copyright. IANAL so I'll refer you to [the Choose a License website](https://choosealicense.com/no-permission/) instead of trying to explain here.

It seems that you already have `SPDX-License-Identifier: GPL-2.0-or-later` on some of the files in the repository, so I assume that's the license you would want to add.

davidjo, 2024-06-16:
Interesting - the SPDX comes from the new patch_cs8409.{c,h} original kernel module source files that was added around kernel 5.13 - which I think was done by Cirrus people (when I started that didnt exist).

Not sure what your plans are but I have to say this is still really alpha code - I know there are issues but I dont really have the kernel knowledge. Currently I know there are issues if you do things like switching audio, plugging in/out headphones fast - I have been meaning to try out using the same locking style that I noticed appeared in  the patch_cs8409 but  thats not likely going to happen.
Also when I started I didnt know about the cs4208.inf file as found on the Bootcamp iso for booting Windows.
This seems to have a completely separate way of doing things compared to the Apple way - at least I couldnt see much overlap between the hda verbs OSX is sending and what is defined in the cs4208.inf file. Apple seemed to do a lot of things in duplicate which I generally ignored (that is when I discovered that  unsolicited responses in the HDA linux module seem to occur in parallel so I hacked up a blocking system (block_unsol) - probably should try using locks for that). It may be better just to try the cs4208.inf way of doing things.

However, this really only applies to machines from approx 2017-2020 (when 8409 chip was used as far as I can see) - when Apple introduced the M series machines it seems to have replaced the 8409 chip with their own silicon - although still seems to be using commercial DAC amplifier chips. I got this from reading about the audio implementation in Asahi linux - where they have done a much better job of implementing audio including DSP processing to make the audio sound much closer to OSX - something Ive known could be done.

As Im planning on getting a new machine soon (hence the interest in Asahi linux) Im not likely to be using this much longer.
As it is Ive not really used linux as much as I used to as never could get a good touchpad setup - have to really remember to keep palms off the pad otherwise get random input.

therainydev, 2024-06-17:
I see.

I'm on a 2017 MacBook Pro (MacBookPro14,3 to be specific). Currently, the audio, touch bar, and graphics switching aren't working. I'm using NixOS, which has an interesting package management and directory structuring approach, so patching the kernel is a bit involved.

This does only apply to machines from 2017-2020, and I think that's going to be more relevant as those Macs lose software support from Apple and we might see more people switching to Linux. At least I am switching to Linux from macOS in part due to Apple dropping software support.

I had the same issues with the trackpad getting random input while I'm typing, though disabling tap-to-click stops it from bothering me, aside from not being able to do tap-to-click.

Thank you for adding the license :)

closed 2024-06-17 by therainydev

davidjo, 2024-06-17:
No probs.

Thats exactly my machine.

You need the macbook12-spi-driver (github) to get the touchbar working - Ive been using it for a while - otherwise no ESC key!!

Plus you need to override the usbmuxd rules file - 39-usbmuxd.rules under /usr/lib/udev - if not usbmuxd grabs the bus and wont let go so touch bar still doesnt work - file data given below.
(Googling should find these updates).

# usbmuxd (Apple Mobile Device Muxer listening on /var/run/usbmuxd)

# systemd should receive all events relating to device
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-5]/*|5ac/8600/*", TAG+="systemd"

# Initialize iOS devices into "deactivated" USB configuration state and activate usbmuxd
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-5]/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="0", OWNER="usbmux", ENV{SYSTEMD_WANTS}="usbmuxd.service"
# but make sure iBridge (T1) doesn't end up in an unconfigured state
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/8600/*", ACTION=="add", ENV{USBMUX_SUPPORTED}="1", ATTR{bConfigurationValue}="1", OWNER="usbmux", ENV{SYSTEMD_WANTS}="usbmuxd.service"

# Make sure properties don't get lost when bind action is called
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-5]/*|5ac/8600/*", ACTION=="bind", ENV{USBMUX_SUPPORTED}="1", OWNER="usbmux"

# Exit usbmuxd when the last device is removed
SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/190[1-5]/*|5ac/8600/*", ACTION=="remove", RUN+="/usr/sbin/usbmuxd -x"