davidjo/snd_hda_macbookpro #128: Gentoo SH Script Issue and issue when configuring kernel..

fspkwon, 2024-07-18:
when entering in hostname <- 6.6.38-gentoo-x86_64 it <- makes a duplicate of the name so the /usr/lib/modules/
folder looks like: 6.6.38-gentoo-6.6.38-gentoo-x86_64 and no module is actually compiled<- is this because of
the Kernel Version? because the same thing happens on a 5.x kernel.

davidjo, 2024-07-19:
are you adding a kernel name argument to install.cirrus.driver.sh?
I usually run without any arguments (ubuntu) for picking the current kernel - are you trying to build for a different kernel to the running kernel??
I have never used Gentoo so dont know its kernel build system

fspkwon, 2024-07-19:
> are you adding a kernel name argument to install.cirrus.driver.sh? I usually run without any arguments (ubuntu) for picking the current kernel - are you trying to build for a different kernel to the running kernel?? I have never used Gentoo so dont know its kernel build system

the kernel argument is provided upon the build kernel installation process of the gentoo handbook is something
sort of like linuxfromscratch or freebsd if you think your geared for that area of interest it can teach someone alot<-
where on the Debian side of things you have to do this whole ton of research which i havent got around
to setting up a source repository <- basically making a apt clone repository that you fill ONLY with your system
packages and allow it to reinstall itself like gentoo, Fedora does this automagically, redhatlinux does this automagically.

fspkwon, 2024-07-19:
:34 https://youtu.be/_jxwQWVI-jA

fspkwon, 2024-07-20:
Xash-3D : 2018 Version:
Half-Life : released in 1998 :
Platform: Windows 95.
XBOX Released: 2001 : Original Game: Half-Life 1, this occured within
the magic 4-year count.

Counter-Intuivity : MacOS 9....................
There was a Port released within 3 weeks.
So this specific Macintosh port thats --Nowhere to be found:
would work theoretically with (WON-2's WON Server Virtualizer, and
Counter-Strike BETA)

"Their leaves without going to say comment is what theyre saying to canadian
at the shoe store asking why would they walk over towards the suki-hana:
lets just get that corrected in their first-offedly second of all<-
I would THINK it said something along the lines of a takes the green plastic
pry-off thing called XBOX and says so "This is what were taking and doing
something with":
Right and to clarify : broaden their horizon theyre at a maceys other town
basically spreading a or so it appears to be message:
How could knowing that your walking around saying 1 thing they want
to sort of say their not DIRECTLY walking around talking about what like<-
and he says ?what is that stupid?:
Thats the XBOX controller where he asks at first glance "i could be stupid
but its not like you'll ever tell me seeing as how your abruptstity at
the store counter<- is it the fact that shes telling the teller
that she wants to get behind the counter and into the back storage room
has 1-answer:

Pirating Effects: The Stock Market Index for your Companies of choice:
it may not be because of a direct decision but when you think they inhibit
ones ability to purchase a disk from gamestop <- these people
contributed to the stock market crash of the 1980s in pertaining to
the "Atari Crash".

That could be what they were insinuatively talking about
going look at this its a brand new super nintendo controller
how did we ever contribute.

So when she says i could understand you want to walk in there
to "Look" for a bit she's saying even I understand i just dont
have the time for this, but it doesnt hurt the 2 seconds it takes
to show someone the atleast i did it look here it says i did it<-
so shes in their living room sitting next to temporary house sitters
that spoke to her so your condoning of:

So Enter: "no is because i remember you pressing phone buttons
this here leads to Blockbuster and a YMCA Pool and shes "Gasping":
"so theyre going DUDE you just said your not gonna ___ my:
in the Sunoco:
So If Justin Timberlake is Climbing the Bridge Overpass
theyre going thats like what i wouldev told you<-

so the entertainment black xmb its saying thats like
saying is it ok if i ask and would see the ephemeral purple glow:
ok so in walking through the door is the blonde girl
going ok this is what were doing today
and the black girl and the one from
taco bell is how their talking all like so he said
like do me and thats the way of society?<-
WELL WELCOME BACK! TO THE OPPOSITE STORIED LEGGO BOX STORY:

fspkwon, 2024-07-22:
OK IT WORKED but now i need to find which modules to include in the kernel -the -amplifiers.

fspkwon, 2024-07-27:
both things i try says it compiles something ive set all audio drivers as modules but depomods nothing