davidjo/snd_hda_macbookpro #129: Linux kernel headers not found on imac 18,2

Stove-git, 2024-07-26:
```
[Steve@Sigma` snd_hda_macbookpro]$ ./install.cirrus.driver.sh
linux kernel headers not found:
Debian (eg Ubuntu): /usr/src/linux-headers-6.10.0-arch1-2
Fedora: /usr/src/kernels/6.10.0-arch1-2
Arch: /usr/lib/modules/6.10.0-arch1-2
Void: /usr/src/kernel-headers-6.10.0-arch1-2
assuming the linux kernel headers package is not installed
please install the appropriate linux kernel headers package:
Debian/Ubuntu: sudo apt install linux-headers-6.10.0-arch1-2
Fedora: sudo dnf install kernel-headers
Arch (also Manjaro): Linux: sudo pacman -S linux-headers
Void Linux: xbps-install -S linux-headers
```
I have linux kernel headers installed, but a newer version: 6.10.1-arch1-1

fspkwon, 2024-07-27:
no tell them your on gentoo because im having an issue also arch you have to build the kernel using the AUR i think

Stove-git, 2024-07-27:
> no tell them your on gentoo because im having an issue also arch you have to build the kernel using the AUR i think what
> was that YOURT? setting up a Pacman and Yourt Repository now that you've sparked my interest come to think about it EVERYTHING CAN BE BUILD USING YAOURT

bro i installed the driver with yay -S and i dont understand a thing

Stove-git, 2024-07-27:
> no tell them your on gentoo because im having an issue also arch you have to build the kernel using the AUR i think what
> was that YOURT? setting up a Pacman and Yourt Repository now that you've sparked my interest come to think about it EVERYTHING CAN BE BUILD USING YAOURT

okay now i understood. i did install it with yay -S

fspkwon, 2024-07-27:
> okay now i understood. i did install it with yay -S

im doing that too im on arch now until i can build the modules against the same kernel version
on gentoo it doesnt but arch it does you have to open up firefox first and play a youtube video
whichll create the pcm device in alsamixer

fspkwon, 2024-07-27:
then theres powerpill/bauerbill and octopi <- bauerbill is like how to optimize the system i do believe

Stove-git, 2024-07-29:
now after succesfully installing the drivers the sound works. BUT the microphone isnt working and i need help with that