davidjo/snd_hda_macbookpro #130: Microphone(Headset) not working on an imac 18,2

Stove-git, 2024-07-29:
The sound works perfectly

davidjo, 2024-07-30:
I have never really attempted to implement the headset microphone (you can always use usb/bluetooth versions - these should just work and are nothing to do with this driver).
I did work on the internal microphone which sort of works but it would require additional pipewire/pulseaudio/alsa fixups to work properly.
Also at the moment mike input switching doesnt work automatically - need to explicitly set the input using your distributions audio settings.
You can try searching the issues for microphone to see what has been found before.
There are also differences in the jack code between my macbook pro 14,3 and imacs - I think that was sorted for headphones.
Try and make sure its not an amplification issue - the internal mike is very quiet but that seems to be what OSX delivers from the raw hardware driver to the OSX higher level audio functions.

Stove-git, 2024-08-05:
> I have never really attempted to implement the headset microphone (you can always use usb/bluetooth versions - these should just work and are nothing to do with this driver).
I use a 3.5mm jack headset that has a microphone in it and it doesnt work
> I did work on the internal microphone which sort of works but it would require additional pipewire/pulseaudio/alsa fixups to work properly.
The internal mic doesnt work at all for me
> Also at the moment mike input switching doesnt work automatically - need to explicitly set the input using your distributions audio settings.
How do i do this on arch
> You can try searching the issues for microphone to see what has been found before.
I found nothing
> There are also differences in the jack code between my macbook pro 14,3 and imacs - I think that was sorted for headphones.
Maybe, but what do i do then, becausr only sound in the headset works for me
> Try and make sure its not an amplification issue - the internal mike is very quiet but that seems to be what OSX delivers from the raw hardware driver to the OSX higher level audio functions.
Its not

davidjo, 2024-08-05:
I just googled for usb audio jack adapter and got multiple devices which would allow you  to use a jack headset with mike via usb.

The internal mike was working for me (and I think still is) but I do think there were problems with imacs.

Try searching in github using the following     repo:davidjo/snd_hda_macbookpro  is:issue microphone
this came up with multiple results for me from the issues (open or closed).

Ive never used Arch so cant help you there - but there must be some gui for doing basic settings - the equivalent of Settings on OSX. On Ubuntu the OSX Settings equivalent lets you choose which audio inputs and outputs to use manually.

The problem is I dont have an iMac or access to one - this whole driver is based on originally being able to log from the OSX driver to see whats happening - without a log from an imac using a headset with microphone running High Sierra (Apples been extensively locking down OSX and I dont think I can do what I did before on current versions of OSX) then we have no idea whats going - there is NO documentation from Apple or Cirrus on the specific chips Apple is using - everything is guesswork.

If you look in the code you will see variables like intmike_nid, intmike_adc_nid and in patch_cirrus_apple.h there is code around the comment
        // setup the intmike and linein nids
        // these are swapped between macbook pros and imacs
which shows a specific difference between the macbook pros and imacs that was fixed so that you do hear audio on your headset/headphone.
Who knows what other  differences might exist for the headset mikes?

Stove-git, 2024-08-05:
> Ive never used Arch so cant help you there - but there must be some gui for doing basic settings - the equivalent of Settings on OSX. On Ubuntu the OSX Settings equivalent lets you choose which audio inputs and outputs to use manually.

Arch is like Gentoo which is barebones and has 0 settings

fspkwon, 2024-08-06:
Fellow iMac User : are you using Gentoo?<- I just got done going over XNINE which is XNINE-TESTS not particularly Gallium9 Wine <- but those specific packages are all screwed <- and Beignet.... <- on Ubuntu you might see that the Internal Mic works, Headphones might work... atleast I THINK* thats what it says ... ArchLinux is simple to get it to work you simply follow the handbook guide up through building packages, hosting a repo, building the kernel, and then move to this tutorial which tells you a couple other packages to install then do the same thing ./installdriver.sh or w./e and it just KNows your using ArchLinux , it doesnt know howeverl; Gentoo which when it tries to compile i dont know yet if its Modularity is the Issue ... is it the fact im not using which specific drivers says it *PATCHES specific drivers but i cant like look at it read it and go oh ok so these are the ones it wants... although it does automagically try to compile <- using the kernel and all just when it outputs the number of driver-updates is 0, the number of modules it actually compiles is 0, it never actually starts the Compiling Process.. is another thing.. and as far as freebsd <- i daisy chained the nids up and down so the mixers all open it just doesnt tell me what gpio ports 0-9 i think are open/closed which im waiting to hear from someone who has it actually working using some kind of a reboot process.... which of course thats a no-go so aesthetically speaking freebsd is able to SHOW everything its just the actual functions gone *if in fact it actually is i dont know if anyone ever got as far as to aesthetically fixing nearly everything that and when it attempts to use the driver the driver says its locking buckling at the audio mixer level

fspkwon, 2024-08-06:
i too have the iMac 18,1 first generation non-retina