davidjo/snd_hda_macbookpro #134: No Audio! Macbook pro 14.1! Running Arch Linux

LukeW2002, 2024-08-23:
I run the make with no errors. 

Heres the specs:
`➜  ~ inxi -Fxxrzc0      
System:
  Kernel: 6.10.6-arch1-Adashima-T2-1-t2 arch: x86_64 bits: 64 compiler: gcc v: 14.2.1
  Console: pty pts/0 DM: startx Distro: Arch Linux
Machine:
  Type: Laptop System: Apple product: MacBookPro14,1 v: 1.0 serial: <superuser required> Chassis:
    type: 9 v: Mac-B4831CEBD52A0C4C serial: <superuser required>
  Mobo: Apple model: Mac-B4831CEBD52A0C4C v: MacBookPro14,1 serial: <superuser required>
    UEFI: Apple v: 499.40.2.0.0 date: 08/22/2022
Battery:
  ID-1: BAT0 charge: 39.8 Wh (100.0%) condition: 39.8/54.6 Wh (72.8%) volts: 12.8 min: 11.4
    model: SMP bq20z451 serial: N/A status: full
CPU:
  Info: dual core model: Intel Core i5-7360U bits: 64 type: MT MCP arch: Amber/Kaby Lake
    note: check rev: 9 cache: L1: 128 KiB L2: 512 KiB L3: 4 MiB
  Speed (MHz): avg: 3591 high: 3597 min/max: 400/3600 cores: 1: 3591 2: 3589 3: 3597 4: 3589
    bogomips: 18406
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel Iris Plus Graphics 640 vendor: Apple driver: i915 v: kernel arch: Gen-9.5 ports:
    active: eDP-1 empty: DP-1, DP-2, HDMI-A-1, HDMI-A-2 bus-ID: 00:02.0 chip-ID: 8086:5926
  Display: unspecified server: X.org v: 1.21.1.13 with: Xwayland v: 24.1.2 driver: X:
    loaded: modesetting alternate: fbdev,intel,vesa dri: iris gpu: i915 tty: 108x28
  Monitor-1: eDP-1 model: Apple Color LCD res: 2560x1600 dpi: 227 diag: 337mm (13.3")
  API: OpenGL Message: GL data unavailable in console, glxinfo missing.
Audio:
  Device-1: Intel Sunrise Point-LP HD Audio driver: snd_hda_intel v: kernel bus-ID: 00:1f.3
    chip-ID: 8086:9d71
  Device-2: Broadcom 720p FaceTime HD Camera driver: N/A pcie: speed: 5 GT/s lanes: 1
    bus-ID: 03:00.0 chip-ID: 14e4:1570
  API: ALSA v: k6.10.6-arch1-Adashima-T2-1-t2 status: kernel-api
  Server-1: PipeWire v: 1.2.2 status: active with: 1: pipewire-pulse status: active
    2: wireplumber status: active 3: pipewire-alsa type: plugin 4: pw-jack type: plugin
Network:
  Device-1: Broadcom BCM4350 802.11ac Wireless Network Adapter vendor: Apple driver: brcmfmac
    v: kernel pcie: speed: 2.5 GT/s lanes: 1 bus-ID: 02:00.0 chip-ID: 14e4:43a3
  IF: wlan0 state: up mac: <filter>
Drives:
  Local Storage: total: 233.76 GiB used: 10.91 GiB (4.7%)
  ID-1: /dev/nvme0n1 vendor: Apple model: SSD SM0256L size: 233.76 GiB speed: 31.6 Gb/s lanes: 4
    serial: <filter> temp: 37.9 C
Partition:
  ID-1: / size: 232.76 GiB used: 10.59 GiB (4.5%) fs: btrfs dev: /dev/nvme0n1p2
  ID-2: /boot size: 1021.9 MiB used: 330.2 MiB (32.3%) fs: vfat dev: /dev/nvme0n1p1
  ID-3: /home size: 232.76 GiB used: 10.59 GiB (4.5%) fs: btrfs dev: /dev/nvme0n1p2
  ID-4: /var/log size: 232.76 GiB used: 10.59 GiB (4.5%) fs: btrfs dev: /dev/nvme0n1p2
Swap:
  ID-1: swap-1 type: zram size: 3.81 GiB used: 0 KiB (0.0%) priority: 100 dev: /dev/zram0
Sensors:
  System Temperatures: cpu: 58.0 C mobo: N/A
  Fan Speeds (rpm): N/A
Repos:
  Packages: pm: pacman pkgs: 769
  Active pacman repo servers in: /etc/pacman.conf
    1: https://mirror.funami.tech/arch-mact2/os/x86_64
  Active pacman repo servers in: /etc/pacman.d/mirrorlist
    1: https://archlinux.uk.mirror.allworldit.com/archlinux/$repo/os/$arch
    2: https://mirror.osbeck.com/archlinux/$repo/os/$arch
    3: https://archlinux.mailtunnel.eu/$repo/os/$arch
    4: https://london.mirror.pkgbuild.com/$repo/os/$arch
    5: https://geo.mirror.pkgbuild.com/$repo/os/$arch
    6: https://mirror.moson.org/arch/$repo/os/$arch
    7: https://packages.oth-regensburg.de/archlinux/$repo/os/$arch
    8: https://mirror.ubrco.de/archlinux/$repo/os/$arch
    9: https://europe.mirror.pkgbuild.com/$repo/os/$arch
    10: https://arch.phinau.de/$repo/os/$arch
    11: https://arch.mirror.constant.com/$repo/os/$arch
    12: https://archlinux.thaller.ws/$repo/os/$arch
    13: https://md.mirrors.hacktegic.com/archlinux/$repo/os/$arch
    14: https://america.mirror.pkgbuild.com/$repo/os/$arch
    15: https://archlinux.za.mirror.allworldit.com/archlinux/$repo/os/$arch
    16: https://archlinux.c3sl.ufpr.br/$repo/os/$arch
    17: https://seoul.mirror.pkgbuild.com/$repo/os/$arch
    18: https://mirrors.neusoft.edu.cn/archlinux/$repo/os/$arch
    19: https://asia.mirror.pkgbuild.com/$repo/os/$arch
    20: https://sydney.mirror.pkgbuild.com/$repo/os/$arch
Info:
  Memory: total: 8 GiB available: 7.63 GiB used: 1.21 GiB (15.9%)
  Processes: 190 Power: uptime: 5m wakeups: 0 Init: systemd v: 256 default: graphical
  Compilers: clang: 18.1.8 gcc: 14.2.1 Shell: Zsh v: 5.9 running-in: pty pts/0 inxi: 3.3.35
`

checked the usual, I dont have an audio out other than "pro audio"  via pavucontrol but that does nothing. Im fairly stumped.

LukeW2002, 2024-08-23:
If I use bluetooth headphones I get audio. If I use the audio-jack I get nothing.

davidjo, 2024-08-24:
Just checked under Ubuntu - I see Built in Analog Audio under pavucontrol Output Device and under Configuration I see Analog Stereo Duplex.
Dont know where pro audio is coming from.
What desktop environment are you running?
Under Ubuntu gnome has a settings app which also shows the audio setup and you can set the output device etc. - havent used pavucontrol for a while.

The audio jack has problems - you need to explicitly set it as the output (it should switch if you plug in while playing) - but headphones work for me for output - input has issues.

KevinDoughty, 2024-08-24:
I have a 14,1 running Ubuntu which also lost sound recently.

$ sudo ./install.cirrus.driver.sh
Ubuntu linux kernel source not found in /usr/src: /usr/src/linux-source-6.8.0.tar.bz2
assuming the linux kernel source package is not installed
please install the linux kernel source package:
sudo apt install linux-source-6.8.0
NOTE - This does not work for HWE kernels

$ uname -mrs
Linux 6.8.0-40-generic x86_64

$ apt-cache search linux-generic
$ sudo apt install linux-modules-extra-6.8.0-40-generic
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
linux-modules-extra-6.8.0-40-generic is already the newest version (6.8.0-40.40~22.04.3).
linux-modules-extra-6.8.0-40-generic set to manually installed.
The following packages were automatically installed and are no longer required:
  apturl apturl-common ffmpeg libavdevice58 libcue2 libdc1394-25 libdecor-0-0 libdecor-0-plugin-1-cairo libdvdnav4 libexiv2-27 libgexiv2-2
  libgsf-1-114 libgsf-1-common liblua5.2-0 libmujs1 libopenal-data libopenal1 libplacebo192 libsdl2-2.0-0 libsixel1 libsndio7.0
  libtracker-sparql-3.0-0 libwpe-1.0-1 libwpebackend-fdo-1.0-1 mpv nautilus-data python3-brotli python3-mutagen
  python3-proton-vpn-connection python3-proton-vpn-killswitch python3-proton-vpn-session python3-pycryptodome python3-pyxattr
  python3-websockets rtmpdump
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.

$ sudo ./install.cirrus.driver.sh
Ubuntu linux kernel source not found in /usr/src: /usr/src/linux-source-6.8.0.tar.bz2
assuming the linux kernel source package is not installed
please install the linux kernel source package:
sudo apt install linux-source-6.8.0
NOTE - This does not work for HWE kernels

davidjo, 2024-08-27:
It may be OK - if you didnt see build errors when running the install script it likely created the module - for a while now new kernel versions havent had changes which cause the module not to build.
One thing to look for in the kernel log files (usually under /var/log either syslog or kern.log) is a line containing
"Primary patch_cs8409 NOT FOUND trying APPLE" - this is my print to indicate my module is loaded.

So the usual kernel structure is that kernel modules are stored under /lib/modules using eg for me with 6.8 kernel and ubuntu the folder 6.8.0-40-generic (if using a generic kernel).  Note that the 40 can vary.
Non ubuntu may see something like 6.10.6-generic.
In that folder, the default kernel module is stored under kernel/sound/pci/hda - snd-hda-codec-cs8409.ko (or possibly snd-hda-codec-cs8409.ko.zst).
My module should be installed in another directory updates (directly under for me 6.8.0-40-generic).
So first check if there is a module snd-hda-codec-cs8409.ko in that folder - if not then you wont get sound - the default module will be loaded.
This directory updates is special for kernel modules - modules with the same name as an existing module (only file name needs to match ie just the snd-hda-codec-cs8409 part) should override an existing module and get loaded in preference to the existing module (this does require a sudo depmod -a being done - the install script should be doing this).

If my module exists under updates you can live dangerously and do the following.
Move the existing kernel module (ie under kernel/sound/pci/hda)  to somewhere else safe in the file system (will need sudo).
Run sudo depmod -a, reboot and see if you get sound.
(I have done this procedure a number of times to really ensure my module is being loaded when having issues).

Obviously at some point should move the original kernel module back - and redo sudo depmod -a.

nate-moo, 2024-09-29:
Hello, I'm not entirely sure if this is the right place to ask this, but I have tried to build this kernel module patch for NixOS but haven't been successful in getting it to install. There is a patch that was applied to enable the cs8409 codec which might be similar. The audio driver loaded by the sound card on my mac is `snd_hda_intel` according to lspci, and looking in the lsmod tree of loaded kernel modules the `snd_hda_codec_cs8409` is loaded seemingly correctly. In dmesg it also seems to be able to find the audio device and detect its ports but I am only getting non-functional pro-audio ports on the audio device. Let me know if I'm talking up the wrong tree, Thanks for your time!

davidjo, 2024-09-30:
This is a good enough place - which model do you have - MacBook Pro 14.1 as in the title??
Unfortunately just because snd_hda_codec_cs8409 is loaded doesnt mean its the module from this repo - as there is a system version (snd_hda_intel is a generic driver which is always loaded for HDA chips - which the cs8409 is - and this repo doesnt change it).
From your description I think its likely its the system module thats loaded.
So - when you ran the build script did any errors occur?
Is there a module snd_hda_codec_cs8409.ko under the /lib/modules /  kernel version / updates directory?
(see my comment just above yours for fuller description and things to try).
If there is a module under updates try running a depmod -a (sudo needed) and reboot - always good to reboot to ensure kernel module updates are included.

nate-moo, 2024-09-30:
I think I'm going to have to switch distros for this to work properly, I was using NixOS with the hardware tree enabled and it mentioned the codec that is patched here simply being enabled not patched. I am not aware of an easy way to write a nix script to build and patch the module because of the way this system builds the patched module. If you have any better insight in how to get that patched, I'd greatly appreciate it :)

davidjo, 2024-09-30:
Ive never used NixOS so cant really help - from what little Ive seen is seems quite different to other distributions.
However, the updates directory is I believe a base linux kernel property so should be able to add modules (and run depmod -a) manually - does NixOS have the /lib/modules/version.../ structure?? (the initial /lib may be different). (Unless NixOS is not using kernel modules just a single file).

nate-moo, 2024-09-30:
Yeah it has something similar to that, but the structure is dependent on the nix store, so for current kernel the `/lib/modules/` doesn't directly exist, for me it resides in `/nix/store/hvsh5z458z2ck3sx05bsb1gb11ks5lq4-linux-6.11/lib/modules/6.11.0/` or `/nix/store/834sznrbkjfjbghywv3bxw4lqfzk336b-linux-6.11-modules/lib/modules` Loading modules is done via the configuration file, of which needs to be in your nix-store to be loadable. Based on the [wiki entry](https://nixos.wiki/wiki/Linux_kernel#Out-of-tree_kernel_modules) I might be able to build it and manually insert it, but that is likely not persistent, I'd have to build it into the correct directory which I havent been successful at (I have never done kernel development before). I'm going to continue looking into it, but after a point it might be more likely for me to just distro hop to something standardized for this to work right :sweat_smile:

davidjo, 2024-10-01:
I just found the following - which seems to have a description on  how to build out of tree kernel modules - and has an example script.

See https://blog.prag.dev/building-kernel-modules-on-nixos

nate-moo, 2024-10-01:
Ah I did see that blog post, I think what I'm getting lost in is reading your scripts and makefile there are some hardcoded paths that I'll have to modify. But also that it doesn't seem like the script builds the module more than it patches the existing module, but I could be wrong? I think I'll have to go through the install script you provide and rewrite that into the nix build script, which if I find time this week I'll try to do.

davidjo, 2024-10-03:
Essentially the install script does 2 things:
1. create a build directory which contains a copy of the main linux kernel source sound/pci/hda files (what I needed was some of the local include .h files - initially didnt know what I needed alter but most of the .c files are not used at all).
2. now that there is an existing cs8409 module we have to edit (ie patch in the sense of the linux patch command) that file to include a code path for the Apple devices (both the patch_cs8409.c  and the patch_cs8409.h). I used a quick hack of creating files that are just included into that main source which contain all my extra code.
3.  the module is built using the mainline linux make system (see the make commands in the Makefile at top of the repo - which is similar to the command in the buildPhase of  the nixOS buildPhase - I also update the sound/pci/hda Makefile so it just does the snd-hda-cs8409.ko module).

(note that when I started I did all this manually, set up the build directory then manually run the kernel  Makefile).

So it seems to me you could update the postUnpack phase to do the file copying/patching and adjust the buildPhase to use the build directory (as far as I can tell the $sourceroot/blackmagic-.. directory is the out of kernel source module directory).
I think the fetchPatch items can be just ignored.

nate-moo, 2024-10-05:
Ohhh ok this was very helpful, so what I'm thinking is getting the sources for the module by hand, applying your patch, and then running a kernel module build on that instead of trying to shoehorn your script into my build script for nix. I'm thinking maybe this should be a different issue so we can document it in case other people come along and try the same thing? Thats up to you though, since I'll have time later today I might give that a try and see how far it will take me :)

nate-moo, 2024-10-07:
Sadly I ran out of time, and painfully ended up hopping distros to fedora, If I end up not getting swamped with school and feel like attempting this again, I might but for now I just wanted a working system. Sorry for taking up your time in this issue :/

davidjo, 2024-10-08:
One of the reasons I stick with one linux OS is precisely this - I just want a working system - just dont want to get involved with multiple OS types.
Over the period of this module Ive discovered there are some quite significant differences between the various Linux OS versions - as you are discovering.
Hopefully I did give you some ideas to follow if you get time to return to it - all of what Ive done has come from googling and trying things out - Ive never been a kernel "expert".

LukeW2002, 2024-10-13:
Hey! Been a while sorry, I decided to just use bluetooth headphones and call it day. 
I returned back to this with fresh eyes. I have just arch linux, and I am not dualbooting, but I have grub anyway.

I ran `sudo dmesg | grep -i cs8409 ` and got that the `[    5.551098] snd_hda_codec_cs8409 hdaudioC0D0: UNKNOWN subsystem id 0x10138409`

I saw in #63 >When I reset the NVRAM, audio works for the first boot. Thereafter, it's dead again, until I reset the NVRAM again. Then  from #80 seems like a boot issue.

Solution which has worked even with reboot, and suspend etc was to simply update my grub bootloader after NVRAM reset

LukeW2002, 2024-10-13:
explicitly I ran 
```
sudo grub-install --target=x86_64-efi --efi-directory=/boot/efi --bootloader-id=GRUB
 grub-mkconfig -o /boot/grub/grub.cfg
```

Thanakrit-Anutrakulchai, 2024-10-15:
> Ohhh ok this was very helpful, ... I'm thinking maybe this should be a different issue so we can document it in case other people come along and try the same thing? Thats up to you though, since I'll have time later today I might give that a try and see how far it will take me :)

So, I came across this thread and also had time to try things out. In case you want to give NixOS another shot, I also found [this post for building and installing the snd_hda_cs8409 as a module](https://micro.thedroneely.com/m/tdro/messages/cs8409/#code-block:8bc7190). I am using NixOS on a macbookpro 14,1, with nixos-24.05. I modified the first `sed` line in the file, replacing `linuxKernel.kernels.linux_6_0` with just `kernel`. Then, I called it as a module from my flake.nix as so:
```nix
{
   # ... description stuff ...
    inputs = {
        nixpkgs.url = "github:NixOS/nixpkgs/nixos-24.05";
        nixos-hardware.url = "github:NixOS/nixos-hardware/master";
    };

    outputs = inputs@{ self, nixpkgs, nixos-hardware }: 
        let 
            system = "x86_64-linux";
            pkgs = nixpkgs.legacyPackages.${system};

           # ... stuff ...
            packages = {config, ...}: ({
                boot.extraModulePackages = [
                    (pkgs.callPackage ./snd_hda_codec_cs8409.nix {
                        kernel = config.boot.kernelPackages.kernel;
                    })
                ];
            });
        in
        {
            nixosConfigurations.nixos = nixpkgs.lib.nixosSystem {
                inherit system;
                
                modules = [
                    packages
                    ./configuration.nix
                    nixos-hardware.nixosModules.apple-macbook-pro-14-1
                ];
            };
        };
}

```
As a reminder, when copying Makefile code over, be wary of (eliminated) tabs ! There should be some before the two "make"s after colons. Fortunately, I didn't waste much time on it, but I know my past self would had.

My plan was to try and build that, see it fail, then modify it using the diff files from this repo. To my surprise, once I rebooted my system, I had another "Built-in Audio Pro" entry listed as a sounds device, and it just works ! I think maybe this has something to do with [this PR on nixos-hardware](https://github.com/NixOS/nixos-hardware/issues/688), but to be honest, this is a bit out of my depth so don't quote me on that.

KiLLeRRaT, 2024-10-21:
Hi Folks.

I'm just posting my findings here and what worked for me.  I'm not 100% sure it will work for you, but it's quick and easy to try... See TLDR at the end if you're impatient .

Here goes:

I've a MacbookPro14,1, Arch, via GRUB. No other OS on the machine.
I'm running kernel: `6.11.4-arch1-1`.

I've installed `https://aur.archlinux.org/packages/snd-hda-macbookpro-dkms-git` which points to this repo.

Audio not working.  I get `Primary patch_cs8409 NOT FOUND trying APPLE` like others have mentioned. Also the Dummy Output problem in `pavucontrol`.

I also have a MacbookPro13,3 and that works fine using similar patches.  Difference being that the 13,3 runs via [rEFInd](https://wiki.archlinux.org/title/REFInd) alongside macOS.  Audio works, flawlessly.

I then by chance read something about trying to boot but by holding the Option key, and manually selecting your EFI boot.  Doing this, the audio worked!

I kept using the machine that way, but each time I turned it on and forgot to hold Option I had to reboot.  Less than ideal.

I then kept reading and found a random old post about EFI, DSDT and CRS.  You can read more here if you want: https://discussions.apple.com/thread/250137979?sortBy=rank&page=1

I then just thought I'd give the `pci=nocrs` a shot in my kernel parameters, and this seems to have done the trick!

I'm guessing rEFInd somehow discards the ACPI PCI init _stuff_ (lack of understanding on my part, calling it stuff...), but GRUB kept it.  Manually specifying the `pci=nocrs` now discards it too.

Anyway, hope this could help someone.

TLDR: Boot by holding option key and selecting your EFI option.  If audio works, try setting the `pci=nocrs` [kernel parameter](https://wiki.archlinux.org/title/Kernel_parameters#GRUB).

BMOCatbug, 2025-02-23:
I have a somewhat similar issue in Manjaro (Arch Linux) on the same hardware, whereby everything seems to be working until I run `./install.cirrus.driver.sh`, and it fails at the following:

`make[1]: *** /lib/modules/6.12.12-2-MANJARO/build: No such file or directory.  Stop.`

I don't seem to have a build directory at that path. But I do have one in /lib/modules/5.10.234-2-MANJARO, despite the fact that this is a fresh install of 6.12.12-2-MANJARO

Happy to provide additional details if there is anything that would be helpful.

davidjo, 2025-02-23:
I have never used Arch so I can only suggest things.
Im guessing you do have /lib/modules/6.12.12-2-MANJARO/ directory - just not the build - as that is a standard path for linux kernels.
Next guess is that you have not installed tools for building kernel modules - which is why it may exist in the old (5.10...) kernel directory but not this one.
I suggest figuring out what you need to install to be able to build kernel modules.