davidjo/snd_hda_macbookpro #135: iMac (iMac 18,2) can't go below 15% sound

asutoshvariar, 2024-09-03:
I've been experiencing a persistent issue with the snd_hda_macbookpro patch, where if the volume for the internal speakers is set below 15%, no sound is produced. It's happened on Arch and Fedora, across multiple kernel and patch versions. Is there any information that I can provide which would help diagnose and remedy this issue?

System information:
- Machine: iMac 18,2
- Sound chip: Cirrus Logic CS42L83 audio codec, via iFixit's teardown
- CPU: Intel Core i5 7400
- RAM: 8 GB, 2400 MHz
- GPU: AMD Radeon Pro 555

snd_hda_macbookpro commit: 259cc39e243daef170f145ba87ad134239b5967f

huntekye, 2024-09-03:
I'll add in that this has always been true for me, over the last ~2 years. I always thought of it as happening at 20%, but @asutoshvariar is right, it happens at 14%—after some testing in pavucontrol, it definitely happens at exactly -51.00 dB, i.e., for me there is sound at -50.99 dB, but none at -51.00 dB. I'll admit that at that point the audio is pretty quiet already, so I don't typically see this as a huge issue for the internal speakers, but I know that some headphones I have are very loud in comparison and I've often wished I could get down to 10% without it going completely silent.

System information:
MacBook Pro 14,1
kernel 6.9.7-arch1-1 (though, again, this has been consistent for me since somewhere around 5.10 I think.)
I have snd_hda_macbookpro is installed via dkms.

davidjo, 2024-09-04:
Unfortunately I did see this myself close to the beginning of doing this but at the time had other things to worry about.
What I did was add logging to the OSX driver to see what HDA commands were being sent under OSX. My module implements those OSX logged commands - there is no documentation. There could be many reasons for the volume inconsistency.
I noticed that OSX just sets max volume on the amps at init time (see functions cs_8409_setup_amps12/cs_8409_setup_amps34 and the specific amp functions cs_8409_setup_amp_max etc) so all volume control is done by the higher level audio setup - which equivalently on linux would be pipewire/pulse audio (possibly some high level Alsa code) - which is what pavucontrol is manipulating.
So it could be some difference between the high level OSX audio driver and the high level linux drivers - but I have as much knowledge as you about those higher level drivers.
The only thing I can think to pin this down at the moment is as follows.
You need to get Audacity - this is sound editing/manipulating program.
If you look under the tests directory of the repo you will see some test waveforms - if you want to use your own tests you must use the format of those .wav files ie 44100 Hz, 32 bit. You can use plain stereo waveforms now rather than 4 channel (note these are NOT 4 channel surround sound as listed under Ubuntu - its a specific 4 channels to drive each amp - but thats what you need to set it to to use the 4 channel waveform - they can be just duplicated).
You can see one of the waveforms is a reduced volume - this was created using Audacity by loading the full volume then using an Audacity operation to reduce the waveform size and hence its volume.
You must write out in .wav format.
My idea is to create a number of versions which reduce to around -50db - both above and below.
Then (NOTA NOTA BENE BE VERY CAREFUL HERE NOT TO USE A FULL VOLUME SOUND!!!) use the following command
aplay -D hw:0,0 "reduced waveform name".wav.
This will play the audio directly to the amps ignoring all high level manipulation on linux - which remember have been set to full volume hence the warning.
If the amps stop playing below the -50db then we know its something to do with the amps setup - if not we know its something to do with the high level setup.

Im thinking it may be something to with audio range eg something is doing volume control non-linear but the amps expect linear (or vice versa). Maybe theres some cutoff in the amps - the amp chips at least are commercial so there is documentation available.

fspkwon, 2024-09-04:
ok so yeah like he said its very loud ok but however i found a new issue when you suspend/hibernate which ive gotten working properly on gentoo along with the card it knocks the soundcard offline for good, and nothing you do can fix the soundcard

fspkwon, 2024-09-04:
Gentoo Here: systemd-less using open-rc and tweaked acpid and elogind and hibernate-script and suspend blocked out
a kde and gnome checker and made it #Deep Suspend: and Tested Hibernate by using LinuxQuestions form of asking how to suspend hibernate
like echo mem/disk > /sys/power/state and yes can report this is in fact authentic it knocks the sound offline for good <-
i don't see feasible booting another OS to fix it ALTHOUGH if i somehow managed to get my hands on a Second USB-Disk to boot a Physical
Install Media then maybe i would find the point in resorting to Recovery if its a Linux Based Desktop<- meaning Partition-wise i do not want a recovery partition , the only reason i see needing suspend is because i knew i had a emachines a while back where it was this BIG issue because someone called something broken but no you fix it and itll work regardless theres even like a special directory for rules that pertain to conflicting modules in my case brcmfmac and cs8409 but that .... k isnt an issue once you leave it running in fact waiting is part of the process you wait 50 seconds after whatever specific action you did attempt to try it and using re-configuration for the Power Button Event you Hit the power button itll start right up , in hibernates case sits at grub shows apple logo everything like a wow it just didnt work but nope let it sit at loading initramfs itll boot right into X11
but like i said the audio card was knocked offline i think because i tried a if i was ever in the case to test so i want to test so with alsa running thats something i think is very specific now hold on because i also heard someone state try plugging in and unplugging the headphone jack
no its reporting a list of control files when it tries to restore mixer levels i loaded the kernel module snd-hda-codec-cs8409 and nothing plug unplug headphone jack nothing:

-> OK VERY SPECIFic the solution is to HOTPLUG the Power Coord holding the power button:

huntekye, 2024-09-05:
@davidjo Thanks for the suggestions! I was able to the the StereoTest file to make tests ranging from -40 dB to -100 dB, and tested them with the command `aplay -D hw:1,0 StereoTest32_reduced_m*dB.wav`. For me the device `hw:0,0` didn't work, and I could only use `hw:1,0` from a tty after exiting my Sway session, otherwise it gives the error `audio open error: Device or resource busy`. The result was that every one of those tests was audible all the way down to -100 dB (if I stick my ear on the speaker), which I guess suggests that this is probably not a problem with how the amps are set up.

For my (plugged) headphones the behavior is basically the same: Audio works if pavucontrol is set at -50.99 dB, it doesn't work set at -51 dB; I can play to them with `aplay -D`; and they are audible below -50 dB, at least down to -90 dB.

davidjo, 2024-09-05:
Thats good - sorry yes the hw: device will depend on what audio hardware you have - on a plain MacbookPro it will be hw:0,0.
You can find the devices by running aplay -l (thats lower case L) (could be affected eg by plugging in monitors or usb audio devices). You want the HDA Intel PCH device analog version.
I dont know what Sway is but if something opens the raw audio device you will need to close that program down as in general only one process is given access to the raw device I think.

So it looks as though it is something to do with the higher level audio setup.
All I can suggest is some googling - which is what I would do.
Have to say Ive found it difficult to get a good idea of what the higher level linux audio drivers are actually doing eg what waveform processing they are actually doing.

davidjo, 2024-09-05:
One idea would be try and pin it down to Alsa vs pipewire/pulse audio.
Some of the older issues discussed how to try and improve the quality by setting up alsa pseudo devices - in particular #23.
You would need to create .asoundrc files which create an alsa pseudo device - you should then be able to use that pseudo device with aplay.

huntekye, 2024-09-05:
Google seems to think that is an old pipewire issue: their open issues [2202](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/2202), [1595](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/1595), and [2256](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/2256) all look like they're exactly the same, and have all been open for years. An important point in those issues is the way that the different hardware mixers are reporting their volume range—they are mainly concerned with pipewire behaving wrong in the presence of correct information, but how sure are we the pipewire is getting the right information about the hardware volume range? i.e., is it something controlled by your driver that could be changed?

fspkwon, 2024-09-05:
for those who cant hotplug if device suspends and wont resume <- command option p & r to pram reset until the Doom chime of ___ Occurs:

davidjo, 2024-09-05:
[fspkwon](https://github.com/fspkwon) please move this to a new issue or add it to #87 - which has discussed this and a fixup that may work.

davidjo, 2024-09-05:
[huntekye](https://github.com/huntekye) - well youve found more than I have - but I alos think you see what I mean about this being very complex.
I looked at [2202](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/2202) which I thought had some interesting suggestions about api.alsa.ignore-dB and api.alsa.soft-mixer (this one might be worth trying) -  see Frank Dana comments - which I really dont understand whats going on.
I dont see any code in my part of the driver that reports the volume range - never heard about this before - however HDA chips are very special and theres a lot the hda driver that is generic which I havent really looked at so maybe its assumed to be part of the generic hda setup (see eg hda_generic.c in the build directory).
But this is what I mean about the higher levels seem to be very complex.
Just read a bit further on the Frank Dana - and he does say that adding api.alsa.soft-mixer does seem to fix this and says how to fix this if your system is using wireplumber (which most linux OSs are now).

davidjo, 2024-09-05:
More I think about this using api.alsa.soft-mixer  is likely the way to go.
From what I learned when trying to sort things out about HDA chips (which are very complex) for  the none-8409 style the DACs, amps are actually on the HDA chip - and Ive just remembered those chips also could have mixer sections.
However there are no mixers on the 8409 so there is nothing to report - and as I said what Apple is doing is all software mixing at the  higher OSX level - which is essentially what setting api.alsa.soft-mixer to true would do if Im understanding things OK.

huntekye, 2024-09-05:
Okay so some more info I've found:
`amixer -c 1` (where one is the sound card number, and might vary) reveals a little of what is going on:
```
→ amixer -c 1 # With the system volume set to 20%
Simple mixer control 'PCM',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 255
  Mono:
  Front Left: Playback 46 [18%] [-41.80dB]
  Front Right: Playback 46 [18%] [-41.80dB]
→ amixer -c 1 # With the system volume set to 10%
Simple mixer control 'PCM',0
  Capabilities: pvolume
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 255
  Mono:
  Front Left: Playback 0 [0%] [-51.00dB]
  Front Right: Playback 0 [0%] [-51.00dB]
```
So for some reason `amixer` thinks there are 256 steps, that step 0 should be at 14%, and step 0 should be -51 dB, but it is actually silent. On the other hand this doesn't prevent the system from going above 100%, even though `amixer` stops at 255.

To try the `api.alsa.soft-mixer` solution, one needs a device name that alsa recognizes; I got one like this: `wpctl status` enumerates different objects using audio or video, and for me the only audio device was `45. Built-in Audio`. From this, `wpctl inspect 45` give a lot of information, but in it is the `device.name` field with the value `"alsa_card.pci-0000_00_1f.3"`, which I think is the one I want.

Then I had to create the path `/etc/wireplumber/main.lua.d/`, and the file `/etc/wireplumber/main.lua.d/99-alsa-volumes.lua`, with the contents:
```
table.insert (alsa_monitor.rules, {
    matches = { 
      {   
        { "device.name", 
          "matches", 
          "alsa_card.pci-0000_00_1f.3" 
        },
      },  
    },  
    apply_properties = { 
      ["api.alsa.soft-mixer"] = true,
    }   
})
```
after which I ran `sudo alsactl store 1`(where the number is again the number of the sound card, and might vary) and rebooted.

The result is that it would seem that possibly the builtin speakers are getting quieter more quickly as I approach -51 dB, but -51 is still at 14%, and  honestly I'm not sure that it's different than before, as I wasn't paying enough attention before I rebooted. For my headphones, nothing seems different, 20% is still reasonably loud, and audio abruptly stops at -51 dB as shown in pavucontrol. I'm not really sure how to tell whether or not wireplumber is even doing anything with that file; I also tried "alsa_card.*" for the device name, but that seems to be the same.

As far as the driver goes, you're right it's rather complicated; as far as having the volume range set there, I can't make much of it, but I didn't find anything to suggest that that's what's happening.

davidjo, 2024-09-06:
When I started this (now 2018/2019) Alsa was the main linux sound, pulseaudio maybe just starting (not default Ubuntu) and pipewire didnt exist.
This soft-mixer reminded me of things I was doing - these are for the .asoundrc text file in your home directory for Alsa control - theres a system wide version /etc/asoundrc if I remember right.

note that these define pseudo Alsa audio devices so should appear with aplay -l and be shown with amixer.

One of the first things was:
pcm.!default {
    type             plug
    slave.pcm        "softvol"
}

pcm.softvol {
    type             softvol
    slave {
        pcm          "hw:0,0"
    }
    control {
        name         "Master"
        card         0
    }
}

A later version seems more relevant - in particular the software volume controller
- softvol means that the audio waveform is being scaled by the Alsa driver rather 
than setting volume levels on the HDA chip.
This was all superseded by pulseaudio - which has its own way of doing things
- now of course superseded by pipewire.
plugin 1 is the one to focus on.

note that originally this driver used Apple ordef of left,left,right,right - at some point it
was changed in the driver to left,right,left,right - not sure exactly which coding was being used here
but the converter was attempting to map channels from plain stereo to 2 stereo channels ie 4 channels to drive all speakers
- again this now done in the hardware driver.

# define the apple hardware device
pcm.apple_hw {
	type hw
	card 0
	device 0
	# default rate and channel count but not necessarily respected
	channels 4
	rate 44100
	hint {
		show on
		description "Cirrus 8409 sound card"
	}
}

# plugin 1: create volume software controller
pcm.apple_softvol {
	type softvol
	slave {
		pcm "apple_hw"
	}
	control {
		name "Apple Cirrus 8409"
		card 0
		count 2 # stereo control
	}
	min_dB -51.0
	max_dB 0.0
	resolution 256
	hint {
		show on
		description "Apple software volume control"
	}
}

# plugin 2: map audio channels, convert stream and apply attenuation
pcm.apple_converter {
	type dshare
	slave {
		pcm "hw:0,0"
		#pcm "apple_hw"
		#pcm "apple_softvol"
		# convert audio to 4 channel, 32-bit little endian
		#channels 4
		#format "S32_LE"
	}
	#ttable {
	#	0.0 1.0
	#	1.1 1.0
	#	0.2 1.0
	#	1.3 1.0
	#}
	bindings {
		0 0
		1 1
		0 2
		1 3
	}
	hint {
		show on
		description "Apple channel converter"
	}
}

# plugin 3: apply a rate converter to force 44.1kHz
pcm.apple_rate {
	type rate
	slave {
		pcm "apple_converter"
		rate 44100
	}
	hint {
		show on
		description "Apple rate converter"
	}
}

# this is setup in /usr/share/alsa/alsa.conf
#defaults.pcm.dmix.max_periods 0
#defaults.pcm.dmix.channels 2
#defaults.pcm.dmix.rate 48000
#defaults.pcm.dmix.format "unchanged"
#defaults.pcm.dmix.card defaults.pcm.card
#defaults.pcm.dmix.device defaults.pcm.device

# well pigs - dmix can ONLY be connected to hardware device!!
# and the following does not work with > 2 channels - just first 1 channels copied
# why isnt there a simple downmix (vdownmix not good as 16 bit only and adds spatialization effect)?

pcm.apple_stereo {
	type plug
	slave {
		pcm "apple_converter"
		channels 2
		rate 44100
		format "S32_LE"
	}
	ttable {
		0.0 1.0
		1.1 1.0
	}
	hint {
		show on
		description "Apple stereo"
	}
}

#pcm.!default {
#	type dmix
#	rate 44100
#}

# plugin 4: apply mixing effects
#pcm.apple_mixer {
#	hint {
#		show on
#		description "Apple mixer"
#	}
#}

davidjo, 2024-09-07:
Unfortunately figuring out the various inter-dependencies between Alsa and pipewire (or pulse) is difficult to figure out.
What I read the .asoundrc as is that it set the max/min range to 0dB/-51dB (which to me seems significant because this is the volume it cuts out - unfortunately I cant remember where that came from - might try searching the issues - in some of the old ones there were various attempts  at .asoundrc files) in 256 steps - Im assuming you could change the min value to some other dB eg -61dB.
Its just a question of googling and trying things out I think - the question I think is what default volume range does pipewire assume if there is no hardware range reported (and I dont think the 8409 chip will report a volume range because its just acting as a digital stream converter).
Heres another idea - most HDA chips use 16 bit audio data. What Apple wanted to do was to use 24 bit audio data - which is why the 8409 is just a stream converter and the separate amps are 24 bit DAC amps (at least thats my guess).
Obviously 24 bit audio data has a larger range than 16 bit data - not sure what the dB conversion is.
On the other hand the range issue seems to be a general issue based on googling - but it could be thats because the problem occurs with pro Audio devices which are using 24 bits

ManeFunction, 2026-11-06:
Hello, guys! I've got an issue with extremely loud headphones and a dead zone on start, so I've applied [this solution](https://github.com/davidjo/snd_hda_macbookpro/pull/231), that is partially relevant with this issue. Hope this help.

huntekye, 2026-11-06:
I realized that I forgot to ever try something here: in the config file from davidjo, there's the software volume control:
```
# plugin 1: create volume software controller
pcm.apple_softvol {
	type softvol
	slave {
		pcm "apple_hw"
	}
	control {
		name "Apple Cirrus 8409"
		card 0
		count 2 # stereo control
	}
	min_dB -51.0
	max_dB 0.0
	resolution 256
	hint {
		show on
		description "Apple software volume control"
	}
}
```
Why does is say -51.0 here? if it were set lower, is that all it would take to solve the problem? I think at the time it seemed too obvious to be worth trying lol, but who knows.

Regarding headphones that seem too loud to be usable, I imagine that they have low impedance; one option would be to buy an impedance adapter. I got some for a device that has a fixed output volume, and it's not the best solution, but it works reasonably well. Impedance adapters are not super easy to find, I think I could only find them on AliExpress.

ManeFunction, 2026-11-07:
For the -51 dB limit my research got me to the softvol: It was an alsa-lib's built-in default for its software volume plugin. 
```
#define PRESET_RESOLUTION       256
#define PRESET_MIN_DB           -51.0
```
And the number itself likely came from calculations like this: 256 steps of 0.2 dB each. Step 0 is the bottom and step 255 is 0 dB, so the range is 255 x 0.2 = 51 dB. This is not a definitive true, it's just my assumptions.

About the high volume, yes, the low impedance was the first thing I've thought as well, but IMO the default use case should be reversed (in case of your ears and hardware safety): low impedance headphones should work out of the box is safe limits, and high impedance one should require additional hardware. But as we have enough power here to bump a high impedance headphones (I suppose, not tested), I've intentionally do not built my limitation within the driver itself, but made a property that everyone can set it up for his environment.

setanarut, 2026-11-09:
This isn't just about the iMac—could you update the title? On a 2016 MacBook Pro, the sound cuts out completely whenever the GNOME volume slider is set below 13%. Has anyone found a solution for this?

This issue occurs in the same way for both the headphones and the built-in speakers.

ManeFunction, 2026-11-10:
@setanarut I've fixed exactly this issue in [this PR](https://github.com/davidjo/snd_hda_macbookpro/pull/231), I've mentioned it a few messages above. Actually no code change needed, if your problem is only a silent zone at the beginning, it's solvable with a config only.

setanarut, 2026-11-10:
@ManeFunction This worked, thanks. I just hope my ears don't explode while I have the headphones on.

huntekye, 2026-11-10:
Okay I can confirm that this config file from @ManeFunction fixes the problem:
```
# ~/.config/wireplumber/wireplumber.conf.d/51-cs8409-volume.conf
monitor.alsa.rules = [
  {
    matches = [ { device.name = "alsa_card.pci-0000_00_1f.3" } ]
    actions = { update-props = { api.alsa.ignore-dB = true } }
  }
]
```
Adding that config and rebooting, I have (quiet) sound down to 1% volume (-116 dB).

Maybe this is still not addressing the real root of the problem, as I imagine that a config with `ignore` in the name should have a more direct solution. But unless someone want to go digging for it, I don't see a reason to not ship this config file in `/etc/`.

Regarding the excess volume with sensitive headphones, perhaps a more simple solution would be to adjust the master volume in the `alsamixer` TUI?