davidjo/snd_hda_macbookpro #136: Trouble installing `./install.cirrus.driver.sh` on 22.04.4 LTS on iMac 19,1 with `6.8.0-45-generic` kernel

dhajnes, 2024-10-10:
Hi, I was trying to go through your installation process and failed on 

```
Ubuntu linux kernel source not found in /usr/src: /usr/src/linux-source-6.8.0.tar.bz2
assuming the linux kernel source package is not installed
please install the linux kernel source package:
sudo apt install linux-source-6.8.0
```

I have tried to run the shell script as:
```
sudo ./install.cirrus.driver.sh -k 6.8.0-45-generic
```
because that's the kernel I have, but with the same output.

Then I tried to install the source for the wanted kernel and got:

```
sudo apt install linux-source-6.8.0

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-source-6.8.0
E: Couldn't find any package by glob 'linux-source-6.8.0'
```

Since I have messed up my Ubuntu install with fiddling with various drivers in the past, I rather open up an Issue hoping you could perhaps give me some informed suggestions.

NOTE: I also looked at your older branches and there is one for Ubuntu 20, the current version says its tested for 24. I am in the middle with 22 :/. 

Hardware and Firmware info:

**System Information:**
- **CPU**: Intel® Core™ i9-9900K @ 3.60GHz (16 cores)
- **RAM**: 40 GB
- **GPU**: AMD VEGA10 (LLVM 15.0.7, DRM 3.57)
- **Disk Space**: 500.3 GB
- **OS**: Ubuntu 22.04.4 LTS 64-bit
- **Desktop Environment**: GNOME 42.9 (Wayland)
- **Kernel Version**: 6.8.0-45-generic

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy

davidjo, 2024-10-10:
The issue is the script only works for the default Ubuntu kernel for whichever Ubuntu version is being used - which was 5.15 something for 22.04 from one of my old root drives - the  only Ubuntu kernel source package will be for 5.15.
The easiest hack is to set the isubuntu variable to 0 in the install script just before the if isubuntu -ge 1 - this will make the script use the mainline kernel distribution files - which Im guessing is what you are using as I dont think you are using an hwe kernel - which needs other tricks (this will download the full mainline kernel source (compressed) into the repo build directory).
I think this should just work then - I think I did use this hack when testing other kernel versions at one stage.

owingst, 2024-10-14:
This worked for me! Thanks!

dhajnes, 2024-10-24:
> The issue is the script only works for the default Ubuntu kernel for whichever Ubuntu version is being used - which was 5.15 something for 22.04 from one of my old root drives - the only Ubuntu [...]

Thank you, setting `isubuntu = 0` in the `.sh` file helped, installation went seemingly without problems, I now have sound after restart, I can also distinguish stereo channels. The bluetooth and AUX jack input also work.

What does not work is the integrated microphone, nor the microphone feature of my bluetooth Sony headphones. I think I have read in some file somewhere, that the microphone is not yet supported so I am closing this issue nevertheless.

Thanks for the driver, it helps a lot that at least I have the sound output.

closed 2024-10-24 by dhajnes

davidjo, 2024-10-24:
The integrated microphone sort of works on a macbook pro 14,3 (iMacs not clear) - but its at a very low volume so might think its not working - it also needs to be explicitly set in the audio settings - it really needs some amplification via a pipewire function (or pulseaudio or alsa).
As far as I know (as dont have a bluetooth headset) the bluetooth system is totally separate so is not dependent on this module at all - it goes through the bluetooth driver - which Im assuming a bluetooth microphone does also.