davidjo/snd_hda_macbookpro #138: No audio on `MacBookPro14,1` on Ubuntu 24.04 and 24.10

osalbahr, 2024-11-05:
Hi!

I am reporting that this driver resolved the "Dummy Output" into "Headphones" but I still do not hear sound. The 2017 MacBook Pro (Intel-based) is running Ubuntu 24.04 but I also tried the driver in a thumb drive with Ubuntu 24.10 
as a sanity check.

(this is mainly a tracking issue)

- First reported in: https://askubuntu.com/questions/1531936/sound-is-dummy-output-on-macbookpro14-1-i-e-macbook-pro-13-inch-2017-two
- Might be relevant: https://github.com/davidjo/snd_hda_macbookpro/issues/134

System info:

```console
$ fastfetch 
                            ....               orange@orange-MacBookPro14-1
              .',:clooo:  .:looooo:.           ----------------------------
           .;looooooooc  .oooooooooo'          OS: Ubuntu oracular 24.10 x86_64
        .;looooool:,''.  :ooooooooooc          Host: MacBookPro14,1 (1.0)
       ;looool;.         'oooooooooo,          Kernel: Linux 6.11.0-9-generic
      ;clool'             .cooooooc.  ,,       Uptime: 24 mins
         ...                ......  .:oo,      Packages: 1626 (dpkg), 12 (snap)
  .;clol:,.                        .loooo'     Shell: bash 5.2.32
 :ooooooooo,                        'ooool     Display (Color LCD): 2560x1600 @ 60Hz (as 1280x800) [Built-in]
'ooooooooooo.                        loooo.    DE: GNOME 47.0
'ooooooooool                         coooo.    WM: Mutter (Wayland)
 ,loooooooc.                        .loooo.    WM Theme: Yaru-dark
   .,;;;'.                          ;ooooc     Theme: Yaru-dark [GTK2/3/4]
       ...                         ,ooool.     Icons: Yaru-dark [GTK2/3/4]
    .cooooc.              ..',,'.  .cooo.      Font: Ubuntu Sans (11pt) [GTK2/3/4]
      ;ooooo:.           ;oooooooc.  :l.       Cursor: Yaru (24px)
       .coooooc,..      coooooooooo.           Terminal: GNOME Terminal 3.54.0
         .:ooooooolc:. .ooooooooooo'           Terminal Font: Ubuntu Sans Mono (13pt)
           .':loooooo;  ,oooooooooc            CPU: Intel(R) Core(TM) i5-7360U (4) @ 3.60 GHz
               ..';::c'  .;loooo:'             GPU: Intel Iris Plus Graphics 640 @ 1.00 GHz [Integrated]
                                               Memory: 3.59 GiB / 7.12 GiB (50%)
                                               Swap: 0 B / 4.00 GiB (0%)
                                               Disk (/): 10.93 GiB / 111.20 GiB (10%) - ext4 [External]
                                               Local IP (wlp2s0): 192.168.100.173/24 *
                                               Battery: 80% [Discharging]
                                               Locale: en_US.UTF-8

                                               ████████████████████████
                                               ████████████████████████
```

davidjo, 2024-11-05:
I am still running 24.04 - so no idea if 24.10 introduces other problems.
What sort of headphones are you using??
Apple defines 3 types of headsets, non-apple headphone (no mike), apple headphone (no mike) and apple headset (with mike).
if its a non-Apple headset with mike that might be an issue (I dont have such a device - all I have is non-Apple headphone and Apple headset).

davidjo, 2024-11-06:
Also, my main test was starting playing through speakers then plugging in headphones - does that work?

osalbahr, 2024-11-06:
I was referring to the built-in sound.

I reinstalled macOS and sound is still not working. So this is probably a hardware issue.

Thx for the sanity check!

closed 2024-11-06 by osalbahr