davidjo/snd_hda_macbookpro #140: Network is gone after install?

michaelahess, 2025-01-15:
I ran the commands on a 2017 Macbook Pro with a clean install of Ubuntu 23.0.4.1 LTS.

After reboot, I have sound, but the network adaptor is just gone.

I don't know what to do to fix this, short of a reinstall?

osalbahr, 2025-01-16:
Have you tried rebooting?

davidjo, 2025-01-16:
What do you mean by network adaptor - is this a plugin USB type network adaptor? the WiFi?

I cant see how at the moment the module could interfere with a network module - the internal audio system is completely independent of the network subsystem and/or usb subsystem.

All the module does in the end is copy the snd_hda_macbookpro.ko file to /lib/modules/{kernel version}/updates.
To remove it just delete the .ko file from updates and run  sudo depmod -a - this should restore your kernel to its state before running this repos install script.

michaelahess, 2025-01-16:
I think it's general stability with Ubuntu 24 on Mac, a few reboots did get wifi working again. Ethernet via USB-C hub did work. I also have stability issues with the display driver after suspend, BT doesn't work at all, all kinds of small things. But sound does work, so thank you very much, and sorry to report incorrectly!

closed 2025-01-16 by michaelahess