davidjo/snd_hda_macbookpro #141: [MBP 13,2] No sound

TheTollingBell, 2025-03-09:
After upgrade to kernel available (6.13.5.arch1-1) sound no longer functions:
- I'm using wireplumber
- I still have all the profile options available in `pavucontrol` 
- No input or output is working from internal speakers, microphone, or headphone jack.

I'd be happy to try anything or provide any logs to help triangulate the issue but I pretty much haven't touched my sound since setting it up so I'm not very knowledgable.

TheTollingBell, 2025-03-09:
Just tested using `aplay` and I still have no sound.

davidjo, 2025-03-09:
Currently Ubuntu is still on 6.8  24.04 (LTS) - looks like 24.10 might be using 6.11 but I dont have this installed.
Was the module build successful - no compile errrors - that has been the issue in the past with kernel version upgrades leading to failed module building because of kernel source code changes.
If you could check what happens running install.cirrus.driver.sh for compilation errors and report them here should be able to figure whats going on.

TheTollingBell, 2025-03-09:
Could this be the issue:
```
At main.c:171:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
```

huntekye, 2025-03-09:
FWIW, I'm also on 6.13.5-arch1-1, but on a macbookpro 14,1 , and for me audio is still working fine. My build log looks basically the same as TheTollingBell's, including the SSL error.

It seems interesting that you say you still see profiles in pavucontrol as if the driver was working correctly? Like for your output device it says "Built-in Audio Analog Stereo" with an option for "speakers" and volume control and another bar that shows the output volume intensity?

I have never had the pavucontrol recognize the output device correctly but not had working output, but I have always had more-or-less working input that has actually never worked, i.e., in pavucontrol my builtin microphone is recognized, and the input intensity is always very responsive if I tap on the microphone, for example, but I have never had any app actually receive input from the microphone... so if I understand correctly, it sounds like your case is similar, but with the input.

TheTollingBell, 2025-03-09:
I'm not seeing anything in pavucontrol and alsamixer really start bugging out whenever i select the internal soundcard.

huntekye, 2025-03-09:
Okay yeah no worries about your input for now; if some application tries to play audio to the speaks, does pavucontrol show the output intensity fluctuating like it thinks the audio is actually playing? 

In any case, you'll definitely want to try moving the module in `/usr/lib/modules/6.13.5-arch1-1/kernel/sound/pci/hda/snd-hda-codec-cs8409.ko.zst`, running `depmod -a`, and rebooting to make sure you're debugging the right module.

TheTollingBell, 2025-03-09:
That's what I'm doing right now. I'm pretty sure that's the issue since for the boots after the kernel update I'm no longer seeing `Primary patch_cs8409 NOT FOUND trying APPLE`.

TheTollingBell, 2025-03-10:
Just moved `/usr/lib/modules/6.13.5-arch1-1/kernel/sound/pci/hda/snd-hda-codec-cs8409.ko.zst`, ran `depmod -a`, however `Primary patch_cs8409 NOT FOUND trying APPLE` is still not showing up in logs and I'm still unable to use any internal input or output.

davidjo, 2025-03-10:
run lsmod (list of loaded kernel modules) and check for snd_hda_codec_cs8409 - if its not there you wont see the message - this means the module is not being loaded - look through the system logs for some error message about the module.

TheTollingBell, 2025-03-10:
No error message about `snd_hda_codec_cs8409` in logs and found line: `snd_hda_codec_cs8409   180224  0`.

TheTollingBell, 2025-03-10:
This was after moving `snd-hda-codec-cs8409`.

TheTollingBell, 2025-03-10:
Just searched my entire drive using `find` and grepped for cs8409. Only locations are where I moved the module to, the update file that your install script installed, and the `/sys/module/snd_hda_codec_cs8409` directory.

davidjo, 2025-03-10:
So Im not sure whats going on - that would definitely suggest its loading my version of the module - but then there should be something in the system logs.

What messages (not just errors) do you see in the system logs for snd_hda_codec_cs8409?

davidjo, 2025-03-11:
OK so that confirms - if you are not seeing that with 6.13 then either my module is not loaded or its being prevented from working (as lsmod does seem to say its loaded).
Im wondering if something has changed about tainting - if you dont sign a kernel module this "taints" the kernel (this is logged specifically in the system logs) - but previously it still ran.
Another option is somethings changed about how sound modules are being called ie its loading my module but for some reason its not getting called.

HDA (HIgh Definition Audio) is an Intel spec for generic audio chips (not just intel chips) - this is implemented as a 2 stage process in linux. First snd_hda_intel is started then specific drivers are loaded for the actual chipset involved eg realtek, the cs8409 etc and their startup function in the specific driver called. So maybe somethings going wrong here ie the module could be being loaded but its not being activated.

Also check the Makefile in the toplevel of the repo - it should have the CONFIG_MODULE_SIG_ALL=n line after install:.
If it does and you are seeing the crypto errors then something has definitely changed with signing.

huntekye, 2025-03-11:
For comparison, with working audio, this is what I see from `journalctl -g snd`:

```
ArchBookPro kernel: snd_hda_intel 0000:00:1f.3: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
ArchBookPro kernel: snd_hda_intel: Primary patch_cs8409
ArchBookPro kernel: snd_hda_intel: Primary patch_cs8409 NOT FOUND trying APPLE
ArchBookPro kernel: snd_hda_codec_cs8409 hdaudioC1D0: autoconfig for CS8409: line_outs=2 (0x24/0x25/0x0/0x0/0x0) type:speaker
ArchBookPro kernel: snd_hda_codec_cs8409 hdaudioC1D0:    speaker_outs=0 (0x0/0x0/0x0/0x0/0x0)
ArchBookPro kernel: snd_hda_codec_cs8409 hdaudioC1D0:    hp_outs=1 (0x2c/0x0/0x0/0x0/0x0)
ArchBookPro kernel: snd_hda_codec_cs8409 hdaudioC1D0:    mono: mono_out=0x0
ArchBookPro kernel: snd_hda_codec_cs8409 hdaudioC1D0:    inputs:
ArchBookPro kernel: snd_hda_codec_cs8409 hdaudioC1D0:      Internal Mic=0x44
ArchBookPro kernel: snd_hda_codec_cs8409 hdaudioC1D0:      Mic=0x3c
ArchBookPro kernel: snd_hda_intel: hda_generic_check_dyn_adc_switch shrinking
```
Despite seeing SSL errors when building the module, I don't see any messages about `snd_hda_codec_cs8409` resulting in a tainted kernel (`journalctl -g taint`), but I do see messages about other things tainting the kernel, e.g., the `facetimehd` module and ZFS when I had that for awhile.

Also here is `lsmod | grep snd` with working audio:

```
snd_seq_dummy          12288  0
snd_hrtimer            12288  1
snd_seq               131072  7 snd_seq_dummy
snd_seq_device         16384  1 snd_seq
snd_hda_codec_hdmi     98304  1
snd_soc_avs           241664  0
snd_soc_hda_codec      28672  1 snd_soc_avs
snd_hda_ext_core       36864  2 snd_soc_avs,snd_soc_hda_codec
snd_soc_core          450560  2 snd_soc_avs,snd_soc_hda_codec
snd_hda_codec_cs8409   180224  1
snd_compress           28672  2 snd_soc_avs,snd_soc_core
snd_hda_codec_generic   114688  1 snd_hda_codec_cs8409
ac97_bus               12288  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_hda_intel          65536  1
snd_intel_dspcfg       40960  2 snd_soc_avs,snd_hda_intel
snd_intel_sdw_acpi     16384  1 snd_intel_dspcfg
snd_hda_codec         212992  6 snd_hda_codec_generic,snd_soc_avs,snd_hda_codec_hdmi,snd_soc_hda_codec,snd_hda_codec_cs8409,snd_hda_intel
snd_hda_core          147456  8 snd_hda_codec_generic,snd_soc_avs,snd_hda_codec_hdmi,snd_soc_hda_codec,snd_hda_codec_cs8409,snd_hda_intel,snd_hda_ext_core,snd_hda_codec
snd_hwdep              20480  1 snd_hda_codec
snd_pcm               200704  8 snd_soc_avs,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_compress,snd_soc_core,snd_hda_core,snd_pcm_dmaengine
snd_timer              57344  3 snd_seq,snd_hrtimer,snd_pcm
snd                   155648  16 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_codec_cs8409,snd_hda_intel,snd_hda_codec,snd_timer,snd_compress,snd_soc_core,snd_pcm
soundcore              16384  1 snd
```
What's interesting here is that for me `snd_hda_codec_cs8409`, has a "used by" of 1 (`snd_hda_codec_generic`), whereas when TheTollingBell had it loaded and it had the right size, it still had a "used by" of 0.

TheTollingBell, 2025-03-11:
Reset `snd_hda_macbookpro` to latest the commit fully and checked makefile. It still has `CONFIG_MODULE_SIG_ALL=n` and the crypto errors are still showing up.

TheTollingBell, 2025-03-12:
NixOS/nixpkgs#375605 may be related to the issue here.

davidjo, 2025-03-12:
I have seen the following used for builds - adding  CONFIG_MODULE_SIG=  in addition to CONFIG_MODULE_SIG_ALL=
(dont know if leaving the =n off is significant or not).
Might try this.
I have seen some suggestions that signing has been changed at 6.13 but still not very clear - cant determine if CONFIG_MODULE_SIG_ALL means all modules now must be signed or not - and how you would know if your 6.13 kernel was built with this on or off.
Unfortunately, Im now using a new arm laptop so  now the old intel laptop is not really being used much - and because Im switching from intel to arm I have a lot of fixups to do - when available it also means Ill be using Asahi linux (which supposedly has a very good fix for the audio subsystem).
Ill see if I can try a mainline kernel 6.13 on the old laptop.

TheTollingBell, 2025-03-12:
Thanks, it'll be nice to know if this is exclusive to me.

TheTollingBell, 2025-03-12:
Actually, I have another 13,2 mbp to test this on, I'll check when I'm back home.

davidjo, 2025-03-13:
Well its not looking good for 6.13.
I used the mainline dpkg (Ive used this before) to install 6.13 and although it boots seemingly OK there is no keyboard (the trackpad and touch bar already need another external module for my 14.2).
Looking through the syslog/kern.log Im seeing error messages applespi spi-APP000D:00: Error writing to device: 00 00 00 00 and DMAR: [DMA Read NO_PASID] Request device [00:1e.3] fault addr 0xffffa000 [fault reason 0x06] PTE Read access is not set.
Also lsmod shows it as 0 ie not active.
applespi used to be another external module I built using another github repo but it has been in the kernel for a number of years now and works fine for 6.8.
(Also apparently 6.13 was apparently compiled with gcc-14 but 6.8 is gcc-13 so my machine refuses to compile 6.13 modules when trying using a usb keyboard).
Though that does beg the question why it seems to be working for huntekye.
Maybe theres some kernel command line config option that needs setting.

davidjo, 2025-03-13:
So I tried 6.11 - that does boot with keyboard working but I tried to add the trackpad/touchbar modules and they did not compile (does require gcc-14) - after trivial removal of compile errors they loaded but did not work so something more needs fixing.
Also tried base 6.13 install but same error with applespi as for 6.13.6.
Clearly there are a number of incompatible code changes going on but no clear idea how to fix yet.
Next step is probably to try and figure what patches arch is applying for 6.13.5-arch1-1 cf. mainline 6.13.5 - that might suggest how to fix the applespi issue.

TheTollingBell, 2025-03-31:
I'm bisecting the issue right now starting at 6.8.9. I'll get back to you once I find whats working.

davidjo, 2025-04-02:
So you had a version that did work - can you restore that??
You need to check the system logs for snd_hda_intel messages  and other snd_ messages - some thing must be going wrong.
One big question is why is lsmod giving 0 for snd_hda_codec_cs8409 - if it still is for these older systems then looking at the logs for some error message is needed.

Also I have to say I would get random errors every so often - in particular I found if it booted into linux directly (which it tended to do after an upgrade) random things would stop working - I always used efibootmgr to ensure it went back into refind (a mac boot helper - google for it) rather direct linux booting.

I also had an error message snd_hda_intel: UNKNOWN subsystem id .... which occasionaly would be logged with the device id (0x10138409 for me) rather than the subsystem id - again I have no idea why - a reboot would be needed to fix this.

TheTollingBell, 2025-06-04:
@davidjo I've tried restoring the kernel all the way back to 5.9 and the exact same problem is persisting so I suspect that this is not due to a kernel update but something else.

davidjo, 2025-06-06:
Sorry - had things to do.

My current guess is theres something left from the dkms which is somehow messing things up - although I dont have a lot of experience with dkms.

Do you have some backup of the system disk you can restore - not just the kernel - is the lsmod still showing 0 for snd_hda_macbookpro?

funsaized, 2026-08-24:
Confirmed working on a MacBookPro13,2 with Omarchy/Arch and kernel 7.1.8-arch1-3. Installed `linux-headers` + AUR `snd-hda-macbookpro-dkms-git`, applied the kernel 7.x DKMS fixes from #183, confirmed the module resolved to `/updates/dkms/snd-hda-codec-cs8409.ko.zst`, then rebooted. Internal speakers now work. (`macbook12-audio-driver` is for MacBook9,1/10,1, not this model.)