davidjo/snd_hda_macbookpro #142: complier differ from kernel even though they look the same

tkalant, 2025-03-18:
Hello. I am tryin this on macbook 2015, running Ubuntu 22.04.5 LTS. Even though I upgraded the gcc to 12, the installer gives me a warning that the kernel and compiler differ. The sound does not work. Here is what I see after running the installer.
```
linux-source-6.8.0/sound/pci/hda/
linux-source-6.8.0/sound/pci/hda/hda_intel.c
linux-source-6.8.0/sound/pci/hda/hda_cs_dsp_ctl.h
linux-source-6.8.0/sound/pci/hda/hda_beep.c
linux-source-6.8.0/sound/pci/hda/patch_cmedia.c
linux-source-6.8.0/sound/pci/hda/cs35l41_hda_property.c
linux-source-6.8.0/sound/pci/hda/cs35l56_hda_spi.c
linux-source-6.8.0/sound/pci/hda/Makefile
linux-source-6.8.0/sound/pci/hda/thinkpad_helper.c
linux-source-6.8.0/sound/pci/hda/patch_cs8409-tables.c
linux-source-6.8.0/sound/pci/hda/hda_auto_parser.h
linux-source-6.8.0/sound/pci/hda/cs35l41_hda_spi.c
linux-source-6.8.0/sound/pci/hda/hda_cs_dsp_ctl.c
linux-source-6.8.0/sound/pci/hda/patch_conexant.c
linux-source-6.8.0/sound/pci/hda/patch_cirrus.c
linux-source-6.8.0/sound/pci/hda/patch_cs8409.c
linux-source-6.8.0/sound/pci/hda/cs35l41_hda_property.h
linux-source-6.8.0/sound/pci/hda/hda_eld.c
linux-source-6.8.0/sound/pci/hda/hda_component.h
linux-source-6.8.0/sound/pci/hda/hda_intel_trace.h
linux-source-6.8.0/sound/pci/hda/cs35l56_hda_i2c.c
linux-source-6.8.0/sound/pci/hda/hda_hwdep.c
linux-source-6.8.0/sound/pci/hda/hda_beep.h
linux-source-6.8.0/sound/pci/hda/hda_tegra.c
linux-source-6.8.0/sound/pci/hda/Kconfig
linux-source-6.8.0/sound/pci/hda/patch_sigmatel.c
linux-source-6.8.0/sound/pci/hda/cirrus_scodec_test.c
linux-source-6.8.0/sound/pci/hda/cs35l41_hda_i2c.c
linux-source-6.8.0/sound/pci/hda/ca0132_regs.h
linux-source-6.8.0/sound/pci/hda/hda_bind.c
linux-source-6.8.0/sound/pci/hda/cs35l56_hda.h
linux-source-6.8.0/sound/pci/hda/tas2781_hda_i2c.c
linux-source-6.8.0/sound/pci/hda/patch_ca0132.c
linux-source-6.8.0/sound/pci/hda/hda_auto_parser.c
linux-source-6.8.0/sound/pci/hda/cs35l41_hda.h
linux-source-6.8.0/sound/pci/hda/patch_realtek.c
linux-source-6.8.0/sound/pci/hda/patch_si3054.c
linux-source-6.8.0/sound/pci/hda/cs35l41_hda.c
linux-source-6.8.0/sound/pci/hda/hp_x360_helper.c
linux-source-6.8.0/sound/pci/hda/hda_local.h
linux-source-6.8.0/sound/pci/hda/hda_controller.c
linux-source-6.8.0/sound/pci/hda/hda_controller.h
linux-source-6.8.0/sound/pci/hda/patch_ca0110.c
linux-source-6.8.0/sound/pci/hda/hda_sysfs.c
linux-source-6.8.0/sound/pci/hda/patch_hdmi.c
linux-source-6.8.0/sound/pci/hda/patch_analog.c
linux-source-6.8.0/sound/pci/hda/ideapad_s740_helper.c
linux-source-6.8.0/sound/pci/hda/hda_controller_trace.h
linux-source-6.8.0/sound/pci/hda/hda_generic.c
linux-source-6.8.0/sound/pci/hda/cirrus_scodec.h
linux-source-6.8.0/sound/pci/hda/patch_cs8409.h
linux-source-6.8.0/sound/pci/hda/hda_codec.c
linux-source-6.8.0/sound/pci/hda/hda_jack.h
linux-source-6.8.0/sound/pci/hda/cirrus_scodec.c
linux-source-6.8.0/sound/pci/hda/cs35l56_hda.c
linux-source-6.8.0/sound/pci/hda/patch_via.c
linux-source-6.8.0/sound/pci/hda/hda_proc.c
linux-source-6.8.0/sound/pci/hda/hda_intel.h
linux-source-6.8.0/sound/pci/hda/hda_generic.h
linux-source-6.8.0/sound/pci/hda/hda_jack.c
Kernel version later than implemented version - there may be build problems
patching file patch_cs8409.c
Hunk #1 succeeded at 1445 (offset 175 lines).
Hunk #2 succeeded at 1454 (offset 175 lines).
Hunk #3 succeeded at 1483 (offset 175 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 304 (offset 2 lines).
Hunk #3 succeeded at 331 (offset 2 lines).
Hunk #4 succeeded at 367 (offset 2 lines).
patching file patch_cirrus_apple.h
make -C /lib/modules/6.8.0-52-generic/build CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1 -Wno-unused-variable -Wno-unused-function" M=/home/user/snd_hda_macbookpro/build/hda modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-52-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  CC [M]  /home/user/snd_hda_macbookpro/build/hda/patch_cs8409.o
  CC [M]  /home/user/snd_hda_macbookpro/build/hda/patch_cs8409-tables.o
  LD [M]  /home/user/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.o
  MODPOST /home/user/snd_hda_macbookpro/build/hda/Module.symvers
  CC [M]  /home/user/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.mod.o
  LD [M]  /home/user/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.ko
  BTF [M] /home/user/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.ko
Skipping BTF generation for /home/user/snd_hda_macbookpro/build/hda/snd-hda-codec-cs8409.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-52-generic'
make INSTALL_MOD_DIR=updates -C /lib/modules/6.8.0-52-generic/build M=/home/user/snd_hda_macbookpro/build/hda CONFIG_MODULE_SIG_ALL=n modules_install
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-52-generic'
  INSTALL /lib/modules/6.8.0-52-generic/updates/snd-hda-codec-cs8409.ko
  SIGN    /lib/modules/6.8.0-52-generic/updates/snd-hda-codec-cs8409.ko
  DEPMOD  /lib/modules/6.8.0-52-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-52-generic'
depmod -a

contents of /lib/modules/6.8.0-52-generic/updates
total 2532
-rw-r--r-- 1 root root 2588800 Mär 18 15:28 snd-hda-codec-cs8409.ko
```
Any ideas on how to fix this? Thank you!

davidjo, 2025-03-18:
The warning is just a warning - the last  few lines show that the module was built and placed into the correct kernel module directory (updates) - this is not why sound is not working.

More information is needed to figure why audio is not working.
First, what exact model are you using - under MacOS you need to run System Report and get the model number (my model number is for example 14,3). This module only works for certain specific models and I only know about the 14,3 module as its the only one I have - other people have loaded it for other models (see other issues) (the machine MUST have the 8409 chip for this module to work).
The system logs should show what is going on when the sound modules are loaded.

I also dont understand using the 6.8 kernel  for Ubuntu 22.04 - from an old boot disk for 22.04 I only saw kernels up to 5.15.
So Im assuming you are loading eg a mainline kernel (which is likely the source of the compiler warning - that would have been compiled with some standard gcc-12 whereas ubuntu is using ubuntus gcc-12 - which apparently are considered different by the kernel build system.

Can you not upgrade to 24.04 which does use the 6.8 kernel - but ubuntus version so shouldnt have above warning?

tkalant, 2025-03-19:
Hello David, 
Thank you for responding. 
I should have checked these beforehand - my apologies.
My model is macbook8, and it doesn't have the HDA chip - it has a built-in audio system. 
I searched a bit to find drivers for macbook8 with internal audio, it seems like there's not a solution for macbook8 with newer versions of ubuntu.
In the meantime, as I found on [Dunedan](https://github.com/Dunedan/mbp-2016-linux?tab=readme-ov-file#audio-input--output)'s repo, I checked with a pair of wired headphones, and the audio jack does work. So that's at least good. 
If you know of any other solutions, please let me know. 
Thank you again.

davidjo, 2025-03-25:
This is what I remember when I was looking for audio solutions - I also found Dunedan.
Now you dont have the 8409 HDA chip but you do have an HDA chip - but much more standard.
However Apple being Apple they use a proprietary version of it - one thing this allows them to do is add so called vendor commands - which are "useful" because there is no documentation of those commands.
There were a number of discussions of this machine on linux kernel lists- try looking for kernel bugzilla for bug 110561.
Curiously theres more info if you attempt to get the Bootcamp disk (for running Windows on the Mac) - in particular the cs4208_...inf files.