davidjo/snd_hda_macbookpro #144: Wifi is gone after install

gabrielsallesrg, 2025-03-31:
This issue was described [here](https://github.com/davidjo/snd_hda_macbookpro/issues/140) but no solution was mentioned apart from rebooting multiple times.
I'm using Lubuntu (24.04 and then 24.10)
To fix it, right after installing **snd_hda_macbookproI**m I run the following command:

`modprobe brcmfmac`

I'm not sure if this needs to be run with sudo, as I run both commands using a script.

davidjo, 2025-03-31:
modprobe would need sudo (or running as root) - maybe the script does it (unless something else in the script is fixing the issue).

I dont think this has anything to do with this module but something missing in the linux pre-boot setup.
For me I found that if I booted into linux (without either power off or booting into macos) a few times randomly at some point things would stop working. This would be fixed by either booting into macos or full power on/off.
(continuously booting into linux for me would happen eg if I was trying different code versions for the sound module - but obviously after figuring this boot problem I avoided such multiple boots).
My best guess was that at some point something got corrupted in the early boot stage memory (ie pre linux/macos) (possibly the nvram) as linux relied on data being set up by eg a boot into macOS - the "bios/uefi" in macs does load some drivers.

It may depend on how you are booting - I  switched to using something called refind for more control.
Another thing that happened is that after a linux install/upgrade the machine would boot directly into linux - which would just about guarantee the above boot issue - I had to keep using efibootmgr to ensure it would boot back into refind.

Wireless was also iffy - for a long time I had fixups to the brcmfmac firmware/setup as suggested by googling - although I think this is not needed for current installations - well I dont seeing being used in my current setup.

gabrielsallesrg, 2025-03-31:
The only OS that I have on this computer is Linux (Lubuntu), so I would assume that there is no relation with MacOS.
What I can say for sure is that everytime the Kernel is updated, I lose sound but the Wifi works.
When I get to this point, I can follow 2 paths:
Path 1:
1. Install **snd_hda_macbookpro**
2. Reboot
3. Now the speakers work, but not the Wifi
4. I run **sudo modprobe brcmfmac**
5. Reboot
6. Now Wifi and speakers work

Path 2:
1. Install **snd_hda_macbookpro** and right after run **sudo modprobe brcmfmac** (I use a script for that so I don't need to type everything)
2. Reboot
3. Wifi and speakers work

davidjo, 2025-03-31:
Ah - yes - every time linux is updated to a new os version you need to rebuild the sound module (kernel modules only work for a specific kernel version) - you could implement dkms to do the rebuild automatically (and if you look through the issues this has been discussed and working dkms systems suggested) - however I do not have this in my base repository as most of the time for me the dkms build would fail because new source changes were needed to the sound module.

Whats going on with wifi I have no idea and I have never seen anything like that.
Check with lsmod if brcmfmac is loaded before doing the sound module install - and then again after the sound install script has run.
Also check the system logs - should be indicating for both modules when they are loaded or unloaded.

As I said there are drivers loaded at early boot stage (ie before either linux or macos kernels have been loaded) - these are the drivers Im was talking about.

davidjo, 2025-04-03:
Sorry - I do have a fixup for WiFi - I have a file brcmfmac43602-pcie.txt (google for it) - without this file my wifi didnt work.