davidjo/snd_hda_macbookpro #147: Mic working until I connect headphones

scuty2000, 2025-05-14:
Hi, as mentioned in the title, I got audio working on my macbook pro late 2016 with touchbar, and also the mic!. But if I plug in my headphones (without a mic) the internal microphone stops working. Has anyone had this problem - and in case how did you solve it? Thanks in advance (also, thanks for this fix to the author)!

davidjo, 2025-05-14:
Sorry - this is not an area I have spent a lot of time on (I was surprised I managed to get the internal mike working at all - my main goal was to get internal speakers working).

I suspect theres some logic issue - although the headset mike should be controlled by have_mike - the presence of a headset determines the jack_present variable - Im guessing somewhere Im either not doing something or incorrectly doing something under the jack_present variable being set but the have_mike variable not set  - theres quite a bit of code under these variables.

Using the internal mike and a headset is not a combination I tried.

scuty2000, 2025-05-16:
Hi, I can confirm that it headset is reported not to have a mike (`[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_headset_type_detect_event headset does not have mike!!`). Also here is the full log for when I plug in the headphones:

```bash
[mag16 09:50] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_jack_unsol_event UNSOL 0xdc000002 tag 0x37
[  +0,000018] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_unsol_event_handler
[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_unsol_event_handler - event unsol_res 0xdc000002
[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_unsolicited_response - NOT UNSOL BLOCKED
[  +0,001213] snd_hda_codec_cs8409 hdaudioC0D0: command read_gpio_status_check 2
[  +0,000005] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_external_device_unsolicited_response - interrupt set
[  +0,000001] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_read_status_and_clear_interrupt start
[  +0,000134] snd_hda_codec_cs8409 hdaudioC0D0: command read_gpio_status_check 2
[  +0,000184] snd_hda_intel: cs_8409_apple_resume
[  +0,000021] snd_hda_intel: cs_8409_apple_init
[  +0,000001] snd_hda_intel: end cs_8409_apple_init
[  +0,000166] snd_hda_codec_cs8409 hdaudioC0D0: cs42l83_read_status_and_clear_interrupt start
[  +0,000003] snd_hda_intel: i2cRead 0x0090 0x1b7b: 1
[  +0,001434] snd_hda_intel: end cs_8409_apple_resume
[  +0,000003] snd_hda_intel: cs 8409 exec verb pin sense 0x003c 0
[  +0,000002] snd_hda_intel: cs 8409 exec verb pin sense 0x002c 0
[  +0,000503] snd_hda_intel: cs_8409_apple_check_power_status
[  +0,000124] snd_hda_intel: cs_8409_apple_check_power_status
[  +0,000223] snd_hda_intel: cs_8409_apple_check_power_status
[  +0,000191] snd_hda_intel: cs_8409_apple_check_power_status
[  +0,000305] snd_hda_intel: i2cRead 0x0090 0x1b7b:  0x7b40 end
[  +0,000003] snd_hda_intel: i2cRead 0x0090 0x1b7c: 1
[  +0,001958] snd_hda_intel: i2cRead 0x0090 0x1b7c:  0x7c00 end
[  +0,000005] snd_hda_intel: i2cRead 0x0090 0x1308: 1
[  +0,001974] snd_hda_intel: i2cRead 0x0090 0x1308:  0x0800 end
[  +0,000003] snd_hda_intel: i2cRead 0x0090 0x1301: 1
[  +0,001955] snd_hda_intel: i2cRead 0x0090 0x1301:  0x0100 end
[  +0,000003] snd_hda_intel: i2cRead 0x0090 0x1302: 1
[  +0,002039] snd_hda_intel: i2cRead 0x0090 0x1302:  0x0200 end
[  +0,000002] snd_hda_intel: i2cRead 0x0090 0x1303: 1
[  +0,001956] snd_hda_intel: i2cRead 0x0090 0x1303:  0x030c end
[  +0,000002] snd_hda_intel: i2cRead 0x0090 0x1304: 1
[  +0,001977] snd_hda_intel: i2cRead 0x0090 0x1304:  0x0400 end
[  +0,000003] snd_hda_intel: i2cRead 0x0090 0x1305: 1
[  +0,002018] snd_hda_intel: i2cRead 0x0090 0x1305:  0x0500 end
[  +0,000002] snd_hda_intel: i2cRead 0x0090 0x130b: 1
[  +0,001977] snd_hda_intel: i2cRead 0x0090 0x130b:  0x0b60 end
[  +0,000003] snd_hda_intel: i2cRead 0x0090 0x130d: 1
[  +0,001997] snd_hda_intel: i2cRead 0x0090 0x130d:  0x0d01 end
[  +0,000002] snd_hda_intel: i2cRead 0x0090 0x130e: 1
[  +0,001998] snd_hda_intel: i2cRead 0x0090 0x130e:  0x0e00 end
[  +0,000002] snd_hda_intel: i2cRead 0x0090 0x130f: 1
[  +0,002019] snd_hda_intel: i2cRead 0x0090 0x130f:  0x0f08 end
[  +0,000003] snd_hda_codec_cs8409 hdaudioC0D0: cs42l83_read_status_and_clear_interrupt end
[  +0,000003] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_read_status_and_clear_interrupt - UNSOL status 0x00400000
[  +0,000036] snd_hda_codec_cs8409 hdaudioC0D0: command read_gpio_status_check 3
[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_read_status_and_clear_interrupt - interrupt 1 clear
[  +0,000001] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_read_status_and_clear_interrupt 0x00400000 end
[  +0,000039] snd_hda_codec_cs8409 hdaudioC0D0: cs42l83_disambiguate_ur_from_int start
[  +0,000001] snd_hda_intel: i2cRead 0x0090 0x1b79: 1
[  +0,001915] snd_hda_intel: i2cRead 0x0090 0x1b79:  0x79a0 end
[  +0,000003] snd_hda_intel: i2cRead 0x0090 0x1b7a: 1
[  +0,001935] snd_hda_intel: i2cRead 0x0090 0x1b7a:  0x7aff end
[  +0,000002] snd_hda_intel: i2cRead 0x0090 0x131b: 1
[  +0,001998] snd_hda_intel: i2cRead 0x0090 0x131b:  0x1b03 end
[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs42l83_disambiguate_ur_from_int end
[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_external_device_unsolicited_response - UNSOL interrupt 0x00400000
[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_external_device_unsolicited_response - UNSOL disambig  0x00a0ff03
[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_external_device_unsolicited_response - UNSOL masked    0x00000000
[  +0,000001] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_external_device_unsolicited_response - UNSOL unmasked  0x00400000
[  +0,000001] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_external_device_unsolicited_response - UNSOL number interrupt actions 1
[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_interrupt_action - plug in
[  +0,000001] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_headset_plugin_event pre  plugin handle_detect
[  +0,000001] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_plugin_handle_detect start
[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs42l83_headset_button_detect_interrupts_off
[  +0,000046] snd_hda_intel: i2cRead 0x0090 0x1b7c: 1
[  +0,002229] snd_hda_intel: i2cRead 0x0090 0x1b7c:  0x7c00 end
[... 200 lines cut ...]
[  +0,000003] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_set_power_state end
[  +0,000001] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_plugin_event_continued headset_plugged_in 1
[... cut ...]
[  +0,000001] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_unsol_event_handler - event unsol_res 0xdc000003
[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_unsolicited_response - NOT UNSOL BLOCKED
[  +0,000038] snd_hda_codec_cs8409 hdaudioC0D0: command read_gpio_status_check 3
[  +0,000002] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_external_device_unsolicited_response - interrupt clear
[  +0,000001] snd_hda_codec_cs8409 hdaudioC0D0: cs_8409_cs42l83_unsol_event_handler end
[  +1,180003] snd_hda_intel: cs_8409_apple_suspend
```

davidjo, 2025-05-17:
Thats great.

Now  need to figure exactly what order you do things.
My primary testing was as follows: boot, start playing audio through speakers, plugin headphones/set (audio switches to headset) unplug headphones (audio switches back to speakers).

One of the issues maybe that the audio is not being switched properly (or being switched when it shouldnt) on headset insertion - what I discovered is this requires a special system call to update the user level audio data - as shown by the settings option for the system settings GUI interface of Kubuntu - you might want to see if this is changing when you plugin - should have options for the output audio and the input audio.

By the way one of the reasons I didnt bother too much with headset/headphones is because usb headsets/headphones are nothing to do with this module and just work.

BrewCoffeeAddict, 2025-07-13:
Hello @davidjo,

I believe that I have tracked down the issue. While looking at the sequence of event, I have noticed that the `cs_8409_inputs_power_nids_off` function is called when plugging headphones without microphone under `!(spec->have_mike)` but also when unplugging the headphones without microphone. 
When I mask out these calls, then the internal microphone keep working when I plug and unplug my headphones.
Please see code changes below:

```
diff --git a/patch_cirrus/patch_cirrus_real84.h b/patch_cirrus/patch_cirrus_real84.h
index 76de5e8..09f6857 100644
--- a/patch_cirrus/patch_cirrus_real84.h
+++ b/patch_cirrus/patch_cirrus_real84.h
@@ -4821,7 +4821,7 @@ static void cs_8409_plugin_event_continued(struct hda_codec *codec)
                 {
 
                         // ensure the intmike/linein nids are powered off
-                        cs_8409_inputs_power_nids_off(codec);
+                        //cs_8409_inputs_power_nids_off(codec);
 
 
                         retval = cs42l83_headphone_sense(codec);
@@ -5615,10 +5615,10 @@ static void cs_8409_unplug_handle_disconnect(struct hda_codec *codec)
                 //unplug_sync_converters2_on(codec);
 
 
-                if (!(spec->have_mike))
-                {
-                        cs_8409_inputs_power_nids_off(codec);
-                }
+                //if (!(spec->have_mike))
+                //{
+                //        cs_8409_inputs_power_nids_off(codec);
+                //}
 
                 // and another headphone sense
 
@@ -5659,13 +5659,13 @@ static void cs_8409_unplug_handle_disconnect(struct hda_codec *codec)
                                 mycodec_info(codec, "cs_8409_unplug_handle_disconnect headset have mike end\n");
                         }
 
-                        if (spec->playing)
-                        {
-                                if (!(spec->have_mike))
-                                {
-                                        cs_8409_inputs_power_nids_off(codec);
-                                }
-                        }
+                        //if (spec->playing)
+                        //{
+                        //        if (!(spec->have_mike))
+                        //        {
+                        //                cs_8409_inputs_power_nids_off(codec);
+                        //        }
+                        //}
 
                         if (spec->have_mike)
                                 cs_8409_inputs_power_nids_off(codec);
```

The only doubt that I have is that the  cs_8409_inputs_power_nids_off function does two things:
```
        hda_set_node_power_state(codec, spec->intmike_adc_nid, AC_PWRST_D3);
        hda_set_node_power_state(codec, spec->linein_amp_nid, AC_PWRST_D3);
```

Should `linein_amp_nid` still be disabled when plugging/unplugging an headphone without microphone?
Also, I only have a headphone without microphone so can't test all the cases.

Please let me know what you think.

BrewCoffeeAddict, 2025-07-15:
Thank you @davidjo. 
No problem, I was just testing a headset with microphone because I found one but I don't really need one. The main thing for me is to have the internal microphone working.
I am really grateful for these drivers, otherwise, I would have no sound on my laptop when booting on linux making it not really usable. 
I do appreciate that you did not specifically look into the microphone but very glad that it's working

Ah, thank you for the clarification, got it now.
I tried restoring line 4822 `cs_8409_inputs_power_nids_off` in function `cs_8409_plugin_event_continued` 
Adding a else with power on in the next `if spec->mike`, like this:

```
diff --git a/patch_cirrus/patch_cirrus_real84.h b/patch_cirrus/patch_cirrus_real84.h
index 76de5e8..2ed49f0 100644
--- a/patch_cirrus/patch_cirrus_real84.h
+++ b/patch_cirrus/patch_cirrus_real84.h
@@ -4870,6 +4870,10 @@ static void cs_8409_plugin_event_continued(struct hda_codec *codec)
                                 cs_8409_headset_mike_buttons_enable(codec);
 
                         }
+                        else{
+                                // turn on the intmike/linein nids
+                                cs_8409_inputs_power_nids_on(codec);
+                        }
                 }
 
         }

```

But this doesn't work and the internal microphone goes and don't come back when I plug my headphone without microphone.
I suspect that it is possibly an issue with `cs_8409_inputs_power_nids_on`. 
`cs_8409_inputs_power_nids_off` does turn off the internal microphone but `cs_8409_inputs_power_nids_on` doesn't seems to bring it back.
So it work for headphone without microphone, if I never run `cs_8409_inputs_power_nids_off` like in my current changes (taking into account what you mentioned above on line 5615 and 5661 blocks)

```
diff --git a/patch_cirrus/patch_cirrus_real84.h b/patch_cirrus/patch_cirrus_real84.h
index 76de5e8..1689bce 100644
--- a/patch_cirrus/patch_cirrus_real84.h
+++ b/patch_cirrus/patch_cirrus_real84.h
@@ -4819,9 +4819,16 @@ static void cs_8409_plugin_event_continued(struct hda_codec *codec)
                 // - headset phase of 2 or more means post boot headset plugin
                 if (spec->headset_phase >= 2)
                 {
-
-                        // ensure the intmike/linein nids are powered off
-                        cs_8409_inputs_power_nids_off(codec);
+                        if (spec->have_mike)
+                        {
+                                // ensure the intmike/linein nids are powered off
+                                cs_8409_inputs_power_nids_off(codec);
+                        }
+                        else
+                        {
+                                // Turn off line in
+                                hda_set_node_power_state(codec, spec->linein_amp_nid, AC_PWRST_D3);
+                        }
 
 
                         retval = cs42l83_headphone_sense(codec);
@@ -5615,10 +5622,16 @@ static void cs_8409_unplug_handle_disconnect(struct hda_codec *codec)
                 //unplug_sync_converters2_on(codec);
 
 
-                if (!(spec->have_mike))
+                if (spec->have_mike)
                 {
+                        // Turn off internal mic and line in
                         cs_8409_inputs_power_nids_off(codec);
                 }
+                else
+                {
+                        // Turn off line in
+                        hda_set_node_power_state(codec, spec->linein_amp_nid, AC_PWRST_D3);
+                }
 
                 // and another headphone sense
 
@@ -5661,14 +5674,27 @@ static void cs_8409_unplug_handle_disconnect(struct hda_codec *codec)
 
                         if (spec->playing)
                         {
-                                if (!(spec->have_mike))
+                                if (spec->have_mike)
                                 {
-                                        cs_8409_inputs_power_nids_off(codec);
+                                        // Turn on internal mic and line in
+                                        cs_8409_inputs_power_nids_on(codec);
+                                }
+                                else{
+                                        // Turn on line in
+                                        hda_set_node_power_state(codec, spec->linein_amp_nid, AC_PWRST_D0);
                                 }
                         }
 
                         if (spec->have_mike)
-                                cs_8409_inputs_power_nids_off(codec);
+                        {
+                                // Turn on internal mic and line in
+                                cs_8409_inputs_power_nids_on(codec);
+                        }
+                        else
+                        {
+                                // Turn on line in
+                                hda_set_node_power_state(codec, spec->linein_amp_nid, AC_PWRST_D0);
+                        }
 
 
                         mycodec_info(codec, "cs_8409_unplug_handle_disconnect headset disable start\n");
```

Sadly, I don't have a usb headset. I am happy with these changes if I can plug/unplug an headphone without microphone without affecting the internal microphone.

No, on Mac OS X, everything works automatically. In the sounds setting, for headphone without microphone, the output change automatically from speaker to headphone and internal microphone is not affected.
With a headphone with microphone, the output change automatically from speaker to headphone and input changes automatically from internal microphone to headset microphone.

davidjo, 2025-07-16:
Something I discovered is Apple does a lot of power offs/power on which I think are purely to reduce power consumption - I skipped some of these but wasnt sure about others so they were left in.

Im guessing the power off of the input nids likely resets them as well (another thing that isnt clear - what exactly is saved during power "offs" (obviously the 8409 chip isnt totally powered off - these power functions just refer to a specific nid block - nids are an HDA architecture feature).
So it looks as though the 4821 power off resets the nid setup so that needs to be redone - input setup (look for functions with capture in their name) is done when linux sends a capture start message (and this could be another difference - when OSX sends capture starts vs when linux does). The primary capture setup function is cs_8409_capture_real.
This calls functions (these are all my names by the way) like cs_8409_intmike_format_setup_enable, cs_8409_intmike_volume_setup, cs_8409_intmike_stream_on_nid, cs_8409_intmike_volume_unmute.
This should probably only be done if the spec->capturing is greater than 0 (like playing this indicates if capturing is currently being done).
Could just not do the power off at 4821 but I think then if you plug in a headset with mike it will turn off the internal mike - so you would need to re-turn the internal mike back on on unplug so just not powering wont work for multiple plugins - cant remember now if I turn on the intmike on unplug.
Just thought another option is to just not do the 4821 turn off if spec->capturing is greater than 0.

By the way if you really want to fix this you could have my fixedup OSX (NOTE!!) kernel module to log HDA commands. However, this only works on High Sierra (my old machine never booted anything later because of Apples messing with the OS to prevent users doing useful things) - you could put High Sierra on a USB boot disk  that still works for these old intel machines.
This is pure assembly so just needs the assembler to build.
Also the log files are huge (>10000 lines eg for simply plugging in headset while playing - a lot of vendor commands expand to many HDA verbs).

BrewCoffeeAddict, 2025-07-16:
Hello @davidjo,

Yes, your suggestion of using ` spec->capturing>0` does work for headphone without microphone line 4821, I did this:

```
diff --git a/patch_cirrus/patch_cirrus_real84.h b/patch_cirrus/patch_cirrus_real84.h
index 76de5e8..ab1ea8f 100644
--- a/patch_cirrus/patch_cirrus_real84.h
+++ b/patch_cirrus/patch_cirrus_real84.h
@@ -4819,9 +4819,16 @@ static void cs_8409_plugin_event_continued(struct hda_codec *codec)
                 // - headset phase of 2 or more means post boot headset plugin
                 if (spec->headset_phase >= 2)
                 {
-
-                        // ensure the intmike/linein nids are powered off
-                        cs_8409_inputs_power_nids_off(codec);
+                        if (spec->capturing > 0)
+                        {
+                                // Turn off line in
+                                hda_set_node_power_state(codec, spec->linein_amp_nid, AC_PWRST_D3);
+                        }
+                        else
+                        {
+                                // ensure the intmike/linein nids are powered off
+                                cs_8409_inputs_power_nids_off(codec);
+                        }
```
Was your worry about checking `spec->have_mike` before running `cs42l83_headphone_sense`?

Sure, I would be interested to have a look at your fixedup OSX kernel module to log HDA commands. It might help debug the headset with microphone situation. I have never done something like that before but it sound interesting.
Yes, Apple has a track record of messing up the OS to prevent doing useful thing. For the facetimeHD driver, it was possible to get the firmware directly from OSX in the past but they have obviously removed this in newer version.
Talking about the facetimehd drivers (https://github.com/patjak/facetimehd), I know they have looked at the windows bootcamp driver for extracting the sensor calibration files (https://github.com/patjak/facetimehd/wiki/Extracting-the-sensor-calibration-files). Is that something that could help for the microphone?

BrewCoffeeAddict, 2025-07-21:
No problem at all @davidjo. Yes, I suspect everything is more work on the M chips being all proprietary.
  
Sorry, I made a lot for changes this weekend. Just for the background, my kernel got updated from 6.8.0-60 to 6.8.0-64. As a result, the driver completely stopped working. 
I was worried that it was because of these recent changes to keep the internal mike working when plugging an headphone without mike.
Turned out the issue is probably because of the difference between main linux kernel and Ubuntu kernel (I saw your comment about this on another issue). My distro is Zorin OS 17 based on Ubuntu 22.04 LTS but somehow the drivers worked without the ubuntu flag set in the script until now (you mentioned this flag in another issue).
Setting the isubuntu flag to 1 solved my issue. So, I updated the code to support other ubuntu based distros and not only linux mint.

I also found out from your comments how to clean up the driver by deleting the ko file, this was something that was puzzling me until now as I could never completely clean up the driver.

I also discovered that some distros like mine, update the kernel to the next LTS to support newer hardware, as a result it is more complicated to retrieve the  kernel source so I added a comment on how to do that. All in this PR:  https://github.com/davidjo/snd_hda_macbookpro/pull/154  

Then, I actually found some issues with the dkms fix (sorry). Basically, as you rightly wanted to make sure, sadly a typo broke the install.cirrus.driver.sh when running without dkms. I fixed that now. I also discovered that even after deleting the driver from dkms and cleaning up, the driver was still actif. So I added a workaround to delete the ko file and to run depmod when cleaning up. I did a lot of testing so I am confident that the code is working well now. All added to the PR: https://github.com/davidjo/snd_hda_macbookpro/pull/119

Finally, I raised a PR for the workaround to keep the internal mike working with headphone: https://github.com/davidjo/snd_hda_macbookpro/pull/155
But as you said, I think this is more a workaround than a proper fix. To properly fix it, looking into the fixedup OSX kernel module is more of a solution.

I combined all these three PRs in a branch on my fork that I have thoroughly tested over the weekend so I am quite confident about the changes.