davidjo/snd_hda_macbookpro #153: Help converting from https://github.com/egorenar/snd-hda-codec-cs8409 to This on Nixos

snoweuph, 2025-07-15:
Hi,

Im stupid and know basicly nothing about Drivers.
ive recently moved from fedora to NixOS on my 2017 Macbook Pro.
With your Patchset on Fedora, I had Speakers and Microphone working.
Ive found somebodies config for https://github.com/egorenar/snd-hda-codec-cs8409 on NixOS and got it working, but only speakers work, and not the microphone.

Ive mirroed https://github.com/egorenar/snd-hda-codec-cs8409 just to be sure, but otherwize this is completley untouched:

```nix
{
  stdenv,
  lib,
  fetchgit,
  linuxKernel,
  kernel ? linuxKernel.kernels.linux_5_15,
  version ? "d8c9001418e6172099a0907f022534f152e29d71",
}:
stdenv.mkDerivation {
  inherit version;
  name = "sna-hda-codec-cs8409-${version}-module-${kernel.modDirVersion}";

  src = fetchgit {
    url = "https://git.euph.dev/snoweuph/snd-hda-codec-cs8409";
    rev = version;
    sha256 = "sha256-uwPz2d9tJP9Tp+j0lFFKy2Dn/sVSqV2MRLec469KU+E=";
  };

  hardeningDisable = ["pic"];
  nativeBuildInputs = kernel.moduleBuildDependencies;

  NIX_CFLAGS_COMPILE = ["-g" "-Wall" "-Wno-unused-variable" "-Wno-unused-function"];

  makeFlags =
    kernel.makeFlags
    ++ [
      "INSTALL_MOD_PATH=$(out)"
      "KERNELRELEASE=${kernel.modDirVersion}"
      "KERNEL_DIR=${kernel.dev}/lib/modules/${kernel.modDirVersion}/build"
    ];

  postPatch = ''
    printf '
    snd-hda-codec-cs8409-objs := patch_cs8409.o patch_cs8409-tables.o
    obj-$(CONFIG_SND_HDA_CODEC_CS8409) += snd-hda-codec-cs8409.o

    KBUILD_EXTRA_CFLAGS = "-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1"

    KERNELRELEASE ?= $(shell uname -r)
    KERNEL_DIR    ?= /lib/modules/$(KERNELRELEASE)/build
    PWD           := $(shell pwd)

    default:
    	make -C $(KERNEL_DIR) M=$(PWD) CFLAGS_MODULE=$(KBUILD_EXTRA_CFLAGS)

    install:
    	make -C $(KERNEL_DIR) M=$(PWD) modules_install
    ' \
    > Makefile

    sed --in-place 's|<sound/cs42l42.h>|"${kernel.dev}/lib/modules/${kernel.modDirVersion}/source/include/sound/cs42l42.h"|'  patch_cs8409.h
    sed --in-place 's|hda_local.h|${kernel.dev}/lib/modules/${kernel.modDirVersion}/source/sound/pci/hda/hda_local.h|'                                                      patch_cs8409.h
    sed --in-place 's|hda_jack.h|${kernel.dev}/lib/modules/${kernel.modDirVersion}/source/sound/pci/hda/hda_jack.h|'                                                        patch_cs8409.h
    sed --in-place 's|hda_generic.h|${kernel.dev}/lib/modules/${kernel.modDirVersion}/source/sound/pci/hda/hda_generic.h|'                                                  patch_cs8409.h
    sed --in-place 's|hda_auto_parser.h|${kernel.dev}/lib/modules/${kernel.modDirVersion}/source/sound/pci/hda/hda_auto_parser.h|'                                          patch_cs8409.h
  '';

  meta = {platforms = lib.platforms.linux;};
}
```

as https://github.com/egorenar/snd-hda-codec-cs8409 was a fork of your project, and since then quite some structure changed in the layout (with semengly multiple patch versions) I though, that you could maybe help me convert over to your patchset.
As far as I see, the config above only creates a custom Makefile and replaces some Imports in some of the patchfiles.

Would be cool to get some input on this (for example what patch set I need)

snoweuph, 2025-07-15:
thanks, gonna give it a try.
as for the kernel version, its 6.12.37 for me currently.
as from where the source files are coming from, thats
`${kernel.dev}/lib/modules/${kernel.modDirVersion}/source`. basicly those are as far as I understand inside a so called nix-store which is basicly their fancy sandboxing for their buildsystem for basilcy everything.

from the commit hash your showing, I guess I should use the patches from the patch_cirrus folder?

the egorenar fork was completely flat when it came to folder structure.

Well, thanks for your response, gonna continue tinkering.

snoweuph, 2025-07-15:
btw, noticed the readme, and that youre testing on 6.8. 
sadly can't try that one without packaging it my selfe, because nixos drops EOL kernels reallly aggressevly.

with 6.12 I get following errors trying to apply the patches from ` patch_cirrus` :
```sh

@nix { "action": "setPhase", "phase": "buildPhase" }
build flags: SHELL=/nix/store/ih68ar79msmj0496pgld4r3vqfr7bbin-bash-5.2p37/bin/bash O=\$\(buildRoot\) ARCH=x86_64 CROSS_COMPILE= INSTALL_MOD_PATH=\$\(out\) KERNELRELEASE=6.12.37 KERNEL_DIR=/nix/store/qqpcpwmjy084bsizysmkm5cnak28iv9l-linux-6.12.37-dev/lib/modules/6.12.37/build
make -C /nix/store/qqpcpwmjy084bsizysmkm5cnak28iv9l-linux-6.12.37-dev/lib/modules/6.12.37/build M=/build/snd_hda_macbookpro-259cc39 CFLAGS_MODULE="-DAPPLE_PINSENSE_FIXUP -DAPPLE_CODECS -DCONFIG_SND_HDA_RECONFIG=1"
make[1]: Entering directory '/nix/store/qqpcpwmjy084bsizysmkm5cnak28iv9l-linux-6.12.37-dev/lib/modules/6.12.37/build'
  CC [M]  /build/snd_hda_macbookpro-259cc39/patch_cs8409.o
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:[m[K In function '[01m[Kcs42l42_mute[m[K':
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:484:59:[m[K [01;31m[Kerror: [m[K'[01m[KCS42L42_REG_HS_VOL_CHA[m[K' undeclared (first use in this function); did you mean '[01m[KCS42L42_REG_AMIC_VOL_MASK[m[K'?
  484 |                                 cs8409_i2c_write(cs42l42, [01;31m[KCS42L42_REG_HS_VOL_CHA[m[K, 0x3f);
      |                                                           [01;31m[K^~~~~~~~~~~~~~~~~~~~~~[m[K
      |                                                           [32m[KCS42L42_REG_AMIC_VOL_MASK[m[K
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:484:59:[m[K [01;36m[Knote: [m[Keach undeclared identifier is reported only once for each function it appears in
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:486:59:[m[K [01;31m[Kerror: [m[K'[01m[KCS42L42_REG_HS_VOL_CHB[m[K' undeclared (first use in this function); did you mean '[01m[KCS42L42_REG_AMIC_VOL_MASK[m[K'?
  486 |                                 cs8409_i2c_write(cs42l42, [01;31m[KCS42L42_REG_HS_VOL_CHB[m[K, 0x3f);
      |                                                           [01;31m[K^~~~~~~~~~~~~~~~~~~~~~[m[K
      |                                                           [32m[KCS42L42_REG_AMIC_VOL_MASK[m[K
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:489:59:[m[K [01;31m[Kerror: [m[K'[01m[KCS42L42_REG_AMIC_VOL[m[K' undeclared (first use in this function); did you mean '[01m[KCS42L42_REG_AMIC_VOL_MASK[m[K'?
  489 |                                 cs8409_i2c_write(cs42l42, [01;31m[KCS42L42_REG_AMIC_VOL[m[K, 0x9f);
      |                                                           [01;31m[K^~~~~~~~~~~~~~~~~~~~[m[K
      |                                                           [32m[KCS42L42_REG_AMIC_VOL_MASK[m[K
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:496:43:[m[K [01;31m[Kerror: [m[K'[01m[KCS42L42_REG_HS_VOL_MASK[m[K' undeclared (first use in this function); did you mean '[01m[KCS42L42_REG_AMIC_VOL_MASK[m[K'?
  496 |                                         & [01;31m[KCS42L42_REG_HS_VOL_MASK[m[K);
      |                                           [01;31m[K^~~~~~~~~~~~~~~~~~~~~~~[m[K
      |                                           [32m[KCS42L42_REG_AMIC_VOL_MASK[m[K
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:[m[K In function '[01m[Kcs42l42_handle_tip_sense[m[K':
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:639:37:[m[K [01;31m[Kerror: [m[K'[01m[Kstruct sub_codec[m[K' has no member named '[01m[Kforce_status_change[m[K'
  639 |         int status_changed = cs42l42[01;31m[K->[m[Kforce_status_change;
      |                                     [01;31m[K^~[m[K
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:641:16:[m[K [01;31m[Kerror: [m[K'[01m[Kstruct sub_codec[m[K' has no member named '[01m[Kforce_status_change[m[K'
  641 |         cs42l42[01;31m[K->[m[Kforce_status_change = 0;
      |                [01;31m[K^~[m[K
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:645:14:[m[K [01;31m[Kerror: [m[K'[01m[KCS42L42_JACK_INSERTED[m[K' undeclared (first use in this function)
  645 |         case [01;31m[KCS42L42_JACK_INSERTED[m[K:
      |              [01;31m[K^~~~~~~~~~~~~~~~~~~~~[m[K
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:657:14:[m[K [01;31m[Kerror: [m[K'[01m[KCS42L42_JACK_REMOVED[m[K' undeclared (first use in this function); did you mean '[01m[KCS42L42_SUB_REVID[m[K'?
  657 |         case [01;31m[KCS42L42_JACK_REMOVED[m[K:
      |              [01;31m[K^~~~~~~~~~~~~~~~~~~~[m[K
      |              [32m[KCS42L42_SUB_REVID[m[K
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:[m[K In function '[01m[Kcs42l42_jack_unsol_event[m[K':
[01m[K/build/snd_hda_macbookpro-259cc39/patch_cs8409.c:690:30:[m[K [01;31m[Kerror: [m[K'[01m[KCS42L42_HSDET_AUTO_DONE[m[K' undeclared (first use in this function); did you mean '[01m[KCS42L42_HSDET_AUTO_DONE_MASK[m[K'?
  690 |         if (reg_cdc_status & [01;31m[KCS42L42_HSDET_AUTO_DONE[m[K) {
      |                              [01;31m[K^~~~~~~~~~~~~~~~~~~~~~~[m[K
      |                              [32m[KCS42L42_HSDET_AUTO_DONE_MASK[m[K
[... cut ...]
    sed --in-place 's|hda_local.h|${kernel.dev}/lib/modules/${kernel.modDirVersion}/source/sound/pci/hda/hda_local.h|'             patch_cs8409.h
    sed --in-place 's|hda_jack.h|${kernel.dev}/lib/modules/${kernel.modDirVersion}/source/sound/pci/hda/hda_jack.h|'               patch_cs8409.h
    sed --in-place 's|hda_generic.h|${kernel.dev}/lib/modules/${kernel.modDirVersion}/source/sound/pci/hda/hda_generic.h|'         patch_cs8409.h
    sed --in-place 's|hda_auto_parser.h|${kernel.dev}/lib/modules/${kernel.modDirVersion}/source/sound/pci/hda/hda_auto_parser.h|' patch_cs8409.h
  '';

  meta = {platforms = lib.platforms.linux;};
}
````

davidjo, 2025-07-16:
So it looks as though nixOS builds from ${kernel.dev}/lib/modules/${kernel.modDirVersion}/source  - is that for the whole kernel so you still have sound/pci/hda under that directory??

Any way - in the appropriate directory  you need to setup the source files with my new filles and the patched files.
My repo is updated for the current version I state ie 6.8. So for this case the only files that need patching are the now original kernel source files patch_cs8409.c and patch_cs8409.h after they were added for Dell laptops.
(From other comments I think its supposed to be working up to 6.12 or 6.13).

So you could try to build this folder of files manually externally to the building - then just copy the result to the build folder.

First you need to  extract your kernels source sound/pci/hda directory to build/hda  in my main repo folder.

The run the following patches in the build/hda folder.

patch -b -p2 <../../patch_patch_cs8409.c.diff

patch -b -p2 <../../patch_patch_cs8409.h.diff

patch -b -p2 <../../patch_patch_cirrus_apple.h.diff
(this is needed due to a kernel source code change)

Going back to the top of my repo this command copies my new files into that build/hda folder.
cp patch_cirrus/Makefile patch_cirrus/patch_cirrus_* build/hda

Unfortunately I think you cant directly use the commit I suggested (which is where the symbol errors you are seeing are comming from) as there have been other kernel source changes.
Eg the symbol CS42L42_REG_AMIC_VOL doesnt exist any more.

snoweuph, 2025-07-17:
> The run the following patches in the build/hda folder.
> patch -b -p2 <../../patch_patch_cs8409.c.diff
> patch -b -p2 <../../patch_patch_cs8409.h.diff
> patch -b -p2 <../../patch_patch_cirrus_apple.h.diff [...]

In that order I get an error tht its missing the patch files in the build/hda, so I guess I copy the stuff over from patch_cirrus first?

also I guess I should also cp over the patch_cs8409 files?
so: `cp patch_cirrus/Makefile patch_cirrus/patch_cirrus_* patch_cirrus/patch_cs8409.* build/hda`?

but when I do so, I get:
```sh
 > patching file patch_cs8409.c
 > Reversed (or previously applied) patch detected!  Assume -R? [n]
 > Apply anyway? [n]
 > Skipping patch.
 > 3 out of 3 hunks ignored -- saving rejects to file patch_cs8409.c.rej
```

Also im now trying to build on top of master (259cc39e243daef170f145ba87ad134239b5967f)

im gonna try skipping the 
`patch_patch_cs8409.c.diff` and `patch_patch_cs8409.h.diff`

snoweuph, 2025-07-17:
also force applying clearly doesnt seem to work:
```sh
> patching file patch_cs8409.c
> Hunk #1 succeeded at 1272 with fuzz 2 (offset 2 lines).
> Hunk #2 FAILED at 1279.
> Hunk #3 succeeded at 1316 with fuzz 2 (offset 19 lines).
> 1 out of 3 hunks FAILED -- saving rejects to file patch_cs8409.c.rej
```
```sh
> patching file patch_cs8409.h
> Hunk #1 FAILED at 19.
> Hunk #2 succeeded at 319 with fuzz 2 (offset 23 lines).
> Hunk #3 FAILED at 323.
> Hunk #4 succeeded at 505 with fuzz 2 (offset 149 lines).
> 2 out of 4 hunks FAILED -- saving rejects to file patch_cs8409.h.rej
```

snoweuph, 2025-07-17:
>  (they were only needed before the kernel added its own version to handle the Dell laptops)
those patches dont exist for me in the kernel sources under ´/sound/pic/hda` 
so `patch_patch_cs8409.c.diff` and `patch_patch_cs8409.h.diff` fail telling me im missing those

snoweuph, 2025-07-17:
correction im only missing the `patch_cs8409.c` the `patch_cs8409.h` exists

snoweuph, 2025-07-18:
oh wait, my sources in General seem to only hold h files. Gonna habe to fix that.

closed 2025-07-18 by snoweuph

snoweuph, 2025-09-21:
wdym?
Just import it: https://nixos.wiki/wiki/NixOS_modules#Imports

jonathanfishbein1, 2025-09-21:
what does you configuration.nix look like? because I'm trying a few different things without success.  importing the the file in imports doing callPackage outside of imports and  getting errors either way

snoweuph, 2025-09-21:
I use flakes instead of configuration.nix

jonathanfishbein1, 2025-09-21:
`environment.systemPackages` ?

snoweuph, 2025-09-21:
Yea

jonathanfishbein1, 2025-09-21:
but isn't systemPackages for executables?

can you link the config you're mentioning here?

```Ive found somebodies config for https://github.com/egorenar/snd-hda-codec-cs8409 on NixOS and got it working, but only speakers work, and not the microphone```

davidjo, 2025-09-21:
Im afraid the microphone is a known issue - it has been discussed in a number of issues.
If you are talking about a headset mike I havent really implemented this (just use a usb headset which is nothing to do with this driver).
If you mean the internal mike the issue is it appears the macOS driver (which this module is essentially implementing) does not amplify the signal but assumes higher level os drivers do this. For linux this would be Alsa or previously pulse audio now pipewire - updates to do this are not part of this module as its not a kernel module function. Also I know it works for macbook pros 14,3 but not clear about eg imacs.
You may think the mike isnt working but it is just very quiet - I discovered this because I was testing using Audacity and discovered there was an input signal if blew hard on the laptop.
The amplification factor needed appears to be 256.

jonathanfishbein1, 2025-09-21:
sorry I probably should have cropped the part about the speakers and microphone.  I'm just looking for the NixOS configuration you're referencing that consumes your nix file

snoweuph, 2025-09-21:
> but isn't systemPackages for executables?
> 
> can you link the config you're mentioning here?
> 
> `Ive found somebodies config for https://github.com/egorenar/snd-hda-codec-cs8409 on NixOS and got it working, but only speakers work, and not the microphone`

don't have the link anymore, I think it was some twiter post, and it did't consume my nix config from above, it is what i based the nix package config above on, as I had to change how and what i patched and the Makefile.

As I'm not the most skilled person and I don't feel like beeing a suporter for packaging something in a format I'm still quite new to, I also decided to not upstream it to nix hardware, as they require you taking responsibility when upstreaming which I don't feel comfortable dooing with my current knowledge.

you can find my personal nix config here: <https://git.euph.dev/snoweuph/nix>
to see what I'm doing, but I'm not the right person to ask for further help.

as for this question:
> but isn't systemPackages for executables?

No its for Packages, its just happens to be, that a package can contain executables
(even multiple in one)

jonathanfishbein1, 2025-09-21:
I see thank you for sharing your nixos configuration.  that is quite a complicated config!

jonathanfishbein1, 2025-09-26:
ok I've successfully imported the file but I'm getting an error realizing the derivation

`Running phase: unpackPhase
@nix { "action": "setPhase", "phase": "unpackPhase" }
unpacking source archive /nix/store/xjcq6f9p7b8y5zjch4gaw9zdc89cpxx9-s>
source root is snd_hda_macbookpro-259cc39
Running phase: patchPhase
@nix { "action": "setPhase", "phase": "patchPhase" }
patching file patch_cs8409.c
Hunk #1 succeeded at 1440 (offset 170 lines).
Hunk #2 succeeded at 1449 (offset 170 lines).
Hunk #3 succeeded at 1478 (offset 170 lines).
patching file patch_cs8409.h
Hunk #2 succeeded at 305 (offset 3 lines).
Hunk #3 succeeded at 332 (offset 3 lines).
Hunk #4 succeeded at 368 (offset 3 lines).
patching file patch_cirrus_apple.h
/build/snd_hda_macbookpro-259cc39
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix { "action": "setPhase", "phase": "updateAutotoolsGnuConfigScripts>
Running phase: configurePhase
@nix { "action": "setPhase", "phase": "configurePhase" }
no configure script, doing nothing
Running phase: buildPhase
@nix { "action": "setPhase", "phase": "buildPhase" }
build flags: SHELL=/nix/store/q7sqwn7i6w2b67adw0bmix29pxg85x3w-bash-5.>
make: *** empty variable name.  Stop.`

Any ideas? Appreciate your help

jonathanfishbein1, 2025-09-26:
before I do that I'm noticing some things in the printed out build flags

`       > build flags: SHELL=/nix/store/q7sqwn7i6w2b67adw0bmix29pxg85x3w-bash-5.3p3/bin/bash O=\$\(buildRoot\) --eval=undefine modules CC=/nix/store/82kmz7r96navanrc2fgckh2bamiqrgsw-gcc-14.3.0/bin/gcc LD=/nix/store/c43ry7z24x3jhnjlj4gpay8a4g2p3x1h-binutils-2.44/bin/ld AR=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/ar NM=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/nm STRIP=/nix/store/c43ry7z24x3jhnjlj4gpay8a4g2p3x1h-binutils-2.44/bin/strip OBJCOPY=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/objcopy OBJDUMP=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/objdump READELF=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/readelf HOSTCC=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/cc HOSTCXX=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/c++ HOSTAR=/nix/store/l19cddv64i52rhcwahif8sgyrd3mhiqb-binutils-wrapper-2.44/bin/ar HOSTLD=/nix/store/l19cddv64i52rhcwahif8sgyrd3mhiqb-binutils-wrapper-2.44/bin/ld ARCH=x86_64 CROSS_COMPILE= INSTALL_MOD_PATH=\$\(out\) KERNELRELEASE=6.12.47 KERNEL_DIR=/nix/store/adl3dfzqwk28d2phvrjd51y16b2smsz7-linux-6.12.47-dev/lib/modules/6.12.47/build
       > make: *** empty variable name.  Stop.`

the buildRoot and out variables not being resolved seems suspicous

snoweuph, 2025-09-27:
that sounds to me like you werent passing

`config.boot.kernelPackages.kernel` as your `kernel` as shown above.

> <https://github.com/davidjo/snd_hda_macbookpro/issues/153#issuecomment-3088737866>
> ```nix
> (pkgs.callPackage ./packages/snd-hda-cirrus {
>   kernel = config.boot.kernelPackages.kernel;
> })
> ```

Techwizz-somboo, 2026-01-21:
> before I do that I'm noticing some things in the printed out build flags
> 
> ` > build flags: SHELL=/nix/store/q7sqwn7i6w2b67adw0bmix29pxg85x3w-bash-5.3p3/bin/bash O=\$\(buildRoot\) --eval=undefine modules CC=/nix/store/82kmz7r96navanrc2fgckh2bamiqrgsw-gcc-14.3.0/bin/gcc LD=/nix/store/c43ry7z24x3jhnjlj4gpay8a4g2p3x1h-binutils-2.44/bin/ld AR=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/ar NM=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/nm STRIP=/nix/store/c43ry7z24x3jhnjlj4gpay8a4g2p3x1h-binutils-2.44/bin/strip OBJCOPY=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/objcopy OBJDUMP=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/objdump READELF=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/readelf HOSTCC=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/cc HOSTCXX=/nix/store/95k9rsn1zsw1yvir8mj824ldhf90i4qw-gcc-wrapper-14.3.0/bin/c++ HOSTAR=/nix/store/l19cddv64i52rhcwahif8sgyrd3mhiqb-binutils-wrapper-2.44/bin/ar HOSTLD=/nix/store/l19cddv64i52rhcwahif8sgyrd3mhiqb-binutils-wrapper-2.44/bin/ld ARCH=x86_64 CROSS_COMPILE= INSTALL_MOD_PATH=\$\(out\) KERNELRELEASE=6.12.47 KERNEL_DIR=/nix/store/adl3dfzqwk28d2phvrjd51y16b2smsz7-linux-6.12.47-dev/lib/modules/6.12.47/build > make: *** empty variable name. Stop.`
> 
> the buildRoot and out variables not being resolved seems suspicous as does the CROSS_COMPILE=

Do you remember what you did to fix this? I am having the same problem.