davidjo/snd_hda_macbookpro #156: Microphone not working on iMac 18,3 27 5K (2017)

Marek-mahav, 2025-08-01:
Hello,

I am writing to report an issue with the built-in microphone on my iMac18,3 (27-inch, 2017) with an AMD graphics card and a Cirrus Logic CS8409 audio codec.

I was using Ubuntu 24.04.02 with the Wayland desktop environment but was experiencing problems with the audio drivers, specifically with the microphone. Due to a lack of proper driver support for this model, I decided to reinstall macOS Ventura in order to collect relevant system information for your development team.

I have performed a system diagnostics and am sending you the complete output. I am willing to provide any further diagnostic information regarding the entire system, not just the audio, if needed.

Thank you for your hard work and effort.

Best regards,
Marek HavlikLast 

login: Fri Aug  1 18:55:33 on ttys000
mahav@iMac-uzivatela-Marek ~ % echo "--- SYSTEM PROFILER AUDIO DATA ---"; system_profiler SPAudioDataType; echo "--- KEXTSTAT AUDIO ---"; kextstat | grep -i audio; echo "--- IOREG CIRRUS LOGIC ---"; ioreg -l | grep -i "cirrus"
--- SYSTEM PROFILER AUDIO DATA ---
Audio:

    Devices:

        Built-in Microphone:

          Default Input Device: Yes
          Input Channels: 2
          Manufacturer: Apple Inc.
          Current SampleRate: 44100
          Transport: Built-in
          Input Source: Internal Microphone

        Built-in Output:

          Default Output Device: Yes
          Default System Output Device: Yes
          Manufacturer: Apple Inc.
          Output Channels: 2
          Current SampleRate: 44100
          Transport: Built-in
          Output Source: Internal Speakers

--- KEXTSTAT AUDIO ---
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
  165    5 0xffffff7f960a6000 0x1dff0    0x1dff0    com.apple.iokit.IOAudioFamily (440.2) 626EA2F4-ED14-3987-AEEE-C183C95B4731 <164 7 6 3 1>
  168    0 0xffffff7f94f7a000 0xffc      0xffc      com.apple.driver.AppleAudioClockLibs (240.1) B4907148-964A-36F2-A97E-E31568A697F8 <9 7 6 3 1>
  190    0 0xffffff7f95f22000 0x3000     0x3000     com.apple.driver.AudioAUUC (1.70) 1F0EAF2B-E5B4-3FE2-941B-E00EDA061DC8 <165 115 17 16 9 7 6 3 1>
--- IOREG CIRRUS LOGIC ---
mahav@iMac-uzivatela-Marek ~ % echo "--- SYSTEM OVERVIEW ---"
system_profiler SPHardwareDataType
echo "--- DISPLAY/GRAPHICS INFO ---"
system_profiler SPDisplaysDataType
echo "--- USB DEVICES ---"
system_profiler SPUSBDataType
echo "--- NETWORK INFO ---"
system_profiler SPNetworkDataType
echo "--- AUDIO DEVICES ---"
system_profiler SPAudioDataType
echo "--- KERNEL EXTENSIONS (DRIVERS) ---"
kextstat
--- SYSTEM OVERVIEW ---
Hardware:

    Hardware Overview:

      Model Name: iMac
      Model Identifier: iMac18,3
      Processor Name: Quad-Core Intel Core i5
      Processor Speed: 3,4 GHz
      Number of Processors: 1
      Total Number of Cores: 4
      L2 Cache (per Core): 256 KB
[... 360 lines cut ...]
  166    2 0xffffff7f95365000 0x3ffc     0x3ffc     com.apple.iokit.IOHDAFamily (440.2) 900235B5-1D57-31BE-B43A-B2562B64C9FB <7 6 3 1>
  167    1 0xffffff7f95241000 0x12ff4    0x12ff4    com.apple.driver.AppleHDAController (440.2) E92DDE1B-681C-33D3-9FA7-610AE5EB249B <166 165 115 17 9 8 7 6 3 1>
  168    0 0xffffff7f94f7a000 0xffc      0xffc      com.apple.driver.AppleAudioClockLibs (240.1) B4907148-964A-36F2-A97E-E31568A697F8 <9 7 6 3 1>
  170    0 0xffffff8001bbf000 0x21000    0x21000    com.apple.driver.AppleThunderboltDPOutAdapter (8.5.1) 032AE0A3-AFE8-3B91-841E-BD8199DB7D9F <101 83 16 7 6 3 1>
  172    0 0xffffff7f95e45000 0x3e000    0x3e000    com.apple.driver.AppleThunderboltIP (4.0.3) FA0DE28D-EFE8-3CDA-A937-9BD0F75DCCF7 <83 51 17 7 6 3 1>
  173    0 0xffffff7f813c3000 0x2ffb     0x2ffb     com.apple.kext.AMDRadeonX4000HWServices (4.1.4) 3C88D443-E039-3F45-B4EB-173411D6AE83 <115 17 16 9 7 6 3 1>
  174    0 0xffffff7f80fbe000 0x3c5ff2   0x3c5ff2   com.apple.kext.AMDRadeonX4000 (4.1.4) 979721D4-AE1C-3A78-B73A-DFE9268D4532 <125 115 93 17 9 7 6 3 1>
  175    0 0xffffff7f95073000 0x5f000    0x5f000    com.apple.driver.AppleGFXHDA (240.1) 780EA320-85E9-387C-8411-9CF9CBC76ADC <165 122 115 17 9 8 7 6 3 1>
  176    0 0xffffff7f95d82000 0x3000     0x3000     com.apple.driver.AppleIntelPCHPMC (2.0.1) A09B9284-7028-3F99-A27A-F32F6F312B3B <111 18 17 16 6 3>
  177    0 0xffffff7f82d84000 0x10a000   0x10a000   com.apple.kext.AMDRadeonX4200HWLibs (1.0) 78332A0E-0770-3AEF-B912-79D7A70EBC92 <17 7 6 3 1>
  178    1 0xffffff7f96374000 0xf000     0xf000     com.apple.kext.OSvKernDSPLib (530) 35810D8D-EDB2-331A-8E17-59879627A177 <8 7 6 3>
  179    1 0xffffff7f95271000 0xe2ff8    0xe2ff8    com.apple.driver.DspFuncLib (440.2) A199D19B-2689-3DFF-B461-A5956BC976ED <178 165 164 67 8 7 6 3 1>
  180    0 0xffffff7f9518a000 0x8fff0    0x8fff0    com.apple.driver.AppleHDA (440.2) D8129302-91D6-382A-9EEA-21DAE11270F1 <179 167 166 165 122 115 8 7 6 3 1>
  181    0 0xffffff7f80f8e000 0x24ff4    0x24ff4    com.apple.kext.AMDFramebuffer (4.1.4) AD3A9718-3CC3-3CC1-BF96-D6AB7942C314 <117 115 17 16 9 7 6 3 1>
  183    0 0xffffff7f80fb8000 0x2ff8     0x2ff8     com.apple.kext.AMDRadeonServiceManager (4.1.4) DC124877-88CB-3EA3-AE49-0AB02361D603 <17 6 3 1>
  184    0 0xffffff7f95f13000 0x3000     0x3000     com.apple.driver.AppleUpstreamUserClient (3.6.9) 9C1B870C-F5F0-370D-8630-9B89E6C03A11 <115 17 16 9 7 6 3 1>
  185    1 0xffffff7f96588000 0xcff7     0xcff7     com.apple.filesystems.exfat (1.4) E4340F72-F8F7-3D27-9A2F-E3AD67A4D4C3 <9 8 7 6 3 1>
  186    0 0xffffff7f9656b000 0x8000     0x8000     com.apple.filesystems.autofs (3.0) 6631673F-0234-3275-BA34-5F3104B29D6B <158 9 8 7 6 3 2 1>
  187    1 0xffffff7f96266000 0x6000     0x6000     com.apple.driver.X86PlatformShim (1.0.0) F95AA991-F15C-371A-8966-E764C030D925 <112 111 18 9 6 3>
  188    0 0xffffff7f95166000 0x1eff8    0x1eff8    com.apple.driver.AGPM (131) 5740686E-249C-3E77-823A-32DB73D4512E <122 116 115 111 17 8 7 6 3 1>
  189    0 0xffffff7f95dde000 0x1fff     0x1fff     com.apple.driver.ApplePlatformEnabler (2.7.0d0) 221D07C8-329A-3C68-9135-7F79FD1C3155 <9 7 6 3>
  190    0 0xffffff7f95f22000 0x3000     0x3000     com.apple.driver.AudioAUUC (1.70) 1F0EAF2B-E5B4-3FE2-941B-E00EDA061DC8 <165 115 17 16 9 7 6 3 1>
  192    0 0xffffff8001b52000 0x1fff     0x1fff     com.apple.driver.AppleUSBMassStorageInterfaceNub (554) 83AAC4DB-2FBB-3387-BD9B-EDD67232629F <38 37 36 35 25 17 7 6 3 1>
  193    0 0xffffff8002f6f000 0x5000     0x5000     com.apple.iokit.SCSITaskUserClient (482.120.2) B2CB70AD-5033-31A1-91FF-81EFB25941A2 <36 35 7 6 3 1>
mahav@iMac-uzivatela-Marek ~ %

davidjo, 2025-08-01:
Unfortunately this is a known issue - for some reason iMac internal microphones dont seem to work using the coding developed for MacbookPros (there are other issues aswell that required code changes - swapped nids, swapped polarity jacks that have attempted fixes).
However I dont have an iMac so I cant really do anything about this - the original code was developed by creating a logging Apple HDA kext driver but that was only for High Sierra and has to be run on the actual hardware (remember there is no information provided by Apple and a lot of the chips involved are custom versions for Apple).
However this has to be done on a pre-sealed boot volume OS so Ventura I believe is out as it uses the sealed boot volume - I think it has to be one before Big Sur.

Your best bet is likely to get the Apple Bootcamp disk, get the cs408_{number}.inf file and look for the AB 14/15 and AB 16 quad APO defs - I think these may refer to the iMacs - as it does define some HDA verbs for setting up - at least I did find some information for MacbookPros (which seems to be AB 57) - never looked for iMacs.
(the 14/15/16 seem to come from definitions in the AppleHDA.kext under Resources in the layout and Platforms files for the TAS audio chips - which seem to be what iMacs of that vintage were using).

PS There is no development team its just something I was able get to work nearly 10 years ago now.

davidjo, 2025-08-02:
Actually lets just make sure the mike really is not working.

You need to check the Ubuntu settings Audio option for input and make sure its set to Internal Microphone - also make sure thats an option - there is no automatic switching for the mikes.

Now another issue is the sound is very, very low even when working (this has to be fixed by adding pipewire or alsa setup files to amplify the input - but this is the hardware output for mikes - amplification needs to be done by higher levels of the audio system as in OSX.

I only noticed the internal mike was working because I was using Audacity for testing.
In Audacity you can set it to record from the microphone and I tried and didnt see any sound initially but then blew on things and saw that Audacities input audio monitor registered a signal.
So then I recorded some audio - which you cant really hear - but Audacity can apply effects and one of these is amplification which I applied and then could hear the audio.

davidjo, 2025-08-03:
As it looks as though the internal mike is working fixing amplification is not a function of this driver.

You might want to look through the issues (in particular #135) for suggestions how to fix this (there have been many discussions of this in the past).
The addition of amplification does not require updates to the kernel but are user level additions.
There are two possible paths - adding an alsa user driver - this requires creating a .asoundrc file in your home directory (look for .asoundrc in issues to see many older discussions).
I think now your best bet is wireplumber - which is the pipewire system for adding sound effects (including amplication) (when this project was started pipewire didnt exist).
However you have as much knowledge as I do about this - all I have done is google for information and try things and have not really used wireplumber at all.

I have no idea what a UCM2 profile is.

As has been noted the sound using this driver is quite different from the macs - and one of the primary reasons does seem to be extensive DSP processing by Apple.
If you look in the Ventura AppleHDA.kext Resources directory you will see a platform file and layout files - only one of the layout files refers to your machine type.
When I looked at it there were may DSP coefficients - in particular one thing I did notice a label to one called DspMozartCompressor but although there are coeffcients we have no idea what the actual DSP functions are (Im sure this is not the same as the APO filters - although yet again the application of DSP processing is  a user level either alsa or wireplumber function).

davidjo, 2025-08-07:
So did a bit of googling and it appears UCM2 profiles are an Alsa thing - not clear how that now interacts with pipewire. (For pro audio you apparently used to use JackD for real time audio but now apparently those functions are also in pipewire).
Heres my analysis - Apple wanted to have a 24 bit audio path for its audio for audio professionals. However most (if not all) standard HDA chips (with internal amplifier nodes) use 16 bit audio. So what Apple did was use the very non-standard 8409 chip which has no internal amplifiers and essentially just transforms the bus bytestream to a TDM bytestream which is what the 24 bit DAC/Amp chips like the TAS5764L in iMacs take as input.
When I looked at the raw Audacity data for my MacbookPro internal mike I could see only the low order 16 bits were actually being used. So it appears the internal mikes are still using 16 bit data but are packed into the low order of 24 bits hence sound silent. Therefore a multiplication of 256 is needed (8 bits) to shift the sound up - which is what I ended up trying in Audacity.

Apparently Asahi linux (linux for Arm machines which is what Im using now) has a completely new audio drivers/setup with proper drivers which give a much closer Apple sound (although Apple has given up on the 8409 chip and now uses its own silicon but still uses commerical DAC/amp chips). Unfortunately Asahi linux isnt working for my machine yet but I have thought of looking at its audio implementation for info about how to handle the higher level audio setup.